lumapu/ahoy #639: "Polling will stop" message even though night-time communication is enabled.

Argafal, 2023-02-01:
Version: 0.5.78, on ESP8266

I added GPS coordinates in the settings, using a decimal point as required. Night-time communication is enabled/unchanged, i.e. the setting "disable night communication" is not checked. Upon save the DTU was rebooted.

On the Ahoy homepage I find the message: "Polling inverter(s), will stop at 1.2.2023, 17:27:38"

I believe that this is incorrect as I night-time communication is enabled.

lumapu, 2023-02-01:
nobody tested that before, thank you for pointing that out. 👍

Argafal, 2023-02-08:
```
index cb16ddc..3a1628b 100644
--- a/src/web/html/index.html
+++ b/src/web/html/index.html
@@ -142,15 +142,15 @@
                     e.addEventListener("click", setTime);
                 }
 
-                if(obj["ts_sunrise"] > 0) {
+                if(obj["disNightComm"]) {
                     if(((obj["ts_sunrise"] - obj["ts_offset"]) < obj["ts_now"])
                         && ((obj["ts_sunset"] + obj["ts_offset"]) > obj["ts_now"])) {
                        commInfo = "Polling inverter(s), will stop at sunset " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
                     }
-                    else if(obj["disNightComm"]) {
+                    else {
                        commInfo = "Night time, inverter polling disabled, ";
                         if(obj["ts_now"] > (obj["ts_sunrise"] - obj["ts_offset"])) {
                            commInfo += "stopped at " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
                         }
                         else {
                             commInfo += "will start polling at " + (new Date((obj["ts_sunrise"] - obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
```

closed 2023-03-27 by lumapu