lumapu/ahoy #643: Exception running 0.5.78 on ESP8266

Argafal, 2023-02-03:
Development binaries from ahoydtu.de, flashed using the webflasher, running on Wemos D1 mini v3

With this version I get random reboots. I caught one of them and the decoded stack trace is below:

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000df64
EXCVADDR: 0x00000000

Decoding stack results
0x4022d816: frame_downheap at /workdir/repo/gcc-gnu/libgcc/unwind-dw2-fde.c line 535
0x40239f26: tcp_slowtmr at core/tcp.c line 1414
0x40238c49: dns_recv at core/dns.c line 1353
0x4023a1f8: pbuf_realloc at core/pbuf.c line 438
0x4023a1f8: pbuf_realloc at core/pbuf.c line 438
0x40234f6d: _fputwc_r at /workdir/repo/newlib/newlib/libc/stdio/fputwc.c line 172
0x4023a1f8: pbuf_realloc at core/pbuf.c line 438
0x4023a134: pbuf_header_force at core/pbuf.c line 658
0x40235ab1: ieee754_acos at /workdir/repo/newlib/newlib/libm/math/e_acos.c line 90
0x402354b1: _isatty_r at /workdir/repo/newlib/newlib/libc/sys/xtensa/isatty.c line 15
0x40235b00: ieee754_acos at /workdir/repo/newlib/newlib/libm/math/e_acos.c line 91
0x4023a134: pbuf_header_force at core/pbuf.c line 658
0x4023e4d5: ip4addr_aton at core/ipv4/ip4_addr.c line 211
0x4023e703: igmp_lookup_group at core/ipv4/igmp.c line 263
0x4023eb62: ip_napt_find at core/ipv4/ip4.c line 539
0x402409d7: tcp_oos_insert_segment at core/tcp_in.c line 1081
0x4022d89c: frame_heapsort at /workdir/repo/gcc-gnu/libgcc/unwind-dw2-fde.c line 568
0x4022e202: search_object at /workdir/repo/gcc-gnu/libgcc/unwind-dw2-fde.c line 598
0x4023167c: strncmp_P at /workdir/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c line 309
0x4022d9c8: read_encoded_value_with_base at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 234
```

stefan123t, 2023-02-03:
Found this in the Espressif docs about LoadProhibited, StoreProhibited:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited

> LoadProhibited, StoreProhibited

> These CPU exceptions happen when an application attempts to read from or write to an invalid memory location. The address which has been written/read is found in the EXCVADDR register in the register dump. If this address is zero, it usually means that the application has attempted to dereference a NULL pointer. If this address is close to zero, it usually means that the application has attempted to access a member of a structure, but the pointer to the structure is NULL. If this address is something else (garbage value, not in 0x3fxxxxxx - 0x6xxxxxxx range), it likely means that the pointer used to access the data is either not initialized or has been corrupted.

Argafal, 2023-02-04:
I caught it one more time and the decoded Exception is below. I wonder if I mixed up the .elf file when decoding it the first time, see below for the new results. Could this be related to #644 as well?

Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000df64
EXCVADDR: 0x00000000

Decoding stack results
0x4023aa73: __d2b at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 779
0x40238a66: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c line 352
0x4023a23d: _Balloc at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 128
0x4023aa76: __d2b at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c line 779
0x40239569: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c line 853
0x4023ad4c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
0x40235861: __cvt at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_float.c line 102
0x4023ac88: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 182
0x402363a1: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x4023ad4c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
0x402363a1: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x4023ad4c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
0x4023ac88: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 182
0x4023ec5d: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x4023ee8b: new_linkoutput at glue-lwip/lwip-git.c line 272
0x4023f2ee: ethernet_output at netif/ethernet.c line 312
0x4024686c: etharp_output_to_arp_index at core/ipv4/etharp.c line 769
0x40246940: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x402482b0: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x4023fb9d: mem_malloc at core/mem.c line 210
0x4024025e: pbuf_alloc_LWIP2 at core/pbuf.c line 284

Argafal, 2023-02-09:
I reproduced the exception with 0.5.83:

```
I: resetPayload: id: 0
I: (#0) Requesting Inv SN 114172221779
I: (#0) enqueuedCmd: 0xb
I: (#0) sendTimePacket
I: TX 27B Ch23 | 15 72 22 17 79 86 99 51 75 80 0B 00 63 E4 CC 8D 00 00 00 01 00 00 00 00 28 A6 D2 
RX whatHappened Ch61 001
RX whatHappened Ch61 001
RX whatHappened Ch40 001
I: RX 27B Ch61 | 95 72 22 17 79 72 22 17 79 01 00 01 01 81 00 E9 03 7F 01 7D 00 BA 02 C3 00 04 83 
I: RX 27B Ch61 | 95 72 22 17 79 72 22 17 79 02 AB 84 00 04 8E 6D 00 7A 00 77 09 5B 13 88 05 FA 64 
I: RX 23B Ch40 | 95 72 22 17 79 72 22 17 79 83 00 01 00 40 03 E8 00 93 00 01 AB D4 51 
I: procPyld: cmd:  0xb
I: procPyld: txid: 0x95
I: Payload (42): 00 01 01 81 00 E9 03 7F 01 7D 00 BA 02 C3 00 04 AB 84 00 04 8E 6D 00 7A 00 77 09 5B 13 88 05 FA 00 01 00 40 03 E8 00 93 00 01 
I: resetPayload: id: 1
I: (#1) Requesting Inv SN 116174403530
I: (#1) enqueuedCmd: 0xb
I: (#1) sendTimePacket
I: TX 27B Ch40 | 15 74 40 35 30 86 99 51 75 80 0B 00 63 E4 CC 97 00 00 00 34 00 00 00 00 4C 99 A9 
RX whatHappened Ch75 001
RX whatHappened Ch61 001
RX whatHappened Ch61 001
I: RX 27B Ch75 | 95 74 40 35 30 74 40 35 30 01 00 01 01 62 00 F5 00 F0 03 61 03 52 00 04 68 0E A2 
I: RX 27B Ch61 | 95 74 40 35 30 74 40 35 30 03 04 76 00 04 79 DA 00 04 81 6B 00 86 00 96 09 4F FB 
I: RX 27B Ch61 | 95 74 40 35 30 74 40 35 30 84 13 88 0E 6C 00 E0 00 9B 03 E6 00 9E 00 34 D7 E9 E2 
W: Frame 2 missing: Request Retransmit
I: TX 11B Ch61 | 15 74 40 35 30 86 99 51 75 82 9D 
RX whatHappened Ch3 001
I: RX 27B Ch3 | 95 74 40 35 30 74 40 35 30 02 00 03 FA AD 00 7B 00 7A 01 6E 01 19 01 38 04 05 8D 
I: procPyld: cmd:  0xb
I: procPyld: txid: 0x95
I: Payload (62): 00 01 01 62 00 F5 00 F0 03 61 03 52 00 04 68 0E 00 03 FA AD 00 7B 00 7A 01 6E 01 19 01 38 04 05 04 76 00 04 79 DA 00 04 81 6B 00 86 00 96 09 4F 13 88 0E 6C 00 E0 00 9B 03 E6 00 9E 00 34 

Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000df64
EXCVADDR: 0x00000000

Decoding stack results
0x40238f32: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c line 352
0x4023b230: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 232
0x40239a4d: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c line 853
0x4023b16c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 182
0x40236885: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x40236289: _printf_float at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_float.c line 330
0x40236900: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 211
0x4023b16c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 182
0x4023f141: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x4023f36f: new_linkoutput at glue-lwip/lwip-git.c line 272
0x4023f7d2: ethernet_output at netif/ethernet.c line 312
0x40246d50: etharp_output_to_arp_index at core/ipv4/etharp.c line 769
0x40246e24: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x40248794: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x40240081: mem_malloc at core/mem.c line 210
0x40240742: pbuf_alloc_LWIP2 at core/pbuf.c line 284
```

Argafal, 2023-02-26:
I have not yet been able to reproduce this exception with 0.5.92. I will keep an eye out for it and update this issue accordingly in a few days.

Argafal, 2023-03-09:
I have not been able to reproduce this after 0.5.92. I propose we close this issue! :)

closed 2023-03-27 by lumapu