lumapu/ahoy #645: ESP32 0.5.78 Nicht mehr erreichbar nach 2-3 Tagen

MiniOh, 2023-02-04:
Hallo, 

ich habe seit der 0.5.78 der Verhalten dass alle 2-3 Tage, erst die Webseite auf dem ESP32 nicht mehr erreichbar ist, aber noch ca. 1h Daten empfangen und auch per MQTT übertragen werden.
Danch werden auch keine neue Daten mehr übermittelt.
Hier hilft bisher nur stromlos machen.

Kann ich hier sinnvollte Daten oder Logs bereitstellen, und falls ja, wie?

Danke schon mal vorab.

Argafal, 2023-02-04:
Was bedeutet denn "die Webseite nicht mehr erreichbar ist"?

MiniOh, 2023-02-04:
Die DTU ist noch per Ping erreichbar, aber der Webserver antwortet nicht mehr per HTTP.

Gerri1, 2023-02-04:
Kann ich bestätigen beim ESP8266, nur das bei mir sich Ahoy auch nicht mehr anpingen lässt!

Aikonflo, 2023-02-04:
Ähnliches Problem bei mir.
ESP ist kurz nach dem Start per Web erreichbar danach „Seite kann nicht angezeigt werden“
Laut Netzwerk-Controller ist er aber noch verbunden.
WLAN-LED leuchtet dauerhaft mit kurzem Flackern. 

ESP 8266 D1 Mini
Ahoy ver. 0.5.65

lumapu, 2023-02-05:
wir arbeiten leider seit anfang an an diesem Problem. Zusätzlich verhält sich jeder ESP anders. Meiner (ESP8266) mit Version `0.5.78` läuft jetzt aktuell seit über 6 Tagen am Stück - alles geht.

MiniOh, 2023-02-06:
Vielen Dank für die Rückmeldung.
OK, das wusste ich nicht, dass es "solche Probelme" schon seit Beginn der Entwicklung gibt.

Ich selbst hatte das Verhalten noch nicht beobachten können.
Lediglich seit der 0.5.78 hatte ich das bereits 3 Mal.

Was "passiert" in dem Fehlerfall? Es ist ja schon etwas merkwürdig, dass zuerst die Funktion des Webservers "ausfällt", MQTT aber noch funktioniert, dann auch irgendwannd die Übermittlung per MQTT wegfällt und erst dann ist quasi die DTU selbst auch per Ping nicht mehr erreichbar ...

E1nste1n-AUT, 2023-02-06:
Wäre ein periodischer Reset eine Option, Hausnummer jeden Tag um 05:00 Uhr? Denn wenn die ahoyDTU 24h ohne Probleme läuft dann wäre das für mich OK.
Dann besteht zwar immer noch mein Problem mit der Zeitsynchronisation, aber das habe ich hier bereits versucht zu erläutern #642

roku133, 2023-02-06:
Das beschriebene Verhalten von 0.5.78 kann ich bei meinem ESP8266 bestätigen, nach drei Tagen ist die WLAN-Verbindung plötzlich unterbrochen, ping auf die DTU-Adresse bleibt erfolglos. Netz AHOY-DTU ist als Alternative vorhanden. Die DTU bleibt über Stunden in diesem Zustand. Nach Anmeldung funktioniert ping auf 192.168.4.1 nicht, die Anmeldung auf die Weboberfläche damit ebenfalls nicht.
Interessant ist, dass nach einigen Zugriffsversuchen auf AHOY-DTU die normale WLAN-Verbindung wieder aufgebaut wird.
DTU funktioniert ohne Bootvorgang (Nachweis durch uptime) wieder normal weiter.

Im Gegensatz dazu lief v0.5.70 mehr als 22 Tage ohne Unterbrechung, DTU war dabei zeitweise nicht mit dem stärksten AP verbunden.

beegee3, 2023-02-08:
Frage in die Runde: wie sind denn die RSSI Werte? Bei schlechter Verbindung ist die Web Verbindung träge. Auch wenn sie zustande kommt, wird die Reaktion im Laufe der Zeit zunehmend schlechter. Die Requests an AhoyDTU laufen asynchron, aber die meisten Seiten erzeugen in regelmäßigen Abständen neue Requests, die nicht zeitnah abgearbeitet werden können und es entsteht ein Stau. Vielleicht wird das System dadurch überfordert und es wäre besser, neue Requests erst dann zu erlauben, wenn der vorherige beantwortet wurde.
Auch WLAN Abbrüche und erfolglose Anmeldeversuche sind bei schlechter Verbindung natürlich häufiger zu beobachten.

Gerri1, 2023-02-08:
Da waren keine Auffälligkeiten vorhanden!
[image]

MiniOh, 2023-02-09:
Hallo, 

ich weiß nicht, ob es bei der Fehlersuche hilft.
Ich hatte jetzt 2 ESP32 mit 0.5.78 laufen, eine hat die Kommunikation mit den Wechselrichtern aktiv, die andere nicht.
Die mit aktiver Kommunikation ist jeweils nach 2 Tagen und 3 Stunden bzw nach 2 Tagen und 14 Stunden nicht mehr erreichbar gewesen. 
Die ohne Kommunikation läuft ohne Ausfall durch.

Argafal, 2023-02-09:
Danke @MiniOh. Auf esp8266 geht ahoy für mich viel schneller kaputt, ich hab eine uptime von Minuten statt Tagen. Das ist natürlich fürs debugging ganz praktisch ;) 
Deine Beobachtung deckt sich mit meiner, und @knickohr hat das auch schon erwähnt: Ohne Kommunikation scheint ahoy stabil zu laufen. Für mich läuft nachts alles stabil durch, Probleme (reboots/nicht erreichbar/webserver weg) scheinen nur tagsüber bei aktiver Kommunikation aufzutreten.

MiniOh, 2023-02-09:
Klingt ja fest etwas danach, dass irgendein Speicher "vollläuft"?
160 vs. 512kB Ram .... was evtl. auch einen zeitlichen Unterschied macht?

Argafal, 2023-02-09:
Ja, das kann gut sein. Vielleicht auch ein Zusammenhang zur Anzahl Inverter? Ich hab drei in AHOY.

MiniOh, 2023-02-09:
Ja, auch möglich, ... aktuell habe ich auch 3, werden aber mehr.
Gibt es evtl. die Möglichkeit, die RAM Auslastung zu überwachen? OpenDTU macht das, wenn ich es richtig in Erinnerung habe.

Argafal, 2023-02-09:
Gibt es zu den zwei new() Statements in web/RestApi.h ein dazugehöriges delete()?
```
 63             AsyncJsonResponse* response = new AsyncJsonResponse(false, 8192);
[....]
103             AsyncJsonResponse* response = new AsyncJsonResponse(false, 200);
```

Gloomyeye, 2023-02-09:
> Ja, das kann gut sein. Vielleicht auch ein Zusammenhang zur Anzahl Inverter? Ich hab drei in AHOY.

Ich habe auch die Version 0.5.78 installiert und nur einen Wechselrichter angemeldet (HM-600). Der ESP ist auch nach einigen Stunden nicht mehr erreichbar.

MiniOh, 2023-02-09:
> Ja, das kann gut sein. Vielleicht auch ein Zusammenhang zur Anzahl Inverter? Ich hab drei in AHOY.

Ich habe mir jetzt mal über einen Zeitraum von einer Stunde die free_heap Size angesehen.
Diese bleibt ohne Kommunkation recht gleich.
Mit aktiver Kommunikation wird diese pro Stunde ca. 7000 kleiner, in meinem Fall von rund 183000 zu 176000.
Hochgerechnet, sollte diese nach ca. 26 Stunden (Kommunikationsstunden) auf 0 sein, was bei mir ungefähr passen würde, dass nach 2 1/2 Tagen die DTU ausfällt.

stefan123t, 2023-02-09:
@Argafal ja die new() Statements und andere dynamisch allokierte Speicherbereiche hat @lumapu aktuell auch im Verdacht und versucht sie sukszessive zu ersetzen.

knickohr, 2023-02-09:
Reden wir wirklich von ESP32 ?  Die Version 76, nicht 78 !, lief bei mir 12 Tage am Stück ohne Probleme mit 2 Invertern durch. Ja, kann gut sein das wir da schon wieder ein neues Problem haben.

Auf dem 8266 habe ich seit der letzten Stable (66) nur für wenige Stunden Erfolg, dann schlug der Watchdog zu 😪

MiniOh, 2023-02-09:
> Reden wir wirklich von ESP32 ? Die Version 76, nicht 78 !, lief bei mir 12 Tage am Stück ohne Probleme mit 2 Invertern durch. Ja, kann gut sein das wir da schon wieder ein neues Problem haben.
>
> Auf [...]

Ja, bei mir 2x ESP32 im Einsatz.
Mit Kommunikation gibts nach etwas mehr als 2 Tagen das Problem, ohne Kommunikation läuft die DTU.

Argafal, 2023-02-09:
Ich hab (wie explizit erwähnt) den Vergleich zu ESP8266 berichtet, für den Fall das das die gleiche Ursache ist und sie beim ESP8266 nur schneller auftritt.
 
Ob jemand von euch die mit ESP32 unterwegs sind die Exception beim Reboot fangen kann?

MiniOh, 2023-02-09:
Wenn du mir ein paar Infos dazu gibts, ... kann ich das evtl. machen.
Alternativ kannst du bei Dir auch mal die free_heap beobachten?

Argafal, 2023-02-09:
> Ich habe mir jetzt mal über einen Zeitraum von einer Stunde die free_heap Size angesehen.
> Diese bleibt ohne Kommunkation recht gleich.
> Mit aktiver Kommunikation wird diese pro Stunde ca. 7000 [...]

Ist das immer noch so, wenn du MQTT deaktivierst? Mit anderen Worten: Können wir eingrenzen, ob dein immer kleiner werdender Heap durch die Kommunikation mit den Invertern an sich ist, oder durch die Weiterleitung der Daten über MQTT (oder beides)?

Argafal, 2023-02-09:
Mach ich schon, deswegen kam mir ja der Gedanke. Aber wie @knickohr ganz richtig angemerkt hat, ich bin auf ESP8266, also hab ich hier in dem Issue nicht viel zu melden. Du schon ;)

Wenn du Lust auf ein Experiment musst du denke ich nur den MQTT server aus den Settings nehmen und speichern und neu starten. Und dann wieder den heap beobachten, so wie du es auch vorher gemacht hast.

beegee3, 2023-02-09:
hab' seit einer Stunde einen ESP32 ohne MQTT laufen (Version 0.5.82): heap_free stabil bei ~190000.
Ist evtl. ein Problem bei MQTT, würde trotzdem andere Bereiche die dynamisch Speicher allokieren nicht ausschließen.

Argafal, 2023-02-09:
> hab' seit einer Stunde einen ESP32 ohne MQTT laufen (Version 0.5.82): heap_free stabil bei ~190000.
> Ist evtl. ein Problem bei MQTT, würde trotzdem andere Bereiche die dynamisch Speicher allokieren nicht ausschließen.

Dem letzten Punkt stimme ich natürlich total zu.

Hier übrigens ähnliche Ergebnisse auf ESP8266 ohne MQTT: heap um die 13000, uptime 30 minuten, Webserver nicht abgeraucht. Ein neuer Rekord für mich auf 0.5.83 tagsüber.

beegee3, 2023-02-09:
> Hier übrigens ähnliche Ergebnisse auf ESP8266 ohne MQTT

dann vielleicht in MQTT als erstes suchen?!

Argafal, 2023-02-09:
Ich hab die String() concatenations in der publish Methode von MQTT im Verdacht. Ich baue die gerade mal testweise um...

beegee3, 2023-02-09:
@Argafal du meinst in `sendDiscoveryConfig`? Gab's da nicht neulich ein size Problem?
Die String() concatenations sind es m.E. nicht, aber vielleicht der lokal erzeugte Buffer ganz unten direkt vor dem publish. Zwar wird am Ende ein `delete` aufgerufen, aber wird wirklich gelöscht, auch wenn das publish noch im Ausgang hängt?
Wenn du eh schon dran bist, ersetze doch mal den Buffer durch einen globalen hinreichender Größe, also z.B.
`char mBuf[4096];` und in `sendDiscoveryConfig`
```
                    memset(mBuf, 0, 4096);
                    serializeJson(doc2, mBuf, size);
                    publish(topic, mBuf, true, false);
```

Argafal, 2023-02-09:
Nein, ich meinte in: `void publish(const char *subTopic, const char *payload, bool retained = false, bool addTopic = true) {`

See the first answer in https://arduino.stackexchange.com/questions/75745/how-to-fix-memory-leak-code-in-esp8266-nodemcu-caused-by-string-concenation 

No idea if this is truly the issue, but I thought it was worth trying.

MiniOh, 2023-02-09:
Kann es sein, dass mit der 0.5.83 nun auch Zero Werte bei inverter/total/YieldDay, inverter/total/YieldTotal und inverter/total/P_AC per MQTT übertragen werden, wenn die WR Abends offline gehn.
Mit der 0.5.78 war das nicht der Fall.

MiniOh, 2023-02-09:
> Mach ich schon, deswegen kam mir ja der Gedanke. Aber wie @knickohr ganz richtig angemerkt hat, ich bin auf ESP8266, also hab ich hier in dem Issue nicht viel zu melden. Du schon ;)
>
> Wenn du Lust [...]

Aus meiner Sicht müsste das Abnehmen der free_heap eher von der Kommunikation mit den Wechselrichtern her rühren.
Hier die heutige Aufzeichnung.
Bis ca. 18 Uhr nahm der Wert kontinuierlich ab, danach waren die WR offline, die MQTT Kommunikation ist aber weiterhin aktiv, dennoch wurde der Wert ab rund 167000 nicht mehr kleiner.

[image]

beegee3, 2023-02-09:
MQTT hat aber auch nichts mehr zu berichten außer minütlich uptime, wifirssi und free_heap

knickohr, 2023-02-09:
Also beim ESP32 geht zu diesen Zeitpunkt der Wert von free_heap bei mir wieder hoch auf fast 190.000

MiniOh, 2023-02-09:
> MQTT hat aber auch nichts mehr zu berichten außer minütlich uptime, wifirssi und free_heap

Werden nicht weiterhin alle Daten der WR bis auf die Total Werte weiterhin übertragen?

MiniOh, 2023-02-09:
> Also beim ESP32 geht zu diesen Zeitpunkt der Wert von free_heap bei mir wieder hoch auf fast 190.000

Hier leider nicht.

Welche Ahoy Version bei dir?

beegee3, 2023-02-09:
kommt auf das MQTT Setting an: bei fixem Interval ja, sonst nein

beegee3, 2023-02-09:
hab nochmal nachgeschaut: in v0.5.78 wurde bei `pubMqtt` in `sendDiscoveryConfig` die Erzeugung von             `DynamicJsonDocument` geändert. @knickohr ist noch auf v0.5.76 (s.o.)
Wenn ich https://arduinojson.org/v6/how-to/reuse-a-json-document/ richtig lese, sollte nach Benutzung eines `DynamicJsonDocument` erst mit `clear()` der Speicher wieder freigegeben werden. Das fehlt bei `doc2`.

@lumapu Vorschlag zum testen:
in der inneren Schleife `for (uint8_t i = 0; i < ((!total) ? (rec->length) : (4) ) ; i++) {` am Ende `doc2.clear();` einfügen und bei der äußeren Schleife `for (uint8_t id = 0; id < mSys->getNumInverters() ; id++) {` würde ich `doc.clear();` nicht an den Anfang, sondern ans Ende stellen.
Dann mal sehen, was der free_heap anzeigt.

lumapu, 2023-02-10:
@beegee3 ja einen Versuch ist es wert.

```
JsonDocument::clear(), but this function is implicitly called by:

    deserializeJson()
    deserializeMsgPack()
    JsonDocument::to<T>()
    JsonDocument::set()
    JsonDocument::operator=
```

demnach erwarte ich nicht zu viel Verbesserung.
Das dynamisch bedeutet ja, dass er automatisch bei zu wenig Buffer size einen größeren Speicherbereich allokiert und den ursprünglichen freigibt. mE. sollten wir uns eher überlegen hier auf statisch zu gehen um nicht unnötig den Speicher zu Fragmenttieren.

Macht es Sinn, vor allem auf die API geschaut sich eine kleine Klasse zu bauen, die sehr simple JSON strings zusammensetzt? 
Als Basis einen `char buf[1024]` und dort wird mittels `addedLen = snprintf(&buf[pos], Len, "Blabla")`  der String zusammengefügt

MiniOh, 2023-02-10:
> > Werden nicht weiterhin alle Daten der WR bis auf die Total Werte weiterhin übertragen?
> 
> kommt auf das MQTT Setting an: bei fixem Interval ja, sonst nein

OK, verstehe.
Ich habe es bei mir auf einem fixen Intervall stehen, 60s. Also findet auch Nachts die MQTT Kommunikation statt, free_heap wird dabei nicht geringer.

Jetzt heute Morgen sieht man nochmal sofort bei aktiv werden der WR, dass die free_heap wieder beginnt kleiner zu werden.

MiniOh, 2023-02-10:
Hallo nochmal, 

Ich habe testweise 2 DTUs im Einsatz. Beide Identisch konfiguriert. Beide 0.5.78.

#1 verhält sich wie bereits hier diskutriert. free_heap nimmt nach und nach ab, daraus ergeben sich die Uptimes von ca. 2.5 Tagen welche fehlerfrei laufen.

#2 macht ca. alle 16 - 22min einen Reboot. Ich kann bisher nicht erkennen, warum das so ist.
Hier die uptime von #2

[image]

lumapu, 2023-02-10:
ich habe 5 Wemos hier (vom Makershop). Jeder verhält sich anders, von identisch würde ich hier nicht sprechen.
Der eine geht ohne NRF auch nur mit zusätzlichem Elko.

Die Hardware scheint eine nicht unwesentliche Rolle zu spielen.

beegee3, 2023-02-10:
@lumapu sorry war im falschen Film. `sendDiscoveryConfig` wird ja eigentlich nur einmal in den Settings benötigt, d.h. im normalen täglichen Gebrauch startet das nie. Und `DynamicJsonDocument` ist damit kein Problem (Umstieg auf statisch lohnt sich nicht; der Vorschlag ist nur 2 Zeilen Anpassung und garantiert Speicherfreigabe, bevor man das DJD wegschmeißt).
Eigentlich wollte ich mir `sendIvData` ansehen und war dann falsch abgebogen ↩️
Hab' da mal nun am Anfang und am Ende `DBGPRINTLN(F("****** ") + String(ESP.getFreeHeap()));` eingefügt: seit 5 Stunden kaum Schwankungen, schon gar kein abfallend. Kannst du das vielleicht mit einem ESP8266 kontrollieren?

lumapu, 2023-02-10:
ich hab noch keine Idee woher die Schwankungen kommen, ich hatte von gestern Abend bis heute Mittag (da habe ich dann einen reboot wegen Test gemacht) praktisch keine Schwankung.
Minimal waren es 14368 und maximal 14400 eigentlich identische Werte. 
Irgendwas scheint je nach Startzeitpunkt den ESP massiv zu beschäftigen

Gerri1, 2023-02-10:
Ich glaube, das Ganze wird noch durch die verwendung eines Displays verstärkt!

knickohr, 2023-02-10:
Wenn @Gerri1 Rect hat, dann würde ich dafür plädieren, das Display beim 8266 weg zu rationalisieren und nur noch beim ESP32 supporten.

knickohr, 2023-02-10:
Nein, es beseitigt nicht das Problem, aber es wird momentan an einer weiteren Implementierung von WRs gearbeitet. Nicht deicht halt das Gefühl das der 8266 irgendwann am Ende ist.

beegee3, 2023-02-10:
hab in `class MonochromeDisplay` geschaut. Das einzige, was mir aufgefallen ist: `uint8_t _mExtra` wird in `setup` nicht initialisiert und ansonsten nur hochgezählt, was letztlich zu einem Überlauf führt. Keine Ahnung, ob das zu einem ESP Problem wird, oder ob dann einfach mit 0 weitergemacht wird. Aber um den Hund nicht zu reizen, sollte das vielleicht korrigiert werden.
Allerdings müsste der Überlauf bereits nach 255 Sekunden auftreten, also sehr viel früher als die hier angegebenen Laufzeiten. Und kann das eigentlich zu einem memory leak führen?

MiniOh, 2023-02-10:
> @knickohr nur damit der ESP etwas länger durchhält? Die Verwendung eines Displays kostet natürlich Speicher, klar das dann der ESP eher am Ende ist. Aber das beseitigt doch nicht das eigentliche Problem (es sei denn, es ist ein Display Problem und wir suchen die ganze Zeit an der falschen Stelle 😆)

Also ich bin ohne Display unterwegs, von daher sollte, das zumindest nicht die Ursache sein, warum der Speicher soll geschrieben wird.
@lumapu hast du bzgl. der heap Problematik in 0.5.84 schon etwas geamcht?

lumapu, 2023-02-10:
@beegee3 der Überlauf ist unkritisch und bei oft sogar 'design-pattern' 😉
Ich komme aus der FPGA Welt und die lebt von Überläufen 

@MiniOh nein, ich stochere genauso wie @beegee3 durch den Code und fixe was mir ins Auge sticht. Bis jetzt war nichts weltbewegendes dabei.

bzgl Display: Ich meine, dass der komplette Display Inhalt kurzzeitig in ESP Speicher ist. Kann durchaus zum Absturz beitragen.
Ein Bekannter von mir hat auch einen ESP8266 mit SH1106 Display und 3 inverter. Er hat noch nie über reboots geklagt

Argafal, 2023-02-10:
Ich hab ein Stabilitätsproblem, aber kein Display. Mit 0.5.80 geht es etwas besser, aber 0.5.83 ist leider eher nicht verwendbar für mich, sorry to say. :(

MiniOh, 2023-02-11:
> @MiniOh nein, ich stochere genauso wie @beegee3 durch den Code und fixe was mir ins Auge sticht. Bis jetzt war nichts weltbewegendes dabei.

@lumapu 
OK, verstehe. Bis 0.5.76 scheint das Verhalten ja nicht da gewesen zu sein, oder? Lässt sich da evtl. ansetzen, was dort verändert wurde?

knickohr, 2023-02-11:
85 !  Neues Spiel, neues Glück 🤭

Argafal, 2023-02-11:
> OK, verstehe. Bis 0.5.76 scheint das Verhalten ja nicht da gewesen zu sein, oder? Lässt sich da evtl. ansetzen, was dort verändert wurde?

Gute Idee. Ich bin jetzt zurück auf 0.5.76, werde das mal ne halbe Stunde laufen lassen und sehen, wie es sich verhält. Dann gehe ich auf 0.5.77, dann auf 0.5.78, etc. Mal sehen, was es zu berichten gibt.

Aber: Ich bin auf ESP8266, also die Frage: will jemand das gleiche mit ESP32 machen und den free_heap und heap_frag und das allgemeine Verhalten beobachten? Vielleicht können wir ja eingrenzen, woher das Problem kommen kann.

MiniOh, 2023-02-11:
Ja, sicher, ich kann gerne gleiches mit dem ESP32 machen.

beegee3, 2023-02-11:
denke die ganze Zeit über die programmiertechnischen Unterschiede ESP8266 und ESP32 nach.
Neue Nebelkerze bzgl. memory leak: könnte es ein F() macro Problem sein? (bin als ESP32 User da nicht so im Thema)
Zumindest in `hmRadio` `getReceived` gibt es die Zeile `DBGPRINTLN("RX whatHappened Ch" + ...`. Die hatte ich eigentlich nur für internes debuggen gedacht und sollte vor dem PR gelöscht oder zumindest auskommentiert werden (ist mir durchgegangen 😞). Schließlich wird dieser Teil permanent bei der Inverter Kommunikation aufgerufen.

lumapu, 2023-02-11:
Das ist der Flashstringhelper, auf der einen Seite ein guter Freund auf der anderen Seite (während dem entwickeln) der Teufel 😈.
Er soll helfen den Speicherverbrauch (der heap wird durch ihn nicht so stark belastet) zur Laufzeit in Zaum zu halten, ohne ihn ist ein ESP8266 heute nicht mehr lauffähig.

lumapu, 2023-02-11:
ich habe mal meine Datenbank ausgequetscht ... diese hatte Daten für die letzen 30 Tage:
In blau der free_heap und in rot die version-patch. Die Version wurde anhand der Git Commits manuell nachgetragen, daher kann es auch teilweise fehlerhaft sein.
Anbei auch nochmal meine Rohdaten die als Basis für den Graph gedient haben.

[image]
[heap_lastMonth.csv](https://github.com/lumapu/ahoy/files/10714820/heap_lastMonth.csv)

Gerri1, 2023-02-11:
dev 0.5.85 heap_free von der Systemseite und MQTT ioBroker!

[image]
[image]

[21 more replies]

closed 2023-03-27 by lumapu