lumapu/ahoy #650: 0.5.78 MQTT disconnects without error message

Argafal, 2023-02-05:
Running 0.5.78 on ESP8266. MQTT connects fine upon boot. At some point the MQTT connection drops and the webpage shows: "MQTT is not connected". 

- I do not see any error messages related to MQTT on the serial output.
- In the mosquitto server log I find the message "1675600063: Client AHOY-DTU has exceeded timeout, disconnecting."

How can I help to further understand this problem, and is there a way to enable more debugging messages on the serial output?

stefan123t, 2023-02-05:
Yes, select it from the settings > [x] serial output menu option with the serial usb logs and/or use a development build with additional debug output based on the flags defined in the platformio.ini.
Note that debug output logs a lot of core library information on serial output too, hence it may be subject to reboots much more frequently than the normal build images.

Argafal, 2023-02-05:
Thanks. Serial output is already enabled in the settings and I did not get message related to the mqtt issue.

 Good point about adding more debug flags when building it, thanks for pointing that possibility out to me. Will do.

Argafal, 2023-02-11:
I know caught an error message. I went back all the way to 0.5.66, this problem already appears there. 
```
13:19:50.207 > I: MQTT disconnected, reason: TCP disconnect
```

On the server side I see
```
1676117932: New client connected from <IPADDRESS>:53787 as AHOY-DTU (p2, c1, k15).
1676117965: Client AHOY-DTU closed its connection.
```

Ahoy's wifi RSSI is -59, the server is connected by cable. Eventually MQTT reconnects. Does anyone have any suggestion how to understand this better?

lumapu, 2023-02-11:
I had the same behavior yesterday with my development Wemos. For testing I removed the NTP address to adjust manually the time.
I don't know if there is any coomon ground.

Für me it looked like a pattern which repeats all the time. I had approx. two disconnects per minute.

lumapu, 2023-02-11:
@Argafal here is a screenshot of my behavior - on my debug ESP without NRF
[image]

lumapu, 2023-02-11:
@beegee3 das schein ein generelles Problem zu sein. Hast du eine Idee was wir falsch machen. Auch mein Produktivsystem zeigt diesen Fehler.

<details>
<summary>Log</summary>

```
22:28:07 I: MQTT disconnected, reason: TCP disconnect
22:28:08 I: MQTT connected
22:28:08 I: MQTT got topic: inverter/ctrl/limit_persistent_relative
22:28:08 I: MQTT got topic: inverter/ctrl/limit_persistent_absolute
22:28:08 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_relative
22:28:08 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_absolute
22:28:08 I: MQTT got topic: inverter/setup/set_time
22:28:08 I: [NTP]: 2023-02-11 21:28:08 UTC
22:28:08 I: MQTT got topic: inverter/setup/sync_ntp
22:28:08 I: [NTP]: 2023-02-11 21:28:08 UTC
22:28:08 I: resetPayload: id: 0
22:28:08 I: (#0) Requesting Inv SN 116172214159
22:28:08 I: (#0) Devcontrol request 0xb power limit 0
22:28:08 I: sendControlPacket cmd: 0xb
22:28:08 I: TX 19B Ch61 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:08 I: retransmit power limit
22:28:08 I: sendControlPacket cmd: 0xb
22:28:08 I: TX 19B Ch75 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:09 I: retransmit power limit
22:28:09 I: sendControlPacket cmd: 0xb
22:28:09 I: TX 19B Ch3 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:10 I: retransmit power limit
22:28:10 I: sendControlPacket cmd: 0xb
22:28:10 I: TX 19B Ch23 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:12 I: retransmit power limit
22:28:12 I: sendControlPacket cmd: 0xb
22:28:12 I: TX 19B Ch40 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:13 W: Time not set or it is night time, therefore no communication to the inverter!
22:28:13 I: retransmit power limit
22:28:13 I: sendControlPacket cmd: 0xb
22:28:13 I: TX 19B Ch61 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:43 W: Time not set or it is night time, therefore no communication to the inverter!
22:28:53 I: MQTT disconnected, reason: TCP disconnect
22:28:54 I: MQTT connected
22:28:54 I: MQTT got topic: inverter/ctrl/limit_persistent_relative
22:28:54 I: MQTT got topic: inverter/ctrl/limit_persistent_absolute
22:28:54 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_relative
22:28:54 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_absolute
22:28:54 I: MQTT got topic: inverter/setup/set_time
22:28:54 I: [NTP]: 2023-02-11 21:28:54 UTC
22:28:54 I: MQTT got topic: inverter/setup/sync_ntp
22:28:54 I: [NTP]: 2023-02-11 21:28:54 UTC
22:28:54 I: resetPayload: id: 0
22:28:54 I: (#0) Requesting Inv SN 116172214159
22:28:54 I: (#0) Devcontrol request 0xb power limit 0
22:28:54 I: sendControlPacket cmd: 0xb
22:28:54 I: TX 19B Ch75 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:54 I: retransmit power limit
22:28:54 I: sendControlPacket cmd: 0xb
22:28:54 I: TX 19B Ch3 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:55 I: retransmit power limit
22:28:55 I: sendControlPacket cmd: 0xb
22:28:55 I: TX 19B Ch23 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:56 I: retransmit power limit
22:28:56 I: sendControlPacket cmd: 0xb
22:28:56 I: TX 19B Ch40 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:58 I: retransmit power limit
22:28:58 I: sendControlPacket cmd: 0xb
22:28:58 I: TX 19B Ch61 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:28:59 I: retransmit power limit
22:28:59 I: sendControlPacket cmd: 0xb
22:28:59 I: TX 19B Ch75 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
22:29:13 W: Time not set or it is night time, therefore no communication to the inverter!
22:29:39 I: MQTT disconnected, reason: TCP disconnect
22:29:40 I: MQTT connected
22:29:40 I: MQTT got topic: inverter/ctrl/limit_persistent_relative
22:29:40 I: MQTT got topic: inverter/ctrl/limit_persistent_absolute
22:29:40 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_relative
22:29:40 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_absolute
22:29:40 I: MQTT got topic: inverter/setup/set_time
22:29:40 I: [NTP]: 2023-02-11 21:29:40 UTC
22:29:40 I: MQTT got topic: inverter/setup/sync_ntp
07:28:16 I: [NTP]: 2036-02-07 06:28:16 UTC
07:28:16 I: resetPayload: id: 0
07:28:16 I: (#0) Requesting Inv SN 116172214159
07:28:16 I: (#0) Devcontrol request 0xb power limit 0
07:28:16 I: sendControlPacket cmd: 0xb
07:28:16 I: TX 19B Ch3 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
07:28:16 I: retransmit power limit
07:28:16 I: sendControlPacket cmd: 0xb
07:28:16 I: TX 19B Ch23 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
07:28:17 I: tickMidnight 2086041601
07:28:17 I: zeroInverterValues
07:28:17 I: zeroYieldDay
07:28:17 I: retransmit power limit
07:28:17 I: sendControlPacket cmd: 0xb
07:28:17 I: TX 19B Ch40 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
07:28:18 I: retransmit power limit
07:28:18 I: sendControlPacket cmd: 0xb
07:28:18 I: TX 19B Ch61 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
07:28:20 W: Time not set or it is night time, therefore no communication to the inverter!
07:28:20 I: retransmit power limit
07:28:20 I: sendControlPacket cmd: 0xb
07:28:20 I: TX 19B Ch75 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
07:28:21 I: retransmit power limit
07:28:21 I: sendControlPacket cmd: 0xb
07:28:21 I: TX 19B Ch3 | 51 72 21 41 59 87 27 16 17 81 0b 00 00 00 00 00 60 01 50 
```
</details>

beegee3, 2023-02-12:
oje, das sieht aus, als würden je zwei Ticker für NTP und MQTT laufen. Das einzige, wo sie gestartet werden ist `app::onWifi` mit dem Parameter `true`. Und da werden zunächst alle Ticker mit `ah::Scheduler::resetTicker();` gelöscht. Eigentlich können so dieselben Ticker nicht mehrfach entstehen 🤔 ... 45 min später ... 💡 
hab's glaub ich gefunden:
ich weiß zwar nicht, was das **erste** MQTT disconnected ausgelöst hat, aber durch die anschließende erneute Verbindung wird in `pubMqtt` das `onMessage` mit allen **MQTT got topic: ...** ausgelöst.
Entscheidend sind hier
**MQTT got topic: inverter/setup/set_time** und **MQTT got topic: inverter/setup/sync_ntp**
Beide lösen über `RestApi` ein `mApp->setTimestamp` aus.
Im ersten Fall mit Parameter `jsonIn[F("val")]`, was eigentlich unkritisch sein sollte, aber wenn der Wert 0 ist, führt das genauso wie der zweite Fall, wo explizit der Parameter 0 ist, zum Aufruf von `mWifi.getNtpTime();`, was die darauffolgenden `[NTP]:...` erklärt.
Also keine mehrfach laufenden Ticker 😅, sondern "nur" einfache NTP Aufrufe, über deren Sinn man natürlich nachdenken kann, die aber eigentlich unschädlich sind. Abgesehen, davon, dass @lumapu uns nun weit voraus im Jahr 2036 lebt 😄, was auch die Aktionen 'tickMidnight 2086041601' ... 'retransmit power limit' erklärt.
Bleibt die Frage, warum es zu **MQTT disconnected** kommt. Ohne sehr in der MQTT Materie zu sein, ist es möglich, dass der MQTT Server die Verbindung beendet, da er eine Meldung erwartet, aber nichts bekommt?
Es ist Dunkel, die Inverter senden nicht, also wird nur 1 x pro Minute ein Status geschickt (wozu es nicht mehr kommt, da die Verbindung vorher abreißt).
Soweit die Erklärung des logs, bei der MQTT Fehlerkorrektur kann ich nur wenig helfen (wie gesagt, bin da nicht wirklich drin)
@lumapu Aber wenn du da eh etwas korrigierst, schmeiß' doch bitte `mReconnectRequest` ganz raus, war ja mal anders gedacht, wird heute alles über die Ticker geregelt: bei `onDisconnect` braucht man nichts, das macht `tickerSecond` sowieso. Und da gehört die Verbindungsprüfung an den Anfang, nicht ans Ende, denn `sendIvData();` macht doch nur bei funktionierender Verbindung Sinn, sonst gibt's einen publish Stau, schlecht für den heap (womit wir bei den anderen Diskussionen sind, die gerade laufen). Also, keine Verbindung, dann aufbauen und erst, wenn die Verbindung steht, die IV Daten senden. Damit `mIntervalTimeout` stimmt, gehört das nach vorne:
```
        void tickerSecond() {
            if (mIntervalTimeout > 0)
                mIntervalTimeout--;

            if(!mClient.connected()) {
                mClient.connect();
                return;
            }

            if(0 == mCfgMqtt->interval) // no fixed interval, publish once new data were received (from inverter)
                sendIvData();
            else { // send mqtt data in a fixed interval
                if(mIntervalTimeout == 0) {
                    mIntervalTimeout = mCfgMqtt->interval;
                    mSendList.push(RealTimeRunData_Debug);
                    sendIvData();
                }
            }
        }
```
Die 'Connect' Geschichte in `payloadEventListener` scheint mir so auch nicht so zu sein, wie sie soll. Bitte denk' darüber nochmal nach und wir können das gerne diskutieren.

lumapu, 2023-02-13:
@beegee3 danke für deine Vorschläge. Habe es so wie vorgeschlagen übernommen. Bzgl. `payloadEventListener` denke ich was du meinst. Es werden Daten verworfen, wenn keine Verbinung besteht. Wir können aber nicht endlos Daten sammeln und dann auf einen Schlag sehr viele `publishen`.
Daher denke ich ist die aktuelle Implementierung gar nicht so übel.
Durch die `std::queue` können wir auch nicht durch die Liste iterieren, ob das neue Command schon vorhanden ist. Wie sind deine Ideen?

beegee3, 2023-02-13:
@lumapu 😆, ich hatte `payloadEventListener` genau umgekehrt interpretiert und dachte, du wolltest unbedingt publishen.
Wenn das nicht so ist, lösch' doch bitte die erste Zeile `connect();` Die Verbindung wird doch über `tickerSecond` im Sekundentakt geprüft.
Durch die `std::queue` iterieren war mir einfach zu kompliziert. Da hätte ich ja nachdenken müssen 😁. `RTRDataHasBeenSent` ging mir leichter von der Hand. Iterieren ist natürlich eleganter. Ändere es nach belieben.

Das von @Argafal und dir beobachtete endlose disconnect - connect scheint ein Problem der MQTT Api zu sein. Zumindest weisen ein paar Issues in https://github.com/marvinroger/async-mqtt-client darauf hin, die dasselbe beschreiben. Dort empfohlene Abhilfe ist ein WiFi.disconnect() zu erzwingen. Ist natürlich brutal (Wifi ist ok, nur Mqtt nicht). Was denkst du?

lumapu, 2023-02-14:
stimmt das `connect()` ist überflüssig.
Ich habe jetzt rausgefunden warum ich in der fernen Zukunft lebe ... Da ich ioBroker verwende hier die richtige Einstellung, dass man wieder zurück ins Jahr 2023 kommt:

[image]

Übrigens: wenn ich den MQTT Broker auf meinen `mosquitto` stelle habe ich keine Reconnects, das passiert nur mit dem MQTT Adapter von ioBroker. Beide sind im gleichen Netzwerk.

beegee3, 2023-02-14:
@lumapu `mMqtt.connect();` in `app::tickNtpUpdate` braucht man auch nicht. Direkt danach wird `PubMqttType::tickerSecond` installiert und das sorgt für das connect. `inline void connect()` in `pubMqtt` kann dann auch gelöscht werden!

Zwei Dinge sind in diesem Zusammenhang ggfs. noch problematisch:
1. wenn das MQTT nicht verbunden ist, startet `tickerSecond` einen Verbindungsversuch. Dauert das Connect länger als 1 Sekunde (z.B. schlechtes WLAN), so werden mehrere Versuche gestartet. Kann das Probleme machen? Ist es besser, die Zeitspanne zwischen zwei Versuchen zu vergrößern?
2. 'States bei subscribe publizieren' kann evtl. auch bei anderen Brokern vorkommen. Von daher wäre es sinnvoll, die `onMessage` Aufrufe während der Subscription Phase zu sperren. Ist dafür das `sessionPreset` in `onConnect` gedacht?
Ansonsten, da man nicht weiß, wie lange es vom Subscribe bis zum `onMessage` dauert (und ob das überhaupt passiert), könnte man für die Sperre vielleicht die Zeit seit dem Connect nutzen.

Beide Punkte lassen sich gemeinsam behandeln, indem man ein `uint32_t mLastConnTs;` definiert und mit `millis()` vergleicht. Beim Connect Versuch **und** bei erfolgreichem Connect wird `mLastConnTs = millis();` gesetzt.
Vorschlag für `pubMqtt` mit 5 Sekunden zwischen zwei Connect Versuchen und auch 5 Sekunden Sperrzeit:
in `void setup(...) `: `mLastConnTs = 0;` und
```
        void tickerSecond() {
            if (mIntervalTimeout > 0)
                mIntervalTimeout--;

            if(!mClient.connected()) {
                if (millis() - mLastConnTs > 5000) {
                    mClient.connect();
                    mLastConnTs = millis(); // timestamp: connection attempt
                }
                return; // next check in a second
            }
            ...
        }
...
        void onConnect(bool sessionPreset) {
            DPRINTLN(DBG_INFO, F("MQTT connected"));
            mLastConnTs = millis();// timestamp: connected
            ...
        }
...
        void onMessage(...) {
            DPRINTLN(DBG_INFO, F("MQTT got topic: ") + String(topic));
            if (millis() - mLastConnTs < 5000) // avoid to publish states on subscribe
                return;
            if(NULL == mSubscriptionCb)
                return;
            ...
        }
```
**Achtung:** während der Sperrzeit werden natürlich alle Einstellungsversuche durch den User (bzw. automatisierte Skripte) ignoriert. Wirft das neue Probleme auf? Werden z.B. Limits direkt nach Connect gesendet?

Argafal, 2023-02-17:
Ich wollte kurz Rückmeldung geben. Mit 0.5.89 kann ich das Problem nicht reproduzieren. Vielleicht kommt's noch, oder es ist jetzt als Nebeneffekt durch die vielen anderen Änderungen gelöst.

That being said, die Analyse und Vorschläge von @beegee3 finde ich echt wertvoll. Vielleicht sollte der Issue dafür offen bleiben, auch wenn das eigentlich MQTT disconnect jetzt behoben scheint.

closed 2023-03-09 by lumapu