lumapu/ahoy #651: Kein Datenempfang am USB Netzteil

der2of6, 2023-02-05:
## Hardware
  * [x] ESP8266
Modelname: Several D1 mini and Pro

### nRF24L01+ Module
* [x] ** nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [ ] circuit board
* [x] external antenna

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

## Software
* [x] AhoyDTU
* [ ] 
### Version / Git SHA: 
Version: 0.5.78 and 66

### Build & Flash Method:
* [x ] ESP Tools  
### Desktop OS:
* [x ] Windows
---------------------
Hallo,
ich habe ein sehr merkwürdiges Problem mit meiner mittlerweilte 2. DTU.

Ausgangsituation
Der ESP und der RF24L01+ sind verkablet und geflashed, hängen aber noch per serial am PC.
Ich mache alle Einstellungen (WLAN, Wechselrichter SN etc.), und starte den ESP neu -> Die daten des Wechselrichters werden sofort angezeigt.

Sobald ich jetzt den ESP an ein normales USB Netzteil anstecke, bekomme ich keine Daten des WR mehr. Ein Elko auf 5V VIN bzw. den 3,3V des RF Modules bringen keine Verbesserung. 

Hat hier noch jemand eine Idee, was das sein könnte?

mroenne2022, 2023-02-05:
Aber im WLAN ist der ESP und du erreichst ihn per IP im Browser wenn er am Netzteil hängt ? 
Eigentlich kann es dann ja fast nur am Netzteil liegen , zu wenig Power vom Netzteil ? 
Oder die Entfernung zum WR ? Es klingt tatsächlich merkwürdig. Ich hab schon ein paar gebaut , sowas hatte ich noch nicht.

Loetnase, 2023-02-05:
Versuche mal ein anderes Netzteil oder Akku.
Ich hatte schon mal ein Netzteil das bei zu geringer Last sich ausgeschalten hat, dann aber sofort wieder anlief.
Der ESP mit Elko sieht dann zwar dauernd Spannung welche aber dauernd rauf und runter läuft.

der2of6, 2023-02-05:
Werde ich morgen (wenn kein schnee kommt) nochmal mit einem anderen NT testen. Dieses läuft auf meinem anderen ESP mit AHOY ohne probleme druch. 
Via WLAN ist der ESP immer erreichbar.

der2of6, 2023-02-06:
Auch nach tauschen der Netzteile bleibt das Ergebnis gleich. 

Der Unterschied zwischen den beiden DTU ist, der funktionierende hat einen älteren D1 mini (clon) mit dem ESP12 und der "Metallplatte" an der WLAN Antenne.

Die nur am PC funktionierenden sind neuere Versionen.

Ich habe jetzt noch zusätzlich nochmal einen alten D1 mini aus meinem Funduns gelötet und eingesetzt.
Dieser funktioniert sofort, auch mit unterschiedlichen Netzteilen. 

Sieht also nach einem sehr merkwüridgen HW Thema aus. 

Da ich für den Anwendungsfall einen ESP mit externen WLAN Antenne benötige, hat jemand hier einen Link zu einem funktionierenden PRO mit externen Antennenanschluss?

mroenne2022, 2023-02-06:
@der2of6 : Leider nein, hab so einen ESP mit ext WLAN noch nicht verbaut.

E1nste1n-AUT, 2023-02-06:
Hatte ein ähnliches Problem. 2x WeMos D1 Pro und 2x NRF24L01+ Module vor einiger Zeit bestellt und aus dem ersten Set eine funktioniere ahoyDTU zusammengebastelt.

Vor 2 Wochen wollte ich testhalber die zweite ahoyDTU bauen, aber keine Chance...bekomme keine Daten vom Hoymiles, obwohl alles identisch war (selbes Netzteil, selber Code, erste ahoyDTU vom Netz genommen) Hab dann auch die NRF module untereinander getauscht und irgendwie hatte ich das Gefühl, der zweite WeMos D1 Pro ist "anders".

Also ich vermute leider hier bei mir auch ein Hardware issue, obwohl ich die Teile wie gesagt alle zusammen bestellt hab (wo man annehmen könnte, die sind identisch - wohl aber nicht in China)

Loetnase, 2023-02-06:
Sobald sich so ein ESP8266 (wemos) komlett komisch verhält, versuche ich den mit der Arduino IDE mit einem kleinen Beispielprogramm zu flashen. Z.B- Blink oder einen WLAN Scanner (ScanNetworks). Dabei in Werkzeuge, Erase Flash, All Flash Contents auswählen. Bei dir würde ich den WLAN Scanner (ScanNetworks) ausprobieren ob der dann vernünftig läuft.
Ich denke das löscht alles überflüssige im Speicher.
Dann den ESP mit der Ahoy DTU wie gewohnt komplett neu aufsetzen und hoffen das es dann geht.
Ich hatte dachte mehrmals der ESP ist kaputt uns so ging er dann wieder.
Ich habe so das Gefühl das der ESP Web Installer nicht immer den ganzen Speicher platt macht.

gent99, 2023-02-07:
Wie ist dein Funkmodul am Wemos angeschlossen? Ich hatte dasselbe Problem. Die Kabel zwischen dem NRF und dem Wemos müssen so kurz wie möglich sein. Auf nem Breadboard mit Jumper Wites haste auf Dauer kein Glück.

dtuuser, 2023-02-07:
Mein NRF ist mit 10cm Kabeln auf Stiftleisten "angeklemmt". Kondensator am NRF. Läuft aktuell immer ca. 7 Tage bis der Wachhund anschlägt.

der2of6, 2023-02-08:
Erst mal danke für die vielen Antworten. Leider hat auch ein komplettes überschreiben des Speichers und dem neu Flashen der DTU nicht geholfen.
Ich hab jetzt nochmal ein anderes Board mit ext. Antennenstecker im Zulauf. Mal schauen ob es damit dann läuft

tastendruecker123, 2023-02-09:
Ein Unterschied zwischen PCs und USB Netzteilen ist, dass die USB Anschlüsse GND-seitig bei PCs geerdet sind, während sie bei USB Netzteilen nur mit einem Kondensator ans Stromnetz gekoppelt sind.

Es wäre interessant, ob das Problem auch mit einer USB Powerbank auftritt, oder wenn man den ESP zusätzlich erdet, z.B. mit Krokodilklemmen zwischen dem Erdkontakt der Steckdose und der Aussenseite des Micro USB Anschlusses auf dem Wemos Board.

der2of6, 2023-02-17:
Kurzes Update:
Ich habe jetzt einen, laut Händler, Original D1 mini Pro bekommen. 
Von der HW völlig überdimensioniert, aber mit externen Antennenanschluss. 
Hier klappt es auf Anhieb ohne Probleme.

Fatzit: Ein Board mit 8266 ist nicht immer gleich einem Board mit 8266

lumapu, 2023-02-17:
> Fatzit: Ein Board mit 8266 ist nicht immer gleich einem Board mit 8266

das predige ich auch immer 😉👍 selbst Boards, die ich in einer Lieferung bekommen habe verhalten sich unterschiedlich

closed 2023-03-09 by lumapu