lumapu/ahoy #654: HomeAssistant errors / auto-discovery not working

andreas-bulling, 2023-02-05:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

don't know

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

None

### Error description

Hi,

I see the following error message in the HomeAssistant logs. Auto-discovery is not working - not sure if both are linked:

`Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Balcony Efficiency', 'state_topic': 'inverter-balcony/Balcony/ch0/Efficiency', 'unit_of_measurement': '%', 'unique_id': 'X_ch0_Efficiency', 'device': {'name': 'Balcony', 'identifiers': 'X', 'configuration_url': 'http://192.168.1.190', 'manufacturer': None}, 'expire_after': 65, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 334, in async_discover config: DiscoveryInfoType = discovery_schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: string value is None for dictionary value @ data['device']['manufacturer'] `

lubeda, 2023-02-07:
Ich denke es ist der selbe Fehler wie bei "meinem" Issue #632

andreas-bulling, 2023-02-07:
duplicate of #632

closed 2023-02-07 by andreas-bulling