lumapu/ahoy #655: Boot loop: ISR not in IRAM!

Argafal, 2023-02-05:
With a85ed6c0897ddc6883da28d8882820abb0333795 I get a boot loop. Commit 94fe61b54432fa4de85c2a538b851ff6d6a019e7 still works fine. 

Could this have been introduce in the commit "reworked `hmRadio.h` completely (interrupts, packaging)" by @beegee? Thanks for having a look :)

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00081770
~ld
I: Initializing FS ..
I:  .. done
W: failed to load json, using default config
I: Settings valid: false
ISR not in IRAM!

User exception (panic/abort/assert)

```

lumapu, 2023-02-05:
fixed check newest commit 😎

Argafal, 2023-02-05:
You were so fast! Nice :) I confirm that it works again with your most recent commit.

closed 2023-02-05 by Argafal

lumapu, 2023-02-06:
you too 😃