lumapu/ahoy #657: Debug target compiler error

Argafal, 2023-02-06:
Compiling the most recent development version (git edefcf1c830b5ea7cd202a9d27395811a6f5a03b) fails for the debug target.

pio run -e esp8266-debug

```
[...]
Compiling .pio/build/esp8266-debug/lib48b/espMqttClient/Transport/ClientSecureSync.cpp.o
In file included from .pio/libdeps/esp8266-debug/ESPAsyncTCP/src/async_config.h:26,
                 from .pio/libdeps/esp8266-debug/ESPAsyncTCP/src/ESPAsyncTCP.h:25,
                 from .pio/libdeps/esp8266-debug/espMqttClient/src/Transport/ClientAsync.h:17,
                 from .pio/libdeps/esp8266-debug/espMqttClient/src/Transport/ClientAsync.cpp:11:
.pio/libdeps/esp8266-debug/ESPAsyncTCP/src/DebugPrintMacros.h: In function '_DEBUG_TIME_STAMP debugTimeStamp()':
.pio/libdeps/esp8266-debug/ESPAsyncTCP/src/DebugPrintMacros.h:23:18: error: 'millis' was not declared in this scope
   23 |   unsigned now = millis() % 1000000000;
      |                  ^~~~~~
Compiling .pio/build/esp8266-debug/lib48b/espMqttClient/Transport/ClientSync.cpp.o
*** [.pio/build/esp8266-debug/lib48b/espMqttClient/Transport/ClientAsync.cpp.o] Error 1
In file included from app.h:21,
                 from app.cpp:6:
In member function 'uint8_t ah::Scheduler::addTicker(ah::scdCb, uint32_t, uint32_t, bool, const char*)',
    inlined from 'void ah::Scheduler::onceAt(ah::scdCb, uint32_t, const char*)' at utils/scheduler.h:69:83,
    inlined from 'void app::tickMidnight()' at app.cpp:228:11:
utils/scheduler.h:127:32: warning: 'char* strncpy(char*, const char*, size_t)' output truncated before terminating nul copying 4 bytes from a string of the same length [-Wstringop-truncation]
  127 |                         strncpy(mTicker[i].name, name, (strlen(name) < 6) ? strlen(name) : 5);
      |                         ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

Argafal, 2023-02-08:
I confirm that this still is an issue with the most recent development version (tested on git 7423de76fb0f42918987a38b4aabe3f5ce60cc04)

stefan123t, 2023-02-08:
Hast Du auch die aktuelle ESP8266 bzw. ESP32 Board Bibliotheken in PlatformIO.
Ich hatte auch schon seltsame Meldungen zu Dingen die eigentlich da sein müssten.
Manchmal hilft es die beiden Boards zu aktualisieren auf die letzte Version.

Argafal, 2023-02-09:
```
$ pio lib update

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg update` instead.

Resolving esp8266-release dependencies...
Already up-to-date.
Resolving esp8266-debug dependencies...
Already up-to-date.
Resolving esp8285-release dependencies...
Already up-to-date.
Resolving esp8285-debug dependencies...
Already up-to-date.
Resolving esp32-wroom32-release dependencies...
Already up-to-date.
Resolving esp32-wroom32-debug dependencies...
Already up-to-date.
```
```
$ pio lib list

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg list` instead.

Resolving esp8266-release dependencies...
Libraries
├── ArduinoJson @ 6.20.1 (required: bblanchon/ArduinoJson)
├── ESP Async WebServer @ 1.2.3+sha.1dde9cf (required: git+https://github.com/yubox-node-org/ESPAsyncWebServer)
│   ├── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
│   └── ESPAsyncTCP @ 1.2.2 (required: me-no-dev/ESPAsyncTCP @ ^1.2.2)
├── RF24 @ 1.4.5 (required: nrf24/RF24)
├── Time @ 1.6.1 (required: paulstoffregen/Time)
├── Timezone @ 1.2.4+sha.7cf1425 (required: git+https://github.com/JChristensen/Timezone)
├── U8g2 @ 2.34.13 (required: olikraus/U8g2)
└── espMqttClient @ 1.3.3+sha.6a3f3a8 (required: git+https://github.com/bertmelis/espMqttClient#v1.3.3)

Resolving esp8266-debug dependencies...
Libraries
├── ArduinoJson @ 6.20.1 (required: bblanchon/ArduinoJson)
├── ESP Async WebServer @ 1.2.3+sha.1dde9cf (required: git+https://github.com/yubox-node-org/ESPAsyncWebServer)
│   ├── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
│   └── ESPAsyncTCP @ 1.2.2 (required: me-no-dev/ESPAsyncTCP @ ^1.2.2)
├── RF24 @ 1.4.5 (required: nrf24/RF24)
├── Time @ 1.6.1 (required: paulstoffregen/Time)
├── Timezone @ 1.2.4+sha.7cf1425 (required: git+https://github.com/JChristensen/Timezone)
├── U8g2 @ 2.34.13 (required: olikraus/U8g2)
└── espMqttClient @ 1.3.3+sha.6a3f3a8 (required: git+https://github.com/bertmelis/espMqttClient#v1.3.3)

Resolving esp8285-release dependencies...
Libraries
├── ArduinoJson @ 6.20.1 (required: bblanchon/ArduinoJson)
├── ESP Async WebServer @ 1.2.3+sha.1dde9cf (required: git+https://github.com/yubox-node-org/ESPAsyncWebServer)
│   ├── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
│   └── ESPAsyncTCP @ 1.2.2 (required: me-no-dev/ESPAsyncTCP @ ^1.2.2)
├── RF24 @ 1.4.5 (required: nrf24/RF24)
├── Time @ 1.6.1 (required: paulstoffregen/Time)
├── Timezone @ 1.2.4+sha.7cf1425 (required: git+https://github.com/JChristensen/Timezone)
├── U8g2 @ 2.34.13 (required: olikraus/U8g2)
└── espMqttClient @ 1.3.3+sha.6a3f3a8 (required: git+https://github.com/bertmelis/espMqttClient#v1.3.3)

Resolving esp8285-debug dependencies...
Libraries
├── ArduinoJson @ 6.20.1 (required: bblanchon/ArduinoJson)
├── ESP Async WebServer @ 1.2.3+sha.1dde9cf (required: git+https://github.com/yubox-node-org/ESPAsyncWebServer)
│   ├── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
│   └── ESPAsyncTCP @ 1.2.2 (required: me-no-dev/ESPAsyncTCP @ ^1.2.2)
├── RF24 @ 1.4.5 (required: nrf24/RF24)
├── Time @ 1.6.1 (required: paulstoffregen/Time)
├── Timezone @ 1.2.4+sha.7cf1425 (required: git+https://github.com/JChristensen/Timezone)
├── U8g2 @ 2.34.13 (required: olikraus/U8g2)
└── espMqttClient @ 1.3.3+sha.6a3f3a8 (required: git+https://github.com/bertmelis/espMqttClient#v1.3.3)

Resolving esp32-wroom32-release dependencies...
Libraries
├── ArduinoJson @ 6.20.1 (required: bblanchon/ArduinoJson)
├── ESP Async WebServer @ 1.2.3+sha.1dde9cf (required: git+https://github.com/yubox-node-org/ESPAsyncWebServer)
│   └── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
├── RF24 @ 1.4.5 (required: nrf24/RF24)
├── Time @ 1.6.1 (required: paulstoffregen/Time)
├── Timezone @ 1.2.4+sha.7cf1425 (required: git+https://github.com/JChristensen/Timezone)
├── U8g2 @ 2.34.13 (required: olikraus/U8g2)
└── espMqttClient @ 1.3.3+sha.6a3f3a8 (required: git+https://github.com/bertmelis/espMqttClient#v1.3.3)

Resolving esp32-wroom32-debug dependencies...
Libraries
├── ArduinoJson @ 6.20.1 (required: bblanchon/ArduinoJson)
├── ESP Async WebServer @ 1.2.3+sha.1dde9cf (required: git+https://github.com/yubox-node-org/ESPAsyncWebServer)
│   └── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
├── RF24 @ 1.4.5 (required: nrf24/RF24)
├── Time @ 1.6.1 (required: paulstoffregen/Time)
├── Timezone @ 1.2.4+sha.7cf1425 (required: git+https://github.com/JChristensen/Timezone)
├── U8g2 @ 2.34.13 (required: olikraus/U8g2)
└── espMqttClient @ 1.3.3+sha.6a3f3a8 (required: git+https://github.com/bertmelis/espMqttClient#v1.3.3)

```

Argafal, 2023-02-09:
I managed to build the debug target. I will document the changes necessary below:

1) Add `include <Arduino.h>` to the beginning of file `.pio/libdeps/esp8266-debug/ESPAsyncTCP/src/DebugPrintMacros.h`. This resolves the error `'millis' was not declared in this scope`.

2) In platformio.ini, add `-DPIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48` to the build flags of the `esp8266-debug` target. Without this flag I run into error ```section `.text1'  will not fit in region `iram1_0_seg'```. With this flag it complies fine.

Argafal, 2023-02-09:
Boot-loop with the debugger build:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000a7e90
~ld

SDK:2.2.2-dev(38a443e)/Core:3.1.1=30101000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
I: Initializing FS ..

I:  .. done
lfs_file_close: fd=0x3fff6b68
                             I: Settings valid: true
I: RF24 Amp Pwr: RF24_PA_I: HIGH
I: Radio Config:
SPI Frequency           = 1 Mhz
                               Channel                  = 3 (~ 2403 MHz)
Model                   = nRF24L01+
RF Data Rate            = 250 KBPS
RF Power Amplifier      = PA_HIGH
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 1000 microseconds
Auto Retry Attempts     = 15 maximum
Packets lost on
                   current channel      = 0
Retry attempts made for
                           last transmission    = 15
Multicast               = Disabled
Custom ACK Payload      = Disabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = Enabled
Primary Mode            = TX
TX address              = 0x3035407401
pipe 0 ( open ) bound   = 0x3035407401
pipe 1 ( open ) bound   = 0x7551998601
pipe 2 (closed) bound   = 0xc3
pipe 3 (closed) bound   = 0xc4
pipe 4 (closed) bound   = 0xc5
pipe 5 (closed) bound   = 0xc6

---------
         AhoyDTU Info:
Version: 0.5.83
Github Hash: 7423de7

---------
         AP MODE
                SSID: AHOY-DTU
PWD: esp_8266
IP Address: http://192.168.4.1
---------

fpm close 1 
            mode : sta(c8:c9:a3:6a:bc:e7) + softAP(ca:c9:a3:6a:bc:e7)
                                                                     add if0
                                                                            add if1
                                                                                   dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
                                                                                                                                                       bcn 100
                                                                                                                                                              [APConfig] local_ip: 192.1
68.4.1 gateway: 192.168.4.1 subnet: 255.255.255.0
                                                 [APConfig] DHCP IP start: 192.168.4.100
                                                                                        [APConfig] DHCP IP end: 192.168.4.200
                                                                                                                             bcn 0
                                                                                                                                  del if1
                                                                                                                                         add if1
                                                                                                                                                dhcp server start:(ip:192.168.4.1,mask:2
55.255.255.0,gw:192.168.4.1)
                            bcn 100
                                   connect to network 'Cinderella' ...
I: added inverter HM114172221779
I: added inverter HM116174403530
I: added inverter HM116174404254
I: resetPayload: id: 0
I: resetPayload: id: 1
I: resetPayload: id: 2
I: resetPayload: id: 3
I: resetPayload: id: 4
I: resetPayload: id: 5
I: resetPayload: id: 6
I: resetPayload: id: 7
I: resetPayload: id: 8
I: resetPayload: id: 9
I: resetPayload: id: 0
I: resetPayload: id: 1
I: resetPayload: id: 2
I: resetPayload: id: 3
I: resetPayload: id: 4
I: resetPayload: id: 5
I: resetPayload: id: 6
I: resetPayload: id: 7
I: resetPayload: id: 8
I: resetPayload: id: 9
D: regularTickers
reconnect in 9 seconds

User exception (panic/abort/assert)
                                   --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:191 __yield

>>>stack>>>

ctx: cont
sp: 3ffffec0 end: 3fffffc0 offset: 0000
3ffffec0:  40253290 40236400 00000000 40236400  
3ffffed0:  000000fe 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 3fff3b44  
3ffffef0:  ffffff80 3ffffde6 4020f4bc 3fff3afc  
3fffff00:  3fffdad0 3ffefed4 00000000 4024ee46  
3fffff10:  3fffdad0 3ffefed4 00000000 4024eeb3  
3fffff20:  00000000 15c28f5c 00000896 4024e509  
3fffff30:  00000051 3ffefed4 3fffdad0 4020e53a  
3fffff40:  00010101 00000000 00000000 00000000  
3fffff50:  00000000 00000000 00000000 40240000  
3fffff60:  00000000 00000000 3fffdab0 40253209  
3fffff70:  3fff02d0 3fff2960 3ffefed8 40100101  
3fffff80:  3fffdad0 00000000 3fff28d0 4020e580  
3fffff90:  3fffdad0 00000000 3ffefed0 40235010  
3fffffa0:  feefeffe feefeffe 3fffdab0 4024e606  
3fffffb0:  feefeffe feefeffe feefeffe 40101d65  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

lumapu, 2023-03-09:
can we close this issue?

lumapu, 2023-03-09:
I never used them, @stefan123t introduced them in the very beginning to determine the cause of random reboots (ESP8266)

lumapu, 2023-05-25:
@Argafal what's your today's state about this issue?

closed 2023-05-26 by Argafal