lumapu/ahoy #661: Rephrase "not producing" message

Argafal, 2023-02-07:
Running 0.5.80. Upon start the inverter status changes to

`Inverter #1: Central (v10016) is available and is not producing`

This is not correct, the inverter is producing. 

How about "... is available, recent production data has not been received"?

lumapu, 2023-02-07:
this is the sight from Ahoy. If it has not received any data it does not know if the inverter is producing. Normaly Ahoy runs all the time and starts receiving data before the inverter boots.

Argafal, 2023-02-07:
Right, I agree. However, it does not say that it "does not know if the inverter is producing". It says it knows what the inverter is doing and that it isn't producing. ;) Hence my suggestion to rephrase it.

Argafal, 2023-02-08:
[image]

Sweet, now there is actually a distinction between "no data (yet)" and "inverter actively answered that it is not producing"? That is fabulous, well done <3

closed 2023-03-27 by lumapu