lumapu/ahoy #662: Please add "state_class": "total_increasing" to homeassistant/sensor/XXXX/ch0_YieldTotal/config

softservicegit, 2023-02-07:
The home assistant energy dashboard doesnt accept the "YieldTotal" values without haveing a "state_class".
It makes scence to add this to every YieldTotal value. But at least the ch0 must have iut to get this value accepted in the energy dashboard. 

Here is my sample which I used for testing:

{
  "name": "Gate YieldTotal",
  "stat_t": "inverter/Gate/ch0/YieldTotal",
  "unit_of_meas": "kWh",
  "uniq_id": "114182932169_ch0_YieldTotal",
  "dev": {
    "name": "Gate",
    "ids": "114182932169",
    "cu": "http://192.168.10.26",
    "mf": "Hoymiles"
  },
  "exp_aft": 65,
  "dev_cla": "energy",
  **"state_class": "total_increasing"**
}

antibill51, 2023-02-09:
@softservicegit 
On which version are you having this problem? For me total_increasing is present on all Yield* entities.
There have been fixes regarding HA auto discovery recently.

softservicegit, 2023-02-09:
I was on the latest stable version 0.5.66 - After switching to the current development build the state_class is present.
No further action needed. Thanks.

closed 2023-02-09 by softservicegit