lumapu/ahoy #665: IPv6 API-access from outside. Portfreigabe, portforwarding

Schnee-Mann, 2023-02-08:
Hallo,
unterstützt die ahoy DTU einen Zugriff über IPv6?

Der ESP müsste das ja können: https://github.com/IPv6-ESP8266/IPv6-ESP8266

Ich versuche von außen auf die API-Schnittstelle über Portfreigabe in der fritzBox auf die DTU zuzugreifen. Leider ohne Erfolg.
Per IPv4 geht es ohne Probleme.

In der Fritzbox habe ich von Hand die IPv6 Interface-ID eingetragen.
(berechnet aus der DTU Mac-Adresse)
- Portfreigabe für IPv6 für Port 80 ist aktiv
- Ping6 ist freigegeben
 
Weder ‘telnet’ noch ‘ping6’ können auf die DTU zugreifen. (getestet von zwei unterschiedlichen Maschinen außerhalb des Heimnetzes)
Schonmal vorab Danke für eure Infos :-)

Ollipop030, 2023-02-08:
Sowas Ähnliches gab es hier schonmal:

https://github.com/lumapu/ahoy/issues/508

Es ist aber nicht sehr schlau so ein Loch in das Netzwerk zu reißen, VPN ist da eine sicherere Lösung

Schnee-Mann, 2023-02-08:
Sicherheitsmäßig ist Portfreigabe immer so eine Sache... Da stimme ich dir voll zu.

Aber wie ist es denn mit der IPv6 Unterstützung? Gibt es Möglichkeiten auf die API über IPv6 zu kommen?
Den #508 hatte ich mir schon angesehen. Da gibt es leider keine Infos. 

Im Quellcode der DTU habe ich nichts speziell IPv6 gefunden.

Ollipop030, 2023-02-08:
Dann ist es wahrscheinlich auch nicht implementiert. Mein Ahoy bekommt intern erst gar keine v6 zugewiesen. Aber wäre folgendes nicht besser?:
VPN zwischen den beiden Netzen über IPv6, danach per v4 weiter zur DTU? Warum muss es denn unbedingt v6 sein?

Schnee-Mann, 2023-02-08:
Es geht um einen Cloud-Server, der die API mehrere DTUs anspricht um die Hoymiles Daten zu speichern.

Eine der DTUs ist an einer Fritzbox, die leider nur eine IPv6 vom Internetanbieter zugewiesen bekommen hat (DS-Lite-Tunnel).

Ich könnte ja quasi über den Portmapper '6tunnel' die Box per (v4) -> (portmapper v4 to v6) -> (v6 an fritzbox) ansprechen. Aber wenn die Fritzbox per v6 angesprochen wird, leitet diese die Ports irgendwie nur an v6 - Geräte weiter.
Eine Möglichkeit dass die Fritzbox als 'Translater' IPv6 to IPv4 als Portmapper funktioniert ist mir nicht bekannt.

Für VPN habe ich in diesem Fall das gleiche Problem.

Also irgendwie müsste die ahoy DTU ein IPv6 Gerät sein.
 
Oder bin ich mit meinen Überlegungen da vollkommen falsch unterwegs.

stefan123t, 2023-02-08:
Es gibt aktuell m.W. keinen ESPAsyncTCP IPv6 Stack für den ESP8266.
Die Kollegen in ESPHome haben es wohl für ESP32 umgesetzt da es dort angeboten wird, aber ESP8266 wird da nicht mitspielen können.
Siehe [esphome/feature-requests IPv6 support #718](https://github.com/esphome/feature-requests/issues/718)

Schnee-Mann, 2023-02-09:
Hi Stefan123t,
irgendwie scheint IPv6 auch mit dem ESP8266 zu gehen.
https://github.com/IPv6-ESP8266/IPv6-ESP8266

Vielleicht wäre das ja ein Plan für ein neues release. 
Damit kenn ich mich aber leider garnicht aus.

stefan123t, 2023-02-16:
Das ist ein IPv6 Proof of Concept für das FreeRTOS-SDK des ESP8266. Wir verwenden das Arduino/ESP-IDF das heißt wir müssten mE sehr viel anpassen damit wir das überhaupt als build target unterstützen können. Der PoC ist fünf Jahre alt und wurde seitdem nicht mehr angepast. Auch sind die genannten Probleme mit Reboots / Stabilität nicht angegangen bzw. Es ist auch offenbar nicht gedacht oder möglich IPv4 und IPv6 anzubieten. Dh man bräuchte eine IPv6-only Version.
Ich glaube nicht dass wir das unterstützen bzw support hierfür anbieten können.

closed 2023-02-16 by stefan123t

Schnee-Mann, 2023-02-16:
Danke für die Info. :-)

Calvin0685, 2023-03-25:
Tasmota has gained great success in implementing IPv6 for ESP32 and ESP8266. Details can be found in the related issue https://github.com/arendst/Tasmota/issues/2521 and their release notes https://github.com/arendst/Tasmota/releases/tag/v12.4.0. Maybe their implementation is helpful or even compatible to reuse in
ahoy.

lumapu, 2023-03-26:
danke für die Hinweise. Ist das weiterhin relevant und interessant?

Schnee-Mann, 2023-03-27:
Bisher hatte ich nur eine Anfrage bezüglich IPv6 (DSLite-Tunnel). Also für die Zukunft könnte das vielleicht interessant werden, aber aktuell ist das noch nicht wirklich häufig, dass die DTU über IPv6-portforwarding erreichbar sein kann.

Calvin0685, 2023-03-27:
Nunja als das Thema DS-Lite-Tunnel trifft quasi alle Kabelnutzer... Denn dort hast du nur noch den Tunnel für IPV4 soweit mir bekannt.
Demnach auf alle Fälle relevant, da sonst kein Portforwarding möglich und somit auch kein Zugriff von außen realisierbar.

Schnee-Mann, 2023-03-28:
Mit VPN hat man doch das gleiche Problem, oder habe ich da einen Denkfehler?! Wenn man nur eine v6-Adresse bekommen hat, kann man doch intern auch nur v6 - Geräte ansprechen. Egal ob Port-Forwarding oder VPN... oder???

Und ob das so cool ist.... Die Netzbetreiber haben einfach nicht genügend v4-Adressen kaufen können. Das sind meisten die kleineren Netzbetreiber, die erst später in den Markt gekommen sind. Telekom, 1und1, ... haben halt schon echt ne Menge v4 IPs gesichert. Also ne ahoyDTU als v6-Gerät wäre wahrscheinlich schon irgenwann sinnvoll.

Siginator, 2024-01-11:
In der aktuellen FRITZ!Box OS wird WireGuard unterstützt. Mit dieser VPN Verbindung sind ipv4 Teilnehmer direkt auch über DSlite zu erreichen.