lumapu/ahoy #666: Confusing configuration for LOLIN D1 mini V4.0

martinriedel, 2023-02-08:
I am using a LOLIN D1 mini V4.0 with Type-C USB Port and i have to switch the yellow and grey pins and have to switch the configuration for CE and IRQ (D3 and D4) as well as i mentioned [here}(https://github.com/lumapu/ahoy/issues/36#issuecomment-1396664875). Can somebody explain me why? I would assume that change the config will revoke the switch of the pins

stefan123t, 2023-02-08:
Dear @martinriedel we do not know why some of the D1 mini versions do not work with our default Pinout setting.
If you can figure this out we are happy to add it to our documentation.
Currently we assume that there might be something different in the default wiring of these modules which prevent us from using them as IRQ / CE (which is actually _CE/NCE, a negated CE).
Maybe the vendors add pullups / pulldowns which conflict with our intended use ?

martinriedel, 2023-02-09:
Sure I can add it to the documentation. If you need a Lolin D1 Mini i can send one to you, just ping me

stefan123t, 2023-02-16:
Maybe adding a picture of your board and a working pinout should do it for the time being. Dont know if some of the testers on Discord would be eager to play with the board.

closed 2023-07-23 by lumapu