lumapu/ahoy #667: [ESP8266/ESP32/RaspberryPi] Problem Description / Beschreibung

MubiTec, 2023-02-08:
### Platform

ESP8266

### Model name

Wemos d1 mini 

### nRF24L01+ Module

square dot

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

Git SHA: f8fe044 :: 0.5.66

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

... 

### Debug Serial Log output

```shell
...
```

### Error description

Hallo,

Ich habe ein persistentes Limit bei 5% gesetzt. Morgens beim Start des WR liegt also das 5% Limit an. Dieses schalte ich dann per MQTT nach Start hoch auf 100%, die Limit Änderung erscheint auch im "Live" Bild. Das klappt immer. 
Nun zum Problem: Ungefähr einmal pro Woche ist die tatsächliche Einspeisung nur bei 5% (30Watt von 600Watt),obwohl die Sonne scheint. Schicke ich dann noch mal ein Limit von 99% hinterher, so scheinen dann erst die 5% aufgehoben zu sein und er produziert mehr Strom.
Wie gesagt, ahoy zeigt das Limit auch schon vorher bei 100%.
Ist vielleicht sogar gar kein ahoy Bug sondern ein WR Bug.
P. S. Ich setze das persistente Limit, weil ich nachts den WR zum Einspeisen von Batterie nutzen will.

stefan123t, 2023-02-08:
@MubiTec das persistente Limit sollte man nur sehr selten setzen da sonst der Flash Speicher des WR irgendwann in Mitleidenschaft gezogen wird.
Das temporäre / non-persisten Limit kann ggf. das persistent Limit übersteuern.
Evtl. bekommt der WR Dein non-persistent Limit um die Uhrzeit noch nicht mit, bzw. entfällt ihm diese Einstellung aufgrund einer DC-Unterbrechung ?
Für einen Bug brauchst Du ein paar mehr Debug Logs damit wir hier etwas sehen können was evtl. schief läuft.
Da es aber bei vielen anderen problemlos funktioniert denke ich es liegt eher an einem Verständnis / Kommunikationsproblem.
Ich ändere das Issue in eine offene Frage.

MubiTec, 2023-02-09:
Ja, das persistent Limt ändere ich auch nicht.
OK, dann noch eine Verständnisfrage: das "active_PowerLimit" ist ja eigentlich das gerade aktive Limit des Wechselrichters, was er selber schickt. Er bestätigt mir dann ja damit, dass mein Wunsch die 100% einzustellen, erfolgreich war.
Dann wundert es mich halt, dass trotz "active_PowerLimit" =100% nur 5% produziert werden.
Ist wahrscheinlich schwierig zu debuggen. Ich versuchs einfach mal damit, den Wert mehrfach zu senden

bassbumper, 2023-02-09:
Hi, ich habe das gleiche Phänomen. 
Sprich ich habe ein persistentes Limit von 600W. Das habe ich irgendwann mal gesetzt. Im Laufe des Tages erhöhe ich das non-persistente Limit je nach Verbaucherlast, z.b. auf 800W. Über Nacht wird der Inverter wohl automatisch wieder auf das bestehende persistente Limit von 600W zuückgesetzt (was ja auch gewünscht ist). Die "Live" Seite und active_PowerLimit zeigt morgens aber immer noch das letzte non-persistent Limit, hier 800W an, obwohl der Interter trotz volleinstrahlung bei 600W definitiv abriegelt, was dafür spricht, dass das persistente 600 Limit aktuell aktiv ist und nicht die angezeigten 800W.

Kommt das active_PowerLimit direkt vom Inverter oder merkt sich das die DTU und wird ggf nicht resetted wenn auch ein permanentes Limit existiert?

beegee3, 2023-02-09:
es sind mehrere Effekte, die sich hier überlagern:

@MubiTec eine Änderung des Limits ist zunächst eine Anfrage/Anweisung an den Inverter. Dieser antwortet aber nicht von sich aus, ob er das auch umgesetzt hat. Die Info muss mit einer neuen Anfrage geholt werden.
In der Version 0.5.66 wird nur die erste Anfrage gestellt und das "neue" Limit angezeigt, davon ausgehend, dass der Inverter das auch umsetzt. Mittlerweile ist die Erkenntnis, dass das wohl nicht immer funktioniert. Daher gibt es in den development03 Versionen (aktuell 0.5.83) zusätzlich die zweite Anfrage. D.h. spätestens in der nächsten "stable" Version ist das auch drin.

@bassbumper ja, das non-persistente Limit ist nur aktiv bis sich der Inverter abschaltet (z.B. zur Nacht). Beim nächsten Anschalten gilt entsprechend das persistente Limit.
Dieser Effekt wird tatsächlich bisher von Ahoy nicht berücksichtigt, so dass morgens noch der alte Wert angezeigt wird.
Ahoy müsste eigentlich nach jedem "längerem" Inverter-Offline das aktive Limit neu abfragen.

@stefan123t also doch ein 🐞, oder?

beegee3, 2023-02-09:
ich meine, in der neuen Version meldet Ahoy nur das Ergebnis der 2. Anfrage.
Wenn es geklappt hat, wird die Änderung auch publiziert (serial log, Webseite, MQTT).
Bin mir nicht sicher, ob auch etwas publiziert wird, falls es nicht geklappt hat, sprich neuer gleich alter Wert ist.
Wahrscheinlich ist es sinnvoll, nach dem Senden des neuen Limits kurz zu warten, bevor gecheckt wird. Evtl. spielt auch das MQTT Interval Setting eine Rolle.
Und bitte beachte, dass die development03 Versionen Testversionen sind, wo nicht unbedingt alles "rund" läuft. Nicht, dass es bei dir plötzlich an anderer Stelle hakt.

beegee3, 2023-02-09:
> Aber vielleicht könnte ahoy das ja direkt intern noch mal versuchen, wenns nicht geklappt hat.

stimmt. Wäre wirklich cool. Die Auswertung der 2.Anfrage ist bisher unabhängig von der 1. Da müsste mit der 1. der Wunschwert gespeichert werden und das mit der Antwort auf die 2. abgeglichen werden, um ggfs. den Prozess zu wiederholen. Das darf natürlich nicht in einer Endlosschleife enden, nur weil der WR partout nicht bereit ist, etwas zu ändern (warum auch immer). Und ich bin nicht sicher, ob der angepasste Wert exakt mit dem Wunschwert übereinstimmen muss, eine kleine Unschärfe wäre auch denkbar. @stefan123t als wandelndes HM Lexikon weiß evtl. mehr darüber 📖

> Gibt's denn schon eine Ahnung, wann die nächste stable fertig ist?

da fragst du den falschen, bin auch nur ein einfacher User, der die Entwickler immer mal mit abstrusen Ideen überschüttet und, begeistert von dem Projekt, hier und da sein Halbwissen verbreitet 😁.
Gerade gibt es ein paar Baustellen mit memory leaks und MI Inverter werden integriert. Letzteres ist wohl auf einem guten Weg. Einfach abwarten, dein Problem hast du ja im Griff 😄

MubiTec, 2023-02-09:
Wo wir gerade bei den Limits sind. Ich hab wie gesagt ein persistentes Limit bei 5% drin.
Wann wird das aktiv? Ja eigentlich wenn der HM eine relativ kurze Zeit stromlos war, oder? Meine irgendwas von max. 1 Minute oder so gelesen.
Der letzte Empfang vom WR war heute 1h, da hab ich die Batterie angemacht und der HM hat sich direkt 11A mit 50V gezogen. Dürfte er ja eigentlich nicht, da auf 5% also 15Watt begrenzt. Hab das mal am nächsten Morgen ausprobiert, also ganze Nacht strom los und dann gings...
Jemand da nähere Infos zu?

bassbumper, 2023-02-10:
@MubiTec 
> Ja eigentlich wenn der HM eine relativ kurze Zeit stromlos war, oder? Meine irgendwas von max. 1 Minute oder so gelesen. 
Im Ahoy Manual steht "... after a power cycle of the inverter (P_DC=0 and P_AC=0 for at least 10 seconds) ..."

lumapu, 2023-02-10:
damit der Inverter neu startet muss auf jeden Fall DC-seitig 0V sein, AC dürfte egal sein

beegee3, 2023-02-10:
wo wir beim Limit sind: wenn man versucht ein Limit zu setzen und der Inverter hat sich gerade abgeschaltet, kommt man in eine Endlosschleife von "retransmit power limit". @lumapu: unbedingt dort auch die Anzahl der Wiederholungen limitieren!

lumapu, 2023-02-10:
@beegee3 kannst du hellsehen? Habe gerade mal in die Webconsole geschaut und das gesehen ...
[image]

stefan123t, 2023-02-16:
@beegee3 das mit dem wiederholten setzen des active PowerLimit bis der Soll-Wert vom WR auch übernommem wurde ist (zumindest für das non-persistent / temporäre) Limit sicher eine gute Idee.
Dazu passt dann evtl auch ein Setting, dass er nach dem Schlaf am nächsten Morgen oder anderweitigem Reset auf das persistent Limit nach DC Outage das aktuelle Limit nochmal abfragt und ggf das non-persistent Limit entsprechend dem Soll-Wert und den Retries (max 5 wie beim Senden/Emofang ?) wiederholt.
ich weiss an der Stelle zwar nicht wie es die Hoymiles DTU Pro macht aber dort hat man ja auch eine SMiles Cloud in der Regel am Week die ggf die Kommandos erneit schickt.

closed 2023-03-27 by lumapu