lumapu/ahoy #668: MQTT entities in Homeassistant randomly unavailable

linuxstony, 2023-02-08:
Hello,

i run the latest version of AHOY 0.5.66 and everything is setup fine.

The only problems I have that the device is unavailable from time to time - which I solved by creating a HA Automation to restart the smartplug when it is detected as unavailable.

The main problem I have though is that homeassistant some times forgets the entities that AHOY provides - only a sending of the discovery command solves this problem. 

So this is not really a bugreport rather a question to your guys if this is an issue possibly originating from AHOY, MosquittoMQTT or more on the Homeassistant side?

Would it be possible to send the discovery command regularly or is this undesireable?

KG3RK3N, 2023-02-08:
The entities are unavailable if the value expired (depending on your configured update interval in ahoy) and no new value received by home assistant. That can caused by different things like bad wifi connection, the inverter is not available or ahoy needs to many time to send the new state.

Your issue that home assistant forgets the entities sounds like a missing retain flag on the mqtt messages. 

You should check the latest dev, possibly your issues are already resolved.

linuxstony, 2023-02-08:
Alright - thanks for taking the time - i will check and report back

stefan123t, 2023-02-08:
@linuxstony we are on 0.5.83 with some changes to how MQTT gets published, especially during auto-discovery or initial promotion of the topics especially for Home Assistant. Though not everything may be working some bugs regarding a broken JSON payload have been fixed in those versions since your 0.5.66 stable.

niekniek89, 2023-02-09:
is version 0.5.83 an stable version, or in beta?

KG3RK3N, 2023-02-09:
0.5.83 is beta

foobarth, 2023-02-27:
> Your issue that home assistant forgets the entities sounds like a missing retain flag on the mqtt messages
> You should check the latest dev, possibly your issues are already resolved.
So the MQTT messages are retained in newer builds? While that totally makes sense if the inverter is not polled at night, i still see vanishing entities after shutdown. Namely all realtime values such as power, irradiation and so on vanish after shutdown while the counters (energy, etc...) retain.

lumapu, 2023-02-27:
we have a documentation which tells you the retained values:

https://github.com/lumapu/ahoy/blob/development03/User_Manual.md#mqtt-publish

closed 2023-03-27 by lumapu