lumapu/ahoy #672: LED leuchtet durchgehend

MsM4711, 2023-02-10:
Warum leuchtet die blaue LED bei den aktuellen Versionen ständig?

Ich nutze 5.83. Aber bei den letzten drei Versionen war das auch schon der Fall.

Kann man das ändern?

## Hardware
  * [X ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

### Antenna:
* [X] circuit board
* [ ] external antenna

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

## Software
* [X ] AhoyDTU
* [ ] OpenDTU

Jumperm, 2023-02-10:
Ich habe das selbe Problem. Ich schätze, dass es seit der DEV besteht, in der man die GPIOs neu einstellen musste. Genauer kann ich es gerade nicht eingrenzen.

lumapu, 2023-02-10:
wird schon in #656 besprochen, ist aber so nicht ersichtlich daher die Verlinkung.
Wir werden es einbauen

mroenne2022, 2023-02-10:
Ich hab 2 ESP32 WROOM , beide mit der 30Pin Version , einmal Ahoy und einmal OpenDTU - bei den beiden leuchtet von Anfang an immer die rote LED, egal welche Version ich raufbringe. Ich glaub das ist bei denen normal - zeigt wohl an das Stromversorgung anliegt !?
Jetzt hab einen ESP32 mit 38Pin mit Ahoy 0.5.66 gebaut , da ist es nicht der Fall - da sind die LEDs aus.

Bjoern3003, 2023-02-10:
Habe es ebenfalls (ESP8266). Ebenfalls DEV Version 0.5.83. In der aktuellen Release Version (0.5.66) war dies auf jeden Fall nicht so.

dtuuser, 2023-02-10:
Habe aktuell 2 AZ Wemos 8266 aktiv. Wenn ich IRQ an D4 nutze, dann leuchtet die LED. Bei Nutzung von D2 ist das nicht so.

mroenne2022, 2023-02-10:
Ich hab wie gesagt ESP32  - keine esp8266 ! 
Da ist es offensichtlich anders.

MsM4711, 2023-02-10:
Mit der 0.5.84 wurde es bei mir behoben.
Viele Dank! Tolles Projekt!

beegee3, 2023-02-10:
@lumapu die `loop` von `hmRadio` in v0.5.84 ist noch nicht ganz korrigiert, hast da etwas verwechselt

1. `mNrf24.startListening();` hinter `mNrf24.setChannel(mRfChLst[mRxChIdx]);` muß wieder rein, sonst wird das erste Listening erst nach 4 ms durch den Channel switch eingeschaltet
2. dafür `mNrf24.stopListening();` hinter `if (getReceived()) {` löschen, sonst geht nach erfolgreichem Empfang die LED bis zum nächsten Empfang doch nicht aus

closed 2023-03-27 by lumapu