lumapu/ahoy #674: 0.5.85 auf ESP32 hängt in Rebootschleife

Ollipop030, 2023-02-11:
Mein ESP32 hängt nach Flash auf 0.5.85 in ein Rebootschleife, das Webinterface ist sehr träge in der Zeit wie er erreichbar ist. Ich komme von der 0.5.83, diese lief jetzt knapp 72 Stunden durchgehend ohne Probleme.

Hat das noch jmd?

knickohr, 2023-02-11:
Kann ich nicht bestätigen, habe heute auch die 85 drauf gehabt, lief flüssig. Bin aber wegen eines anderen Problems eben wieder zurück auf die 78.

Ollipop030, 2023-02-11:
Keine Chance: Clean ESP32. 0.5.66 stable installiert, dann auf 0.5.85 aktualisiert. Ein paar Netzwerkeinstellungen gemacht und der ESP startet neu, Webinterface ist sehr langsam, Uptime kommt nur über ein paar Sekunden und staret wieder bei 0. Über das Webinterface dann die 0.5.83 installiert (ein bisschen tricky weil der ESP immer wieder neu startet). Läuft auf Anhieb ohne Neustarts und das Webinterface ist auch viel schneller.

lumapu, 2023-02-11:
ok, ich hatte dieses Verhalten durch meine Neuerung die Werte der Wechselrichter sofort nach erfolgreichem NTP sync anzufragen.
Für die veröffentlichte Version habe ich dann eine Verzögerung von 3 Sekunden eingebaut was bei mir das Problem gelöst hat.
Danke fürs Feedback, dann deaktiviere ich es, bis ich verstehe warum er in meinen Fall sofort neu startet wenn ich die 3 Sekunden weglasse

lumapu, 2023-02-12:
0.5.86 sollte das nicht mehr machen, biite um Feedback

Ollipop030, 2023-02-12:
Danke für die schnelle Hilfe, jedoch verhält sich die 86 genau so.

knickohr, 2023-02-12:
Ahhh jetzt. Ich wundere mich schon warum mein ESP32 das nicht macht (das Booten). Der läuft stabil und hat auch kein Watchdog-Problem. Es ist ein ESP32 DevKit C (keine V4 !) von AZ mit 38 Pins.

Ollipop030, 2023-02-12:
Ja hätte ich vielleicht direkt dazuschreiben sollen, sorry. Die 38 Pin Varinate war damals leider nicht lieferbar, deshalb habe ich mir die "kleinere" Variante geholt. Bisher gab es auch absolut keine Probleme, die Boards laufen tagelang ohne einen Neustart.

teletapi, 2023-02-12:
> Ahhh jetzt. Ich wundere mich schon warum mein ESP32 das nicht macht (das Booten). Der läuft stabil und hat auch kein Watchdog-Problem. Es ist ein ESP32 DevKit C (keine V4 !) von AZ mit 38 Pins.

Ich nutze den Vemos ESP32 D1 Mini der hat bisher auch keinerlei Probleme diesbzüglich angezeigt

Ollipop030, 2023-02-12:
@lumapu 
Was meinst du, ist das zu fixen oder soll ich lieber auf ein 38 Pin Board umsteigen? So teuer sind die ja nicht. Evtl. kommen sonst ja noch andere Probleme in Zukunft auf mich zu.

lumapu, 2023-02-12:
ich weiß bisher nicht was der Unterschied ist. Wenn es wirklich nur die Pins sind dann tippe ich eher auf ein Power-Supply Problem. Das kann auch der auf dem Board verbaute Spannungsregler sein.
Könntest du das ausschließen indem du den ESP noch einen Kondensator spendierst

Ollipop030, 2023-02-13:
Ich habe jetzt noch ein wenig hin und her getestet. Scheinbar macht die DTU nur Neustarts, wenn sie "was zu tun" hat. Sprich wenn sie mit den WR kommunizieren soll. Wenn keine Antenne dran ist, läuft sie stabil. Wenn ein dran ist, aber keine WR eingetragen sind, läuft sie auch. Sobald ich die WR aber eintrage, kommen die Neustarts.
Bei diesem Verhalten würde man auf Hardware tippen, folgendes spricht aber dagegen: Die DTU läuft fehlerfrei mit Version 0.5.83 und darunter. Ich habe mehrere Antennen probiert, mit externer und ohne externer Antenne, aber alle mit Kondensator. Die Antennen selbst habe ich noch an einem 8266 getestet, da laufen sie.
Verschiedene ESP32 30PIN, unterschiedliche Netzteile (eines von meinem PiHole mit 3000mA, der läuft aktuell damit). Sobald eine Version größer 0.5.85 installiert wird laufen die Dinger nicht mehr. 

Es scheint also, dass die Boards ab Version 85 nicht mehr funktionieren. Sind übrigens diese hier: https://www.makershop.de/plattformen/nodemcu/esp32-board-stiftleisten/

soelli22, 2023-02-13:
Das mit den Reboot unter „Last“ kann ich bestätigen. Sobald auf den WR‘s Aktivität herrscht, läuft nix mehr.

lumapu, 2023-02-13:
> Es scheint also, dass die Boards ab Version 85 nicht mehr funktionieren. 

Das klingt stark danach, dass die frisch eingebauten MI inverter den Speicher sprengen. Komisch dabei ist aber, dass es sich um einen ESP32 handelt.

Ollipop030, 2023-02-16:
Heute morgen spaßeshalber mal die 0.5.89 installiert, damit läuft alles wieder. DTU hat mittlerweile 7 Stunden Uptime.

lumapu, 2023-02-16:
und hoffentlich jetzt schon 12

Ollipop030, 2023-02-25:
Kann zu, keine Probleme mehr gehabt seit einer Woche.

closed 2023-02-25 by Ollipop030