lumapu/ahoy #677: Beobachtungen zu NTP und MQTT nicht erreichbar

HFarryranz, 2023-02-13:
### Platform

ESP8266

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.86

### Github Hash

ca25f16

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

 NTP
ntp1.t-online.de 123

### Error description

Folgendes Szenario:

AHOY startet und erreicht den NTP nicht.
Ohne Zeit keine weiter (vollständige) Verarbeitung.

Im WEBIF kann man nun SYNC FROM BROWSER auswählen und es läuft erst mal alles.

Folgende Beobachtungen:

-macht man nichts (Kein SYNC FROM BROWSER)  kommt auch keine weitere Abfrage an den NTP (oder erst sehr spät?).
-solange die Zeit nicht vorhanden ist, wird nicht zum MQTT verbunden.

- ist der NTP später wieder nicht erreichbar, wird die Systemzeit genullt, statt mit der alten Zeit weiter zuarbeiten. Folge: ohne Zeit keine Verarbeitung (siehe oben)
- Wird im laufenden Betrieb der MQTT weggenommen (MQTT Dienst stoppen), baut sich das WEBIF nicht mehr vollständig auf, weil das System sich im Sekundentakt sich mit dem MQTT verbinden will.

MiniOh, 2023-02-13:
Mit ptbtime1.ptb.de als Zeitserver habe ich keine Probleme in den letzten Versionen.

HFarryranz, 2023-02-13:
Das der NTP bei mir zeitweise nicht erreichbar ist, liegt hier an meinem Netzwerk ( oder an einem Geräte das das stört).
Ich wollte damit auf die Problematik im Programmablauf hinweisen.

beegee3, 2023-02-13:
> -macht man nichts (Kein SYNC FROM BROWSER) kommt auch keine weitere Abfrage an den NTP (oder erst sehr spät?).
-solange die Zeit nicht vorhanden ist, wird nicht zum MQTT verbunden.

solange AHOY noch keine Zeit mitgeteilt bekommen hat (egal, ob  durch NTP oder SYNC FROM BROWSER), wird im 5 Sekundentakt versucht, die Zeit vom NTP Server zu erhalten. Dies geschieht still, also ohne serielle Ausgabe.
Und die Verbindung zu MQTT wird erst nach erhaltener Zeit aufgenommen, um zu vermeiden, ungültige Zeitstempel zu publishen.
> Wird im laufenden Betrieb der MQTT weggenommen ...

Ist bekannt, s. #650

HFarryranz, 2023-02-15:
Mit der 0.5.88 sind meine oG Probleme weg. Laufzeit von 1d6h hatte ich noch nie.
Kann von meiner Seite aus hier zu gemacht werden.

closed 2023-03-27 by lumapu