lumapu/ahoy #678: Limit Verständnis Frage

dtuuser, 2023-02-14:
Mal eine Frage zum Limitsetzen über CURL Aufruf.
Ich habe das so verstanden, dass ein über CURL gesetztes Limit erst beim nächsten normal geplanten Datenabruf gesetzt wird. Ist das richtig? 
Oder wird das unabhängig vom geplanten Abrufintervall sofort nach Erhalt an den WR übermittelt? 
Was passiert,  wenn ich innerhalb von wenigen Sekunden zwei Limitwünsche an die AhoyDTU sende?
Danke

Ollipop030, 2023-02-14:
Das Limit wird sofort gesetzt, sieht man auch in der Console. Bei mit wird über ein Script alle paar Sekunden das Limit Richtung Nulleinspeisung gesetzt, funktioniert wunderbar.

dtuuser, 2023-02-14:
Danke für super schnelle Antwort.

Mist...dann muss ich mein Tool anpassen. Dachte, dass das über das Intervall läuft.

closed 2023-02-14 by dtuuser

Ollipop030, 2023-02-14:
Kannst dir das ja mal ansehen, funktioniert ganz gut:

https://github.com/reserve85/HoymilesZeroExport