lumapu/ahoy #680: ahoydtu regularly unavailable

niekniek89, 2023-02-14:
## Hardware
  * [*] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [*] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [*] circuit board
* [ ] external antenna

## Software
* [*] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.66

——————————————-
ISSUE:

Hello,

I have a HM-1500 inverter with 3 panels since last week.
So I read the inverter with ahoydtu.

since I added the inverter, the ahoydtu drops out regularly.
I can no longer ping the ahoydtu (icmp), but I can still see it on the network (unifi).
if I let the device connect again (via unifi controller), the ahoydtu comes online again.
sometimes this has to be +- 6 to 8 times before it works normally again.
I then see from the uptime that it has probably restarted?
it starts again from 0 (uptime).

hardware:

wemos d1 mini v3
NRF24L01+

the problem seems to have started since the inverter + panels were installed.
anyone have an idea?

niekniek89, 2023-02-18:
Anyone??

lumapu, 2023-02-18:
the version you are using is a released one which were tested by many people without problems.
Maybe it makes sense in your case to upgrade to the latest development version 0.5.89

niekniek89, 2023-02-19:
We’re can i download de development release?

niekniek89, 2023-02-19:
it has now succeeded. the dev release is on it.

niekniek89, 2023-02-19:
it seems to work better with the latest dev release.
I'm watching it.

niekniek89, 2023-02-20:
@lumapu 

Good morning,

the ahoydtu has just fallen away again.
interface not reachable, and pinging to IP address is no longer possible.
the only thing that helps is power off and on.

Could this have something to do with the soldering, causing something to break?

lumapu, 2023-02-21:
do you track the free Heap value using MQTT? Maybe you see a decreasing value.

How many inverters do you have in your setup?

Have you tried another power supply and another USB cable? Do you have soldered a capacitor next to the radio module?

niekniek89, 2023-02-21:
Hello,

thanks for the help.
I currently have 1 inverter (HM-1500) with 3 panels.
i am using MQTT to Home Assistant.
I have not soldered a capacitor.

I have tried a different USB cable and an adapter.
it has just been restarted (and with the dev version it will also come back online).
the stable version did not come back online.

[image]
[image]
[image]

niekniek89, 2023-02-21:
@lumapu 

is it recommended to solder a capacitor on it? and do you have an image, where it should go?

niekniek89, 2023-02-22:
after exact 24 hours:

[image]

no data, ping reply

niekniek89, 2023-02-22:
after 24 hours stable, no live dashboard anymore:

[image]

I still receive data via mqtt, and the ahoydtu is still available on ICMP.

niekniek89, 2023-02-22:
And in the end the ICMP Pings get stuck, the GUI stuck .... and is he restarted?

[image]
[image]

lumapu, 2023-02-22:
> @lumapu
> 
> is it recommended to solder a capacitor on it? and do you have an image, where it should go?

yes it's defenetly necessary

niekniek89, 2023-02-23:
@lumapu 

Hello,

thanks for your help.
can you perhaps indicate which one I need, and on which pins I have to solder it?
I assume on the wemos d1 mini, but on which pins?

lumapu, 2023-02-23:
a 100uF will be enough, place as close as possible to the power pins of the nrf24 board.

niekniek89, 2023-02-24:
thanks for your help!
does it really have to be soldered on pin 1 and 2 (on red and black)?

[image]

lumapu, 2023-02-24:
yes, correct 👍

niekniek89, 2023-02-24:
again many thanks for your help!
is this the right one (see various):

Condensator 100UF 50V

lumapu, 2023-02-24:
6.3V or 16V will be ok

niekniek89, 2023-02-24:
Thansk @lumapu 

is this the right one?

https://elektronicavoorjou.nl/product/condensator-100uf-16v/

roku133, 2023-02-24:
ja, dieser Elko passt.

Es können alle Elkos eingesetzt werden, die eine Nennspannung (max. Spannung) von mindestens 6,3 V unterstützen. Die Bauform des Elkos nimmt normalerweise mit steigender Nennspannung zu.
Wichtig:   Beim Anschluss auf Polarität achten (Plus- und Minuspol):
https://www.elektronik-kompendium.de/sites/praxis/bauteil_elektrolyt-kondensator.htm

roku133, 2023-02-24:
sorry, now in English:
The referenced electrolytic capacitor fits.
You may use any e-capacitor supporting a nominal voltage (maximum voltage) with at least 6.3 V. The size of an e-capacitor usually increases with increasing nominal voltage.
Important:   Consider minus and plus pole when connecting.

niekniek89, 2023-02-24:
thanks for your help.
Then I will immediately order a capacitor.

niekniek89, 2023-02-27:
@roku133 @lumapu 

Hello,

Unfortunately I have the same issue with a capacitor.
can it be that if he loses his wifi connection (for a few seconds), he goes into AP mode, until wifi is available again, and thus restarts?

roku133, 2023-02-27:
In case of a missing or lost wifi connection it is intended to switch back to the AP mode to be able to further access the AhoyDTU, therefore this behaviour is ok.
The question is why your connection disappears from time to time, maybe it is a load issue.
My personal experience with v0.5.78, v0.5.88 and v0.5.89 has been that these versions have been less robust than the older v0.5.70. So I currently stick to this development version (my configuration: ESP8266, HM-400, one panel, wifi_rssi in the range from -72 dBm to -78 dBm, mqtt connection to mosquitto broker). Best practice has been 20 days without any interruption.
However, the feedback of other users seems to be different.

niekniek89, 2023-02-27:
@roku133 

Thanks for your help.
What do you mean by load issue?
load on the software or on the power supply?
I'm currently on 0.5.90, do you happen to have an earlier version?
can it hurt that it regularly goes into AP mode (device reboot), not that this is harmful to the hardware?

roku133, 2023-02-27:
Since you have already tested another USB power adapter (5 V, ≥ 1A) I rather assume a SW load issue.
Your wifi power level is excellent.
Earlier SW versions are available in the ahoy github repository.
Device reboots will not damage your HW, they are more or less a flaw of the system. Important data such as daily and total yields will be available after reboot.
Hope this helps.

niekniek89, 2023-02-28:
@roku133 

Hello,

thanks for your help.
which version are you currently using, and do you happen to know where I can download the specific dev version?
I would like to try the same version as you.

0.5.66 and 0.5.90 don't work very well for me, the 0.5.90 is better than the stable.

roku133, 2023-02-28:
My version currently used is 0.5.70.
The source code may be downloaded at:
https://github.com/lumapu/ahoy/tree/d8e255ddc2c9c5b09ee14e739a744ed0597c980c

As mentioned above it is not guaranteed that it will improve your situation. Version 0.5.70 shows heavy heap fragmentation. Nevertheless, it does not matter in my case. My experience may be completely different to others.
One reason is that the various variants of ESP8266 boards behave differently:
I have swapped the pinouts of CE and IRQ:   CE -> D3, IRQ -> D4.

niekniek89, 2023-02-28:
@roku133 

Hello,

is it recommended to change the pinouts? and why did you change it
I have now :

CE on D4 and IRQ on D3

can I just change these settings?
then I will let 0.5.90 run for the time being

roku133, 2023-02-28:
The standard configuration concerning my Wemos D1 mini v3 did not work at all.
After swapping the HW wiring and of course in parallel in the Ahoy setup the connection to the inverter has been successfully established (amplifier power level set to low - distance AhoyDTU to inverter about 4 m).
In your case it makes sense to stay at 0.5.90 and wait for the version 0.5.93 coming soon.

niekniek89, 2023-02-28:
@roku133 

thanks for your answer.

then changing the pins won't make a difference for me I think.
I notice that if I use the gui very often, it frequently reboots.
Do I have the right hardware, or can it not handle it?

it seems that the gui is too heavy for the wemos.

roku133, 2023-02-28:
In case there is no improvement with newer SW releases a possible step is to change the microcontroller:
ESP32 which is also supported by Ahoy is more powerful compared to a ESP8266.

roku133, 2023-02-28:
Anyway, I would recommend to wait for 0.5.93 since @lumapu will introduce some changes in the context of web handling.
ESP8266 boards should be sufficient considering your configuration with one inverter and three panels.
Of course, it is up to you ...

niekniek89, 2023-03-06:
@roku133 @lumapu 

Haven't been home for the past 5 days, and therefore couldn't access the gui (wasn't on the internal network).
It's now more than 6 days up with no reboots.
As soon as I often go to the gui (via browser), then eventually the ahoydtu crashes again, and it reboots.

So I notice now that it is not hardware but software problem.

niekniek89, 2023-03-07:
Hello,

I see that 0.5.92 is no longer downloadable, and only 0.5.88 is ready.
is it recommended to install it?

niekniek89, 2023-03-07:
@lumapu 

Hello,

where can i download other dev releases?
so that I can test several, and check which one works best for me.
it crashes when I access the URL multiple times.

lumapu, 2023-03-07:
you can simply download it from https://ahoydtu.de/web_install

niekniek89, 2023-03-10:
Hello,

I suspect it was fixed with version 0.5.97 (see issue #660).
Thank you so much for your support and help, keep it up!

closed 2023-03-10 by niekniek89

ScumbagSteve, 2024-03-25:
Since a few versions I experience exactly the same issue again. WebIf becomes unreachable after a while, but device is still connected to the network. After the nightly reboot everything works again, until later that day the same issue appears again.
Some memory becoming full seems reasonable. I am using ESP8266 and am on the latest nightly (issue exists since a few months though).
@niekniek89 can you reopen this one or should I create a new issue?