lumapu/ahoy #681: 0.5.88 Max. Anzahl Inverter? DTU nicht mehr erreichbar.

MiniOh, 2023-02-15:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Hallo, 

gibt es Infos über die max. Anzahl der nutzbaren Inverter?
nachdem ich für mich beschlosssen hatte, dass die Version 0.5.88 für mein Empfinden wieder eine recht gute und einsetzbare Version ist, hatte ich heute versucht die Inverter einzurichten.
8 Stück an der Zahl. Nach dem Save Button war die DTU nicht mehr ereichbar im WLAN.
Auch das neu Flashen per USB war alles andere als einfach, da auch hier erst nach dem ca. 10. Versuch der Verbindungsversuch erfolgreich war.

Hab ihr eine Idee was das sein kann? Kann evtl. nicht genug Speicher allokiert werden, um die Anzahl der Inverter bzw. Module zu speichern?

Danke schon mal und sorry schon mal vorab, dass ich euch täglich heue Arbeit mache. ;-)

Ollipop030, 2023-02-15:
@lumapu Könnte das geplante Update auch #674 beheben?

lumapu, 2023-03-09:
sollte mit aktueller Dev nicht mehr auftreten. Bitte nochmal testen `0.5.96`

MiniOh, 2023-03-10:
Hallo @lumapu 

Ja, funktioniert jetzt problemlos.
8 Inverter mit allen Panels und deren Leistung und Bezeichnung lassen sich jetzt ohne Fehler anlegen.

closed 2023-05-25 by lumapu