lumapu/ahoy #682: 0.5.88 "Live" Ansicht bei vielen Invertern "leer"

MiniOh, 2023-02-15:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Hallo, 

bei 9 Invertern, aber testweise nur einem mit aktiver Kommunikation, ist bei mir die "Live" Ansicht leer.

[image]

[image]

MiniOh, 2023-02-16:
Hallo, 

es hat sich offenbar wirklich etwas verändert, auch was die Live Anzeige angeht.
Mit der 0.5.89 wird die Live Anzeige nun angezeigt.

Allerdings gibts es noch Unterschiede, "an welcher Stelle" der aktive Inverter in der Liste steht.
In dem Fall, wie oben dargestellt, dass nur der 8. Inverter aktiv ist, werden keine Channel Daten außer der U_DC ausgegeben.
Lässt man aber die gleichen Inverter drin, und setzt den auf Position 3 aktiv, werden dort Daten ausgegeben.

[image]

MiniOh, 2023-02-24:
@lumapu 
In der 0.5.90 hat sich an der Live Ansicht noch nichts geändert, oder?
Zumindest bei mir wird weiterhin nur die U_DC mit Werten ausgegeben.

lumapu, 2023-02-24:
ja genau, bin schon dabei das zu verbessern, ist aber nicht nicht veröffentlicht

MiniOh, 2023-02-25:
Ok. Super. Das klingt gut.
In 5 Wochen soll unsere Anlage in Betrieb gehen, dann ist Ahoy bestimmt auch in nem sehr guten Zustand. 👍

MiniOh, 2023-02-25:
> ja genau, bin schon dabei das zu verbessern, ist aber nicht nicht veröffentlicht

Hallo, 

die 0.5.92 hat es bei mir aber eher "verschlechtert".
Bei der 0.5.89 war die Live Ansicht grundsätzlich OK, allerdings wurden, wie oben dargestellt, bei den Channels nicht alle Daten angezeigt.

In der 0.5.92 sieht "Live" wie folgt aus:

[image]

[image]

MiniOh, 2023-03-08:
Hattest du in der 0.5.95 etwas angepasst bzgl. der leeren Live Ansicht?

Ich habe nach wie vor 8 Inverter konfiguriert.
Sofern nur die "unteren" mit aktiver Kommunikation definiert sind, ist die Live Ansicht immer noch leer.
Wenn ich Inverter 0 mit dazu "aktiv" nehme, auch wenn dieser noch nicht in Betrieb ist, zeigt die Live Ansicht etwas an.

beegee3, 2023-03-08:
kann es sein, dass in `visualisation.html` `function parse(obj)` ein Fehler ist?
Da steht
```
                    for(var i = 0; i < obj.iv.length; i++) {
                        if(obj.iv[i])
                            getAjax("/api/inverter/id/" + i, parseIv);
                            break;
                    }
```
Ist `obj.iv` das gleiche wie `ivEn`? Oder sollte es so lauten:
```
                    for(var i = 0; i < ivEn.length; i++) {
                        if(ivEn[i])
                            getAjax("/api/inverter/id/" + i, parseIv);
                            break;
                    }
```

lumapu, 2023-03-08:
ich denke in der `RestApi.h` wird bereits das `enable` ausgewertet.
Der Hinweis von @MiniOh ist sehr gut, es heißt, dass beim Start der Iteration der Fehler passiert.

beegee3, 2023-03-08:
> ich denke in der RestApi.h wird bereits das enable ausgewertet.

ähh, nein. Genau dafür ist doch `ivEn`, oder bin ich im falschen Film?
Du rufst mit `getAjax("/api/live", parse)` genau die von mir zitierte `parse` Funktion auf. Die soll doch nach dem ersten enabled Inverter suchen. Das ist die zitierte `for` Schleife. Meine Frage ist, ob `obj.iv` das gleiche ist wie `ivEn`? Ansonsten wäre da wohl der Fehler. Denn dann wird `parseIv` ggfs. nicht aufgerufen (wo eine analoge `for` Schleife mit `ivEn` steht.
Warum beschreibe ich dir, was du programmiert hast? Ich bin echt im falschen Film 🎞️ 😭

beegee3, 2023-03-08:
hab mir gerade die Frage selbst beantwortet: `obj.iv` das gleiche ist wie `ivEn`.
Ist trotzdem die richtige Fehlerquelle, da fehlen die Klammern für die `if` Anweisung, richtig ist
```
                    for(var i = 0; i < obj.iv.length; i++) {
                        if(obj.iv[i]) {
                            getAjax("/api/inverter/id/" + i, parseIv);
                            break;
                        }
                    }
```
Hab's gerade mit einem virtuellen 2. Inverter erfolgreich getestet!

lumapu, 2023-03-08:
@beegee3 ich glaube wir waren gleich schnell - der commit ist bereits draußen und ich wollte gerade schreiben, dass die Klammern die Ursache waren 😂. Ich habe ewig diesen issue hier gesucht ...

MiniOh, 2023-03-09:
Danke ;)
Sieht gut aus.

closed 2023-03-27 by lumapu