lumapu/ahoy #683: different Pinlayout esp-wroom-32

uwe98, 2023-02-15:
Hello, i am irritated with the pinlayout (different to grafics)
[image]
backside: NodeMCU-ESP32   ESP32 DEVKITV1

Please, can someone help me.

vin gnd D13 D12 D14 D27 D26 D25 D33 D32 D35 D34 VN VP EN

3v3  gnd  D15  D2  D4  ax2  tx2  D5  D18  D19  D21 ax0 tx0 D22 D23

mroenne2022, 2023-02-15:
Hallo Uwe:

Ich habe 2 Ahoys mit jeweils ESP-WROOM-32  mit jeweils 30PIN (gibt Unterschiede zu 38PIN !!!)
wie folgt zum laufen bekommen!
Den Tip hab ich hier vor einiger Zeit bekommen.
Es geht immer um die GPIO's , ich musste nur den IRQ ändern,
im Setup und Verkabelung auf GPIO 01 , damit läuft es.
Einstellung im Ahoy-Setup

CS - D1 (GPIO5)
CE - D2 (GPIO4)
IRQ - TX (GPIO1)

Auf dem ESP und dem NRF sieht es wie folgt aus:

Farbe 	ESP32 	nRF24L01+
Schwarz  GND 	Pin 1
Rot 	     +3.3V 	Pin 2
Blau 	D4 	        Pin 3
Lila 	        D5 	        Pin 4
Braun 	D18 	Pin 5
Gelb 	D23 	Pin 6
Orange 	D19 	Pin 7
Grün 	TX0 	Pin 8

Anbei eine Übersicht : 
[image]

marco050969, 2023-05-01:
Hi,

hättest du ein komplettes Foto oder Pinoutbeschreibung vom esp32 mit 30 Pin und von der Antenne oder einen Anschlussplan bitte.
Ich verzweifele gerade und werde nicht schlau draus.

Grüße

Marco

mroenne2022, 2023-05-01:
Moin Marco,  ich kann dir was aus meiner eigenen Doku anbieten, musst mal gucken ob du das damit hinbekommst, sollte aber möglich sein.
[image]
[image]

closed 2023-07-23 by lumapu