lumapu/ahoy #685: Verbindung zum NRF24 Moudl

MaBo0, 2023-02-16:
Hallo,

meine erste (gelötete) AhoyDTU hat funktioniert bis sie in Kontakt zu 230 V gekommen ist.
## Hardware
  * ESP8266
  * NRF24L01+ PA LNA SMA mit Antenne
  * 100 uF Kondesator

Nun habe ich zwei weitere aufgebaut (gelötet und durchgepiept) und bekomme die Fehlermeldung "your NRF24 module can't be reached, check the wiring and pinout". Ich habe einen mit Antenne und einen ohne aufgebaut und hatte bei beiden auch einen Kondensator angelötet.
Gerne würde ich wie empfohlen ein anderes Pinout versuchen, doch ich weiß nicht wie.
Hat jemand einen Tipp für mich?

Grüße

Matthias

## Hardware
  * ESP8266
  * NRF24L01+ /  * NRF24L01+ PA LNA SMA mit Antenne

Modelname: ______
Retailer URL: https://www.ebay.de/itm/153891459582

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [X] external antenna

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |
Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [X ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: _______

### Build & Flash Method:
* Webinstaller

### Desktop OS:
* [ ] Linux
* [ X] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

RealGue, 2023-02-16:
Was sagt denn deine System Config?

MaBo0, 2023-02-16:
Ich hoffe ich antworte hiermit auf deine Rückfrage:

System Information
sdk                  2.2.2-dev(38a443e)
cpu_freq          80
heap_free        18424
sketch_used     486
version           0.5.66
wifi_rssi          -62
ts_uptime      52
esp_type        ESP8266
core_version   3.0.2
flash_size       4096
heap_frag      21
max_free_blk   7816
reboot_reason  Power On

Radio
nrf24l01+        is not connected
RX success       0
RX fail              0
RX no answer   0
RX frames received   7575
TX count      0

closed 2023-02-16 by MaBo0