lumapu/ahoy #687: Ahoydtu live dashboard

niekniek89, 2023-02-17:
Hello,

I have had the HM-1500 with 3x panels since last week.
now also got the ahoydtu working.
however, I see 4 channels in my live dashboard.
How can I make sure that I only get 3 channels on screen?
I have not entered a value for channel 1, because there is no panel on it.

niekniek89, 2023-02-18:
Is it possible to remove or delete channel 1 (is not connected).

lumapu, 2023-02-19:
not planned, and makes no sense to me. Not connected an everything is zero, just the state of your inverter.
I think it's an very individual problem and my idea how to fix that would be another panel connected to this input 😉

Mart-Muc, 2023-03-05:
I think it is relevant- makes it much more clearer and gives better overview on small screens (e.g. phones)
No need to scroll down.
Should be easy to implement. Do not show if maxpower=0 (or disabled =-1)

i would love this feature:-)

niekniek89, 2023-03-05:
that would be a nice addition indeed.
i now only have 3 panels but see 4 on the dashboard. the 4th obviously doesn't do anything either.

niekniek89, 2023-03-10:
Hello,
I see that with version 0.5.97 the disconnected channel is gone!
thank you very much.

closed 2023-03-10 by niekniek89