lumapu/ahoy #690: ESP8266 0.5.89 - wifi-Settings cannot be saved

humus2002, 2023-02-17:
### Platform

ESP8266

### Model name

WEMOS D1 mini 

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.89

### Github Hash

b6a4150

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*

#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [ ]
    - Serial Debug: [ ]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

- reset of my WEMOS D1 mini completely by flashing a 1MBblank.bin using the tool tasmotizer1.2.exe
- upload of new AHOY-DTU firmware 0.5.89 using tasmotizer again
- disconnect & reboot 
- as expected, the access point ahoy-dtu can be connected using 192.168.4.1 (but slow) 
- tried to enter WiFi credentials first, the scan button triggered WiFi search
- several attempts needed, then my WiFi occured (-55db) 
- I choose my WiFi and entered my password 
- SAVE & reboot
- the AHOY-DTU access point occured again, neither the WiFi nor the password has been saved 
- I checked carefully password / settings and tried to save my WiFi-settings several times 
- no success, after reboot / power-off/power-on or reset button, the access point occurs again
- I flashed 0.5.17 and everything is running smoothly - the AHOY-DTU gets data from a hm600 and a hm1500 since several hours now
- I flashed another WEMOS D1 mini (without RF-module) and found the following problems
  (which maybe have to do with the RF-module??) 
- the button ADD Inverter does not show any effect
- system config / display config section are not complete 
-

humus2002, 2023-02-17:
[image]

humus2002, 2023-02-17:
sorry, I closed this wrongly

humus2002, 2023-02-23:
re-tested with firmware 0.5.90 - same behaviour...

lumapu, 2023-03-11:
please check again with most recent verison `0.5.98`

theunknownstarman, 2023-03-12:
I have the same problem with 5.99

humus2002, 2023-03-13:
I have re-testet this using 5.98 - same behaviour 

also, the scan wifi buttion does not work always, i.e. either there is no reaction when I push the button or the network
is not found it does not matter whether how close the ESP8266 is to the access point / router

I tried also to enter the network credentials directly without using the scan function but is not working, too

no matter what I try - I can't save the network settings such as the ESP8266 can be accessed via WiFi (instead of 
direct access by 192.168.4.1) 

I am using a fritzbox 7590 and everything is fine with my second productive ESP8266 on firmware 5.17

lumapu, 2023-03-13:
do you have special characters in your SSID? Once you enter your SSID manually make sure that there is no space added in the end

theunknownstarman, 2023-03-13:
no special characters and no spaces

closed 2023-03-27 by lumapu