lumapu/ahoy #691: 16x2 Display in AhoyDTU

BastyESP, 2023-02-17:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [x ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
* [ ] 

Hallo,
seh hier auf der Github Seite noch nicht ganz so durch, hoffe aber das ist der Bereich wo man Anregungen oder Fragen äußern kann. Bin bisher mit der AhoyDTU noch gar nicht großartig warm geworden, aber aufspielen und einloggen geht, da meine "Balkon" Solaranlage noch nicht läuft, sehe nur in eBay-KA auch genügend AhoyDTU mit Display, dort gibt es die 2 Oled (1,3 und 0,96") und das Nokia Display, beim Nachbar Projekt OpenDTU, wird auch offen über diese 3 Typen berichtet. Wäre es machbar noch einen weiteren Typ einzubauen, das normale 16 Zeichen/2 Zeilen Display angeschlossen über i2c. 

Vorschlag:
1. Zeile "Online 237,8w"
2. Zeile "Today/Total im 10sek wechsel"

Wäre jetzt mal das was mir so vorschweben würde, wenn der jenige der das erstellen kann, mit Abkürzungen Arbeiten möchte ist das auch kein Problem, die Ablesbarkeit für mich ist jedenfalls deutlich höher wie mit den mini Grafik Displays und wenn ich es genauer haben möchte logge ich mich ein und sehe es.

Hoffe ich habe hier die Anfrage richtig Platziert und es kann jemand umsetzen.
Schönen Freitag noch.
Gruß Bastian

Loetnase, 2023-02-18:
Hallo Bastian,
schau dir mal das Nokia Display genauer an, das ist gerade mein "größter" Favorit.
Es hat zwar wenig Pixel ist aber eigentlich das größte aktuelle Display.
Auf dem Bild siehst du zwei nicht aktive 0,96# das aktive 1,3# Oled und das Nokia Display.
Das weiße Oled war schon woanders länger im Einsatz , deshalb ist es schon ziemlich eingebrannt und nicht mehr schön ablesbar. Das sollte beim Nokia besser sein, bei bedarf kannst du das auch noch beleuchten. 
Das benötigt dann 8,3 mA Strom bzw.  nimmt 0,024 Watt elektrische Leistung auf, pro Tag sind das dann 0,65 Wh.
[image]

BastyESP, 2023-02-19:
Hallo Loetnase, Danke für deine Bilder, aber habe die 16x2 LCD hier eben rumliegen, in Invertiert mit weißer Hintergrundbeleuchtung, die sind eigentlich recht gut Abzulesen. Von den 0,96er Oled habe ich auch noch ein paar da, nun noch das Nokia Display kaufen? Weiß nicht so richtig, ob das Sinn macht. Wusste ja nicht wie Aufwendig das einfügen eines solchen Display über die i2c Schnittstelle ist, hatte einfach die Hoffnung das es jemand einbauen kann, daher hatte ich die Anfrage erstellt. Aber bevor ich noch ein Nokia Display kaufe, würde ich wohl die 0,96er Oled erstmal verbrauchen, wobei ich auf deinem Bild sehe, das die nicht die Beste Option sind um Dauerhaft etwas Anzuzeigen.
Aber evtl. findet sich ja noch ein Findiger Programmierer und baut es ein. Die Hoffnung stirbt bekanntlich zuletzt.
Schönen Sonntag noch, Gruß Bastian

closed 2024-08-16 by lumapu