lumapu/ahoy #694: PowerLimit-Doku unvollständig bzw. erschließt sich mir nicht

ntfrnd, 2023-02-18:
Im https://github.com/lumapu/ahoy/blob/main/User_Manual.md steht zum PowerLimit folgendes:

The AhoyDTU will publish on the following topics (aus der Tabelle zusammenkopiert):
`<TOPIC>/<INVERTER_NAME_FROM_SETUP>/ch0/#/PowerLimit`
tatsächlich heißt es aber inzwischen `<TOPIC>/<INVERTER_NAME_FROM_SETUP>/ch0/#/active_PowerLimit`, da ist das User-Manual noch nicht auf dem aktuellen Stand.
In der Ahoy 0.5.66 fehlt auch noch der Unterstrich, das wurde aber schon in einem anderen issue (https://github.com/lumapu/ahoy/pull/684) behandelt. 

Desweiteren steht dort "The AhoyDTU subscribes on three topics `<TOPIC>/ctrl/#`, `<TOPIC>/setup` and `<TOPIC>/status.`"
Das `<TOPIC>/ctrl/#` ist gut beschrieben, habe ich angelegt und funktioniert soweit!

Allerdings finde ich nichts über `<TOPIC>/setup` und `<TOPIC>/status` dokumentiert. Auch legt mein AhoyDTU beim Verbinden dies nicht an. Muss ich das selbst tun? Ich dachte, es sind Status-Meldungen, die Ahoy published und im Broker automatisch angelegt werden, wie alle anderen (bis auf ctrl auch)?

Nächste Frage: Gibt es eigentlich das `<TOPIC>/devcontrol/` aus früheren Versionen noch, oder ist das durch `<TOPIC>/ctrl/` ersetzt worden?

Danke!

lumapu, 2023-02-19:
danke für die Hinweise,

devcontrol gibt es nicht mehr.

ntfrnd, 2023-02-19:
Danke für die Antwort. 

Kannst Du noch was zur Frage in der Mitte sagen bzgl.
`<TOPIC>/setup` und `<TOPIC>/status` ?

lumapu, 2023-02-19:
ja, die sind quasi Abfallprodukte der JSON API.
über `setup` geht `set_time`, `sync_ntp`, `serial_utc_offset` und `discovery_cfg`

zu finden ganz unten hier: https://github.com/lumapu/ahoy/blob/main/src/web/RestApi.h

drahdiwaberl, 2023-07-07:
ich seh nach update auf 0.7.8 weder auf /active_PowerLimit noch auf  /PowerLimit Daten über MQTT kommen..

ich darf im übrigen anmerken dass auch das <TOPIC> in der  Dokumentation eher unglücklich bzw. falsch ist - ein "topic" bei MQTT ist der gesamte string/"pfad" .. ich würd mich da an tasmota halten, dort heisst der definierbare Teil den ihr meint <PREFIX> afair

und auch die # wildcard in den Headlines ist eher irreführend für Anfänger- @ntfrnd hats ja dann im ersten post hier gleich mal falsch gezeigt - ein topic mit /#/ kann es nicht geben, /# muss wenn dann am Ende stehen..

wenn mir wer verrät wo sich das powerlimit jetzt versteckt werd ich mir die mqtt-doku vielleicht mal zur Brust nehmen ;)

drahdiwaberl, 2023-07-07:
zumindest laut https://github.com/lumapu/ahoy/blob/dbb352801e9076f50679efd0a7f432b9fa34dbd4/src/hm/hmDefines.h#L29 scheint es wohl schon noch active_PowerLimit zu sein, aber irgendwie bekommt das wohl keine Daten..

ntfrnd, 2023-07-07:
Was heißt hier, ich habe es falsch gezeigt? 
Ich habe den Unterschied zwischen Doku und Realität aufgezeigt. Für die aktuelle Umsetzung kann ich nix 😀

drahdiwaberl, 2023-07-08:
> @drahdiwaberl Was heißt hier, ich habe es falsch gezeigt?

"falsch" = so wie es die Doku einen glauben machen könnte - was aber eben nicht korrekt ist.

egal ob active_PowerLimit oder PowerLimit, dieses topic kann nicht funktionieren:

`<TOPIC>/<INVERTER_NAME_FROM_SETUP>/ch0/#/active_PowerLimit`

weil die multilevel wildcard (#) nur [am Ende eines topics stehen kann](https://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices/#mqtt-wildcard--multi-level).

lumapu, 2023-09-16:
das ist mittlerweile ganz anders gestaltet und dadurch für mich gefixt:
https://github.com/lumapu/ahoy/blob/main/User_Manual.md#control-via-mqtt

closed 2023-09-16 by lumapu