lumapu/ahoy #695: Nokia 5110 Display geht zeigt abends nichts mehr an

Loetnase, 2023-02-18:
### Platform

ESP8266

### nRF24L01+ Module

round dot

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.88

### Github Hash

fe54599

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

{"wifi":{"ssid":"XXXXX","pwd":"","dev":"BK3","adm":"","prot_mask":61,"ip":"","mask":"","dns1":"","dns2":"","gtwy":""},"nrf":{"intvl":30,"maxRetry":5,"cs":15,"ce":2,"irq":0,"pwr":1},"ntp":{"addr":"192.168.178.1","port":0},"sun":{"lat":48.84799957,"lon":8.923999786,"dis":true,"offs":0},"serial":{"intvl":5,"show":true,"debug":true},"mqtt":{"broker":"","port":1883,"user":"","pwd":"","topic":"inverter","intvl":0},"led":{"0":255,"1":255},"plugin":{"disp":{"type":1,"pwrSafe":false,"logo":true,"pxShift":true,"rot180":false,"wake":0,"sleep":0,"contrast":60,"pin0":5,"pin1":4}},"inst":{"en":false,"rstMidNight":false,"rstNotAvail":false,"rstComStop":false,"iv":[{"en":true,"name":"BJ","sn":18973052646805,"yield":[15,15,0,0],"pwr":[380,380,0,0],"chName":["Modul Rechts","Modul Links","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]}]}}

### Debug Serial Log output

```shell
123
```

### Error description

@knickohr, @stefan123t 
eine Frage an die Nokia 5110 Display User.
Abends wenn der Wechselricher offline geht schaltet sich das Nokia Display nach kurzer Zeit ab.
An der zweiten DTU mit Oled und gleichen Einstellungen ist das nicht so, da läuft die Anzeige mit Pixelshifting usw. schön die ganze Nacht durch, morgends werden dann die Werte zurückgesetzt wie gewünscht.
Werden da beim Nokia, wenn der Wechselricher offline ist, irgendwann Resets an die Anzeige geschickt?
Ich kann dann einen Reset am Wemos ausführen, es kommt dann immer die normale Bootanzeige zeigt eine Weile offline an und dann geht das Display aus.
Sobald ich aber die Jumperleitung vom Ausgang D0 ausstecke und auf 3,3V (logisch 1) stecke, dann Reset ausführe funktioniert die Anzeige wie gewünscht die ganze Nacht vergleichbar zum Oled.
Ich denke die Resetleitung hat schon ihren Sinn an D0, ein passender Reset wird schon manchmal nötig sein. 
Können andere User dieses Verhalten bestätigen oder ist das nur bei mir so. 
Ist das ein Bug oder gewünscht?   
[image]

knickohr, 2023-02-18:
Das sollte es eigentlich nicht machen, außer Du hast das „ Turn off while inverters are offline“ aktiviert. Ich habe allerdings seit geraumer Zeit das Nokia mit dem 8266 (wegen Waldi“. Ichs mehr im Betrieb.

Recommended Settings für das Nokia :
[image]

Loetnase, 2023-02-18:
Ich denke das ist richtig eingestellt.
Aber vielleich sollte ich mal deine Einstellung probieren. Probieren kostet ja nix :-)
[image]

knickohr, 2023-02-18:
Probier’s auch mal andersrum. Wir hatten schon den Fall da war an=aus und umgekehrt 🤪

Reboot nicht vergessen, viele Optionen ziehen erst beim Boot.

Loetnase, 2023-02-21:
Ich denke da hat sich beim Updaten oder so ein Bit veklemmt gehabt.
Beim ersten Versuch mit dem Setup von @knickohr hat es dann sofort wieder funktioniert.
Dann habe ich alle einstellbaren Möglichkeiten eingestellt aber der Fehler trat nicht mehr auf.
Mit „Turn off while inverters are offline“ aktiviert war es dann wieder so wie mit "meinem Fehler".
Jetzt funktioniert es wieder und somit ist es gut.
Danke für die Hilfe.

closed 2023-02-21 by Loetnase