lumapu/ahoy #696: MQTT subscribe Topics kommt 2x

HFarryranz, 2023-02-18:
### Platform

ESP8266

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.89

### Github Hash

b6a4150

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
  
### Inverter
#### Inverter 0
     - Name: HM-600
 
 

### Error description

Die Serielle Console im WEBIF meldet mir das doppelt (war aber auch schon in Vorgängerversionen):

17:11:07 I: MQTT connected
17:11:08 I: MQTT got topic: inverter/ctrl/limit_persistent_relative
17:11:08 I: MQTT got topic: inverter/ctrl/limit_persistent_absolute
17:11:08 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_relative
17:11:08 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_absolute
17:11:08 I: MQTT got topic: inverter/setup/set_time
17:11:08 0
17:11:08 I: MQTT got topic: inverter/setup/sync_ntp
17:11:08 0
17:11:21 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_absolute
17:11:21 I: MQTT got topic: inverter/ctrl/limit_persistent_absolute
17:11:21 I: MQTT got topic: inverter/ctrl/limit_persistent_relative
17:11:21 I: MQTT got topic: inverter/ctrl/limit_nonpersistent_relative
17:11:21 I: MQTT got topic: inverter/setup/set_time
17:11:21 0
17:11:21 I: MQTT got topic: inverter/setup/sync_ntp
17:11:21 0

beegee3, 2023-02-18:
das hängt m.E. auch mit https://github.com/lumapu/ahoy/issues/692#issuecomment-1435739211 zusammen (Einstellungen MQTT Broker).

lumapu, 2023-02-19:
ich habe noch keine gute Idee dafür, da es nur ioBroker bisher betrifft. Man kann nicht einfach generell die erste empfangene Message pro topic verwerfen.

beegee3, 2023-02-19:
so wie mein Lösungsvorschlag ist, werden Messages nur innerhalb der ersten 5 Sekunden nach einem MQTT Connect ignoriert. Die 5 Sekunden kannst du ja nach deinen ioBroker Erfahrungen anpassen, vielleicht reichen 2 Sekunden. Jedenfalls halte ich das für besser, als ein [generelles Limit 0 Verbot](https://github.com/lumapu/ahoy/issues/692#issuecomment-1435946016). Insbesondere da auch die unsinnigen 'set time' und 'sync NTP' Messages unterdrückt werden. Evtl. sind es auch diese Messages nach einem MQTT Connect, die zu der Connect - Disconnect Endlosschleife und letztendlich einem Reboot führen.

lumapu, 2023-02-21:
@beegee3 weißt du wer den Fehler macht? Ahoy oder ioBroker? Ich nehme mal ioBroker an, da es mit mosquito geht.

Ich empfinde die 5s Unterdrückung als einen Workaround und keinen fix.
Das Limit 0 Verbot wäre auch ein Workaround.

beegee3, 2023-02-21:
ich glaube das ist ein Design Problem. Eigentlich soll mit den Subscriptions etwas analog zu den Web Commands erzeugt werden. Die Subscriptions haben aber in der Regel Doppelfunktionen. Einerseits 'Ist' Zustände anzeigen, andererseits 'Wunsch' Einstellungen aufgrund von User Änderungen. Bsp. Lichtschalter oder Heizungsregler.
Mosquitto und ioBroker haben offensichtlich verschiedene Konzepte, was Messages zu Subscriptions im Fall von Verbindungsaufbau (Client Anmeldung oder Server Neustart) angeht. Das von ioBroker ('Ist' Werte melden) macht eigentlich viel Sinn, das von mosquitto (nichts melden) ist z.Zt. Ahoy freundlicher.

Bei Ahoy sollen die Subscriptions im momentanen Design nur 'Wunsch' Einstellungen sein. Damit braucht man Defaults, die anzeigen, dass keine Anpassung gewünscht ist. Momentan ist das 'null', geeignet wäre z.B. -1. In der `onMessage` Routine muss man auf Default prüfen und darf diese nicht an die Api weiterleiten (s. [hier](https://github.com/lumapu/ahoy/issues/692#issuecomment-1437219266)). Wenn der User einen Wert übermittelt, sollte nach der Verarbeitung der Default wiederhergestellt werden. Einerseits wäre das eine Art Empfangsbestätigung, andererseits wäre es sonst irreführend (z.B. Vorgabe: limit relativ = 90%, später limit absolut 100W. Man erkennt in den Topics nicht, was nun aktuell gilt. Aber auch bei set_time ist die angegebene Zeit schon 1 Sekunde später 'Schnee von gestern'). Und bei ioBroker würden bei einem Server Neustart wieder Probleme entstehen.
Der 'Ist' Wert des vom Inverter gemeldeten Limits sollte ein publish in einem eigenen Subtopic haben.

Auch eine Design Anpassung wäre zu überlegen. Gleichzeitige Vorgaben von limit relativ und limit absolut machen keinen Sinn.
Man könnte mit nur einem limit Wert, der entweder '%', oder 'W' als letztes Zeichen hat, auf beides reagieren. Default z.B. 'n/a' (analog zu Web Live).
Letztlich braucht der User noch eine entsprechende Dokumentation, um Fragen in Form von neuen Issues zu vermeiden 😄

lumapu, 2023-02-22:
ok dann wird hier ein Feature Request draus =)

beegee3, 2023-02-23:
> ok dann wird hier ein Feature Request draus =)

da die limit Einstellungen im Broker erhalten bleiben, erscheint mir das zwingend notwendig. Sonst kann bei einem Reboot (MQTT Server oder Ahoy) ein falsches limit gesetzt werden, s. Bsp. oben.
Noch ein Schritt weiter: 'p' für persistent voranstellen, dann braucht man für limits nur ein (!) topic. Persistent sollte ja nur spärlich eingesetzt werden. Auf 'n' für non-persistent würd ich verzichten, ließe sich aber auch variabel gestalten.
Entsprechend können auch die Zeiteinstellungen zusammengefasst werden, sync_ntp ist dasselbe wie set_time mit dem Wert 0.

Unabhängig davon solltest du 'reset to default' nochmals überlegen.
Es wird doch immer wieder darauf hingewiesen, persistent limits möglichst selten zu setzen, um den Inverter zu schonen.
Hab' keinen ioBroker zum ausprobieren, aber wenn ich das richtig verstehe, würde ein eingestelltes persistent limit bei Reboots immer wieder übertragen. Gleiches gilt für eine mit set_time gesetzte Zeit, was unweigerlich zu falschen Zeitangaben führt.
Ich wiederhole mich hier gerne: die Werte mit Default -1 belegen (vor `subscribe`), in `onMessage` Werte < 0 abfangen (machen nirgendwo Sinn) und alle anderen Werte nach Erzeugung von `root`, d.h. direkt vor dem Aufruf von `(mSubscriptionCb)(root);` durch ein publish wieder auf den Default setzen. Löst eine neue Message aus, die aber dann ignoriert wird.
Auch beim non-persistent limit ist ein 'reset to default' vernünftig, denn von einem zum nächsten Tag wird das vom Inverter gelöscht und der Broker würde auch hier eine falsche Einstellung suggerieren. Das aktuelle limit (was der Inverter meldet!), gehört in ein eigenes topic.

lumapu, 2023-03-08:
in neueren Development Versionen behoben, spätestens ab `0.5.93`

closed 2023-03-27 by lumapu