lumapu/ahoy #700: [ESP8266] Schlechte bis keine Verbindung bei hoher Einspeisung HM-1500

DReffects, 2023-02-19:
### Platform

ESP8266

### Model name

WeMos D1 MINI MICRO USB

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 123456789
    - Name: HM1500
    - Active Power Limit: variable
    - Active Power Limit Control Type: via MQTT
    - Max Module Power (Wp): 4x400
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 192.168.XX
    - Port: 1883
    - Username (optional): XXX
    - Password (optional): XXX
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

### Error description

Habe ein seltsames Problem mit einem HM-1500. Kommunikation mit dem Inverter lief problemlos für einige Tage während ich lediglich zwei der vier Eingänge beschaltet hatte.

Seit ich alle vier Eingänge des WR belegt habe, bringt die Kommunikation zwischen AHOY DTU und dem HM-1500 zusammen, wenn die Einspeiseleistung etwa 900 Watt erreicht. 

Die DTU liegt ~1m vom Inverter entfernt an einem 2Ampere Netzteil, Kondensatoren sind 2x verbaut (1x 820 uF 35V am 3,3V Strang und einmal bei 5V)

Kann ich dies irgendwie debuggen? :D

lumapu, 2023-02-22:
Evtl. mal die Ahoy-DTU nicht auf der selben Phase betreiben? Testen ob sich das auch am Laptop / Powerbank so verhält

DReffects, 2023-03-08:
> Evtl. mal die Ahoy-DTU nicht auf der selben Phase betreiben? Testen ob sich das auch am Laptop / Powerbank so verhält
Danke für die Anregung, teste ich!

DReffects, 2023-07-02:
Leider bislang keine Veränderung feststellbar.

Habe nun einen Weiteren HM-1500 hier (auch andere Firmware Version). Selbes Problem.

Habe mittlerweile getestet:

DTU mit Wemos D1 und billiger Antenne
DTU mit Wemos D1 und externer Antenne
DTU mit ESP32 und externer Antennte

jeweils mit und ohne Kondensator.

Betrieb über Powerbank, USB Port an einem PC und USB 2A Netzteile an verschiedenen Phasen.

Immer selbes verhalten:
0-20% beim HM-1500: Super Empfang, alle 5 Sekunden werte wie ein Uhrwerk
Je mehr es wird, desto sporadischer wird es.
bei 100% (also 1500W) nur noch alle 30-120 Sekunden ein Udpate

Senden von Einspeiselimit funktioniert jedoch problemlos

HM-300 zeigt das Problem nicht.

lumapu, 2023-08-05:
es gibt inzwischen die Version `0.7.25`, die sollte gegenüber der genannten `0.5.66` signifikante Verbesserungen bringen.

DanielR92, 2023-08-26:
Hallo zusammen, da hier das Problem im Feb/Juni aufgekommen ist und seit drei Wochen kein Update von @DReffects gab. Close ich das mal hier.

DanielR92, 2023-08-26:
Hallo zusammen, da hier das Problem im Feb/Juni aufgekommen ist und seit drei Wochen kein Update von @DReffects gab. Close ich das mal hier. - Bitte neu öffnen wenn sich doch noch was geändert hat. :)

closed 2023-08-26 by DanielR92