lumapu/ahoy #702: Version 0.5.66 MQTT logout

KLM30330, 2023-02-21:
## Hardware
  * [x] ESP32

Modelname: ESP-WROOM-32

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x] external antenna

### Power Stabilization:
None

### Connection diagram:

### Connection diagram I used:
| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  **D16**  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU

### Version / Git SHA: 
Version: 0.5.66

### Build & Flash Method:
* [x] ESP Tools

### Desktop OS:
* [x] Windows

### Debugging:
* [x] Setup settings (use our templates ... to be added)

## Problem
Hi,
I use AhoyDTU with home assistant and since about one month I lost MQTT connection.
I have this message in serial control :

`06:30:24 I: MQTT disconnected, reason: TCP disconnect`

Is there a solution ?

Thanks in advance

Argafal, 2023-02-21:
I have seen this with 0.5.66, too. I have not seen it occur in the latest development version yet. So it might not longer be an issue once the next release comes out.

KLM30330, 2023-02-21:
Hi, witch development version do you used and how did you installed it?

beegee3, 2023-02-21:
you find a zip file with the latest Development Build on https://ahoydtu.de/web_install/
Please download and unzip it then use the AHOY update web surface (if you didn't change the setting it's http://ahoy-dtu/update), select the right bin file (depending on your ESP it ends with _esp8266.bin or _esp32.bin) klick on UPDATE and see what happens (message should be: update started -> Update: success  rebooting ... auto reload after 20s) -> dev. version -> 😃

Argafal, 2023-02-21:
I used 0.5.89. @beegee3 answered how to install it, thanks to them! :)

closed 2023-03-27 by lumapu