lumapu/ahoy #704: Kein Datentransfer zwischen Ahoy und WR

cibis8, 2023-02-21:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [X ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [X ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

cibis8, 2023-02-21:
Was mir noch fehlt ist ein Elko 100uF, aber der sollte nur für eine Spannungsstabilisierung auf der WLAN Seite beitragen. Ist unterwegs vom ALI ;) leider hab ich auf die schnelle keinen bei mir gefunden. Bin schon bissl am "Verzweifeln" und hoffte hier im Forum etwas zu finden (auch mit dem Vertauschen von D3 und D4).

martinriedel, 2023-02-21:
Interessant für mich wäre noch was steht unter `System `-> `Radio`, also RX success, RX fail usw
und schau mal unter `Serial / Control`, da wird die Kommunikation gezeigt, lass das mal eine Minute offen und kopiere den Inhalt hier rein.
Ich weiß natürlich nicht ob `is connected` auch bei den falschen NRF Modulen angezeigt wird

cibis8, 2023-02-21:
hier mal System -> Radio: 
mac
DC:4F:22:2A:08:ED
sdk
2.2.2-dev(38a443e)
cpu_freq
80
heap_free
17728
sketch_used
486
version
0.5.66
wifi_rssi
-77
ts_uptime
2201
esp_type
ESP8266
core_version
3.0.2
flash_size
4096
heap_frag
18
max_free_blk
7552
reboot_reason
External System
Radio
nrf24l01+
is connected
Datarate
250 kbps
Power Level
LOW
RX success
0
RX fail
0
RX no answer
73
RX frames received
0
TX count
438

cibis8, 2023-02-21:
und ein Mitschnitt der Serial Console:
15:08:00 I: resetPayload: id: 0
15:08:00 I: (#0) enqueuedCmd: 1
15:08:00 I: (#0) enqueuedCmd: 11
15:08:00 I: (#0) enqueuedCmd: 5
15:08:00 I: (#0) sendTimePacket
15:08:00 I: sendTimePacket 1
15:08:00 I: TX 27B Ch40 | 15 84 23 93 83 82 75 47 97 80 01 00 63 f4 d0 40 00 00 00 00 00 00 00 00 1d 0c 12 
15:08:01 W: while retrieving data: last frame missing: Request Retransmit
15:08:01 I: (#0) sendTimePacket
15:08:01 I: sendTimePacket 1
15:08:01 I: TX 27B Ch61 | 15 84 23 93 83 82 75 47 97 80 01 00 63 f4 d0 40 00 00 00 00 00 00 00 00 1d 0c 12 
15:08:03 W: while retrieving data: last frame missing: Request Retransmit
15:08:03 I: (#0) sendTimePacket
15:08:03 I: sendTimePacket 1
15:08:03 I: TX 27B Ch75 | 15 84 23 93 83 82 75 47 97 80 01 00 63 f4 d0 40 00 00 00 00 00 00 00 00 1d 0c 12 
15:08:05 W: while retrieving data: last frame missing: Request Retransmit
15:08:05 I: (#0) sendTimePacket
15:08:05 I: sendTimePacket 1
15:08:05 I: TX 27B Ch3 | 15 84 23 93 83 82 75 47 97 80 01 00 63 f4 d0 40 00 00 00 00 00 00 00 00 1d 0c 12 
15:08:07 W: while retrieving data: last frame missing: Request Retransmit
15:08:07 I: (#0) sendTimePacket
15:08:07 I: sendTimePacket 1
15:08:07 I: TX 27B Ch23 | 15 84 23 93 83 82 75 47 97 80 01 00 63 f4 d0 40 00 00 00 00 00 00 00 00 1d 0c 12 
15:08:10 W: while retrieving data: last frame missing: Request Retransmit
15:08:10 I: (#0) sendTimePacket
15:08:10 I: sendTimePacket 1
15:08:10 I: TX 27B Ch40 | 15 84 23 93 83 82 75 47 97 80 01 00 63 f4 d0 40 00 00 00 00 00 00 00 00 1d 0c 12

cibis8, 2023-02-21:
Habe mir noch 2 von den RF Modulen mit aufgeätzer Antenne bestellt (kommen aber erst in 2-3 Tagen).Ich hatte 4 dieser RF mit ext Antenne und Verstärker geholt. Wollte diese DTUs für mich, meine Mutter und die Verwandtschaft selbst bauen und nicht auf die teurere Lösung von Hoymiles gehen.

martinriedel, 2023-02-21:
Es sehe jetzt auch nichts was falsch sein könnte. Allerdings habe ich manchmal das Problem dass der Wechselrichter keine Verbindung aufbauen mag. Dann hat nur ein geholfen den Stromstecker ziehen, warten und wieder einstecken geholfen

cibis8, 2023-02-21:
Also bei mir kam es noch zu überhaupt keinem Transfer. Vielleicht habe ich ja auch beim Löten irgendwas falsch gemacht oder zerstört.. hätte noch 3 weitere Versuche ;) ich dachte nur, weil du geschrieben hattest, dass man in der Config noch irgendwelche Werte ändern müsste zu D3, D4. Echt schade wenn es mit den RF Modulen mit ext. Antenne Probleme macht, da ich schon eine höhere Reichweite bei wenigstens 2 der PV Balkonkraftwerken benötigen würde.

Argafal, 2023-02-21:
Der Elko hat mit dem WLAN nichts zu tun. Der ist dafür da, dass das NRF24 Modul stabil mit Strom versorgt wird. Das kann also schon wichtig und ggf. die Problemursache sein.

Mach doch auch mal Fotos und/oder eine Skizze von deiner Verkabelung. D3 und D4 musst du in den Settings so einstellen wie sie im echten Leben tatsächlich bei dir angeschlossen sind.  Der Tipp probeweise D3 und D4 zu tauschen heißt dann natürlich auch, sowohl die Hardwareseite zu tauschen als auch die Settings. Bei mir ging's übrigens mit D4 und D3 auch nicht, ich bin dann auf D1 und D2 (Hardware und Software) und alles ging.

martinriedel, 2023-02-21:
@cibis8 wie hoch muss die Reichweite sein?

martinriedel, 2023-02-21:
Ich habe bisher keinen einzigen Kondensator verbaut und bisher auch keine Probleme mit der Verbindung gehabt. Allerdings ist der WR auch nur ca. 8 Meter entfernt

cibis8, 2023-02-21:
Aktuell sind es (Testaufbau) ca 3-4m. später werden es dann ca 10-12m durch 2 Wände (Garagendach und eben Hausmauer. Ich möchte beide WR für 2 Wohnungen unter dem Schrägdach der Garage montieren (dort ist auch eine kleine Werkstatt untergebracht mit extra Absicherung, glaube sogar ein FI. Fotos bzw. Skizzen muss ich später machen.. kann ich dann hier mal einfügen. Ich dachte nur, es ist relativ einfach mit dem Ahoy, ist wohl leider doch net so (ausser der Fehler liegt bei mir ;)

Argafal, 2023-02-21:
> Ich habe bisher keinen einzigen Kondensator verbaut und bisher auch keine Probleme mit der Verbindung gehabt. Allerdings ist der WR auch nur ca. 8 Meter entfernt

Da hattest du Glück. Auf dem Discord gibt's jede Menge Erfahrungsberichte von Leuten, bei denen die Kommunikation erst in dem Moment (stabil) lief, als ein Kondensator am NRF24 angelötet wurde. Deswegen ist das auch oft die allererste Empfehlung. Hoffe, es hilft @cibis8 sein Problem zu lösen. :)

cibis8, 2023-02-21:
Leider hängen die Dupont (Steck)Kabel ebenso wie die Elkos noch irgendwo zwischen China und DL fest. sons hätte ich dass auch schnell mal umstecken können.. aber wie du ja geschrieben hast, in der Config muss das dann auch angepasst werden und da scheitert es dann (bei mir)

cibis8, 2023-02-21:
> als ein Kondensator am NRF24 angelötet wurde. Deswegen ist das auch oft die allererste Empfehlung. Hoffe, es hilft @cibis8 sein/ihr Problem zu lösen. :)

Das hoffe ich auch mal, des ich das Ahoy dann zum Funktionieren bringe... bin übrigens M und Name ist Uwe.

martinriedel, 2023-02-21:
@Argafal dann keine Ahnung was ich anders mache als die Anderen :D
@cibis8 ich kann dir testweise mal meine Komponenten zur Verfügung stellen die definitiv funktionieren wenn du willst, schick mir eine Mail an donmarten@googlemail.com

Argafal, 2023-02-21:
@martinriedel Ich glaube es ist eine große Komponente Zufall drin, in der Elko-Frage. Für manche geht's ganz ohne (da war ich auch dabei), für manche gibt's ohne Elko nur unstabil/manchmal Kommunikation, und für viele geht's nur mit Elko richtig gut. Schaden tut der Elko aber nie. Daher das dringende Gebot, das als mögliche Problemursache als erstes zu eliminieren.

cibis8, 2023-02-21:
> ich kann dir testweise mal meine Komponenten zur Verfügung stellen die definitiv funktionieren wenn du willst

Danke martinriedel für das Angebot. Ich denke das sollte schon laufen, nur bei mir tut es aktuell nicht und es nervt schon ein wenig ;) Ich versuche mal heute noch einen 100uF bei einem Bekannten zu besorgen.. wenigstens einen und vielleicht habe ich ja Glück und es funktioniert damit, wenn nicht gehe ich mal heute Abend an den 2. Aufbau. Hoffe ja nicht, dass der WR eine Knacks hat bei seinem Sendemodul.

cibis8, 2023-02-21:
Ja das hatte ich gelesen., Sorry wenn ich da fragen muss.. wie kann ich aber das Pinlayout dann softwareseitig anpassen?
Den D1 mini v3.0.0 habe ich... na mal schauen ob der Elko hilft... werde das dann wieder so umlöten wie es am Anfang war:
D4 auf CE und D3 auf IRQ.

martinriedel, 2023-02-21:
Auf der Weboberfläche unter `Settings `-> `System Config`

cibis8, 2023-02-21:
auch hier das gleiche: Inverter #0: HM800UR (v0) is not yet available  :(

cibis8, 2023-02-21:
Hurra... Erfolg! .. irq (gelb) auf pin 1 gelegt und in der System Config angepasst.. funktioniert... da muss irgendwas nicht passen mit dem D3/D4... Störung vielleicht weil nebeneinander?

JuKi1994, 2023-02-22:
Hallo zusammen,

bei mir hatte die Verkabelung zuerst auch nicht funktioniert, ich habe den CE und IRQ Anschluss beliebig zwischen D1-D4 variiert und die System Config entsprechend angepasst...funktioniert hat es dann aber erst mit RX (GPIO3) und TX (GPIO1).

```
+-----------+          +-----------+
|  ESP8266  |--colour--| nRF24L01+ |
|           |          |           |
|       GND |---black--|[GND]      |
|     +3.3V |----red---| VCC       |
|        RX |---grey---| CE        |
|        D8 |--purple--| CSN       |
|        D5 |---blue---| SCK       |
|        D7 |---green--| MOSI      |
|        D6 |---brown--| MISO      |
|        TX |--yellow--| IRQ       |
+-----------+          +-----------+
```

[image]

Vielleicht hilft das auch wem weiter...
VG,
Julian

cibis8, 2023-02-22:
> Vielleicht hilft das auch wem weiter...
> VG,
> Julian

Vielen Dank für den Tip. Aktuell funktioniert alles perfekt. Deshalb lass ich das mal so laufen und werde bei der nächsten Verkabelung es laut deinem Vorschlag durchführen.

rolliracker, 2023-03-06:
> funktioniert hat es dann aber erst mit RX (GPIO3) und TX (GPIO1).

Kann ich bestätigen. Vorher mit CE und IRQ auf D4 und D3 ständige Unterbrechungen der Kommunikation (manchmal nach mehreren Stunden, manchmal schon nach ein paar Minuten). Seit der Änderung auf RX und TX vor 3 Tagen keine Unterbrechung mehr. (Einen Kondensator hab ich noch nicht verbaut. Amplifier Power Level: High).

Vielen Dank für den Tip!

bernd63571, 2023-03-11:
Ich bin neu in dem Thema, daher bitte um Entschuldigung, wenn ich irgendwo ne offene Tür einrenne. Ich habe leider mit meinem Bastelprojekt ähnliche Probleme. Ich kann Daten empfangen, sieht ähnlich aus wie oben und ich denke, die sehen auch recht gut aus.
Mal ein Beispiel, empfangen habe ich:
13:36:41 I: Payload (42): 00 01 01 5c 00 8e 01 ee 01 5d 00 94 02 07 00 00 33 0d 00 00 36 2b 01 83 01 a2 09 1f 13 8a 03 c7 00 01 00 29 03 e8 00 88 00 0d, dazu bekomme ich dann eine Fehlermeldung, dass 30 Byte erwartet werden.
Die Daten machen nach meinem Eindruck durchaus Sinn, ich würde daraus lesen:
U1=34,8V; I1=1,42A; P1=49,4W; U2=34,9V; I2=1,48A; P2=51,9W;Kollektor 1 hat 13,069kWh und Kollektor 2 hat 13,867kWh geliefert, heute kamen 0,387 bzw. 0,418kWh Ertrag rein, Netzspannung 233,5V bei 50,02Hz und 96,7W Leistung; 0,41A Stromabgabe, Innentemperatur des Wandlers 13,6 Grad. Die Seriennummer des Invertes erscheint korrekt in den Datenpaketen, das sieht alles vernünftig aus. Einige Bits scheinen Status-Bits zu sein '03E8' erscheint, wenn der HM600 normal Energie liefert, bei schlechtem Licht, also ohne Leistungsabgabe wird das dann '03E9' und dann am Ende '0000'. Die Daten aus der Payload(42) finden sich in den empfangenen Datenpaketen, das sieht alles vernünftig aus.
Wenn ich aber anstelle der stabilen Version von der Internet-Seite die aktuelle Developers draufspiele, dann wird gleich gar nichts empfangen.
Kondensator am Empfänger ist verlötet, auch D3/D4 habe ich schon geändert, es ändert sich nichts. Hat jemand ähnliche Erfahrungen und vielleicht einen Tip? Gibt es irgendwelche Probleme mit bestimmten Serien des 8266?

PapaDox, 2023-03-11:
Hi Leute, 
nach unzähligen Stunden rumprobierens, bin ich heute auf diesen Thread gestoßen!
Ich habe den Wemos seit Tagen bearbeitet. Ich habe 2 HM-600. Den aktiven am BKW, den konnte ich nach Zusammenbau nicht abfragen, aber einen Test HM-600, da ging es auf einmal, mit der Standard Konfig vom Ahoy-DTU. Also ging ich davon aus, dass die SN des aktiven nicht korrekt sei. Da habe ich einfach den funktionierenden ausgetauscht. Obwohl es 1x funktionierte, ging es dann in der Praxis wieder nicht. Ich habe den nrf24l01+ mit Antenne dran. Ich sitze ca. 3m neben dem WR, aber trotz der kurzen Strecke lief es nicht. Das tauschen von D3/4 und CE/IRQ hin und her auch auf D1/2 funktionierte nicht. Auf die  letzte DEV Version brachte keine Änderung (da ging übrigens der SAVE button nicht). Dann habe ich mir einen 2. Wemos gelötet und das gleiche Bild wieder nicht. Anschließend diesen Thread gefunden und erstmal den 100yF Kondensator dran gelötet und damit geht es, selbst mit der korrekten D3/4 Konfiguration. Der HM-600 wird ausgelesen. Also habe ich meine 2. Antenne mit dem Kondensator versehen und den ersten Wemos angeschlossen, weil ich dachte es geht. Aber Fehlanzeige! Ging nicht. Dann fand ich einen der letzten Posts hier mit TX/RX, das habe ich dann umgesteckt und konfiguriert und siehe da, es geht. Das ist echt strange, einer geht mit Standard Konfig und Kondensatur und der andere ging trotz Kondensator erst, nachdem ich umgesteckt habe. Krass. Vielen Dank auf jeden Fall für die vielen Infos im Thread. Das erspart mir weitere graue Haare :-D

bernd63571, 2023-03-11:
Nutze ich RX und TX so funktioniert der Empfang gar nicht. Es zeigt sich keine Änderung, egal ob mit Original-Belegung D3/D4 oder D1/D2. Ich habe das auch mit unterschiedlichen Antennen ausprobiert. Bei der kleineren Antenne ist es sinnvoll den Pegel auf High umzustellen, da es bis zum Kollektor schon gut 20 m sind, das Modul mit der längeren Antenne läuft auch mit Pegel Low gut. Für die Daten der Payload(42) hab ich mir inzwischen mit OO Calc nen kleinen Decoder gestrickt, da kann man dann den Text einfach reinkopieren. Falls das wer haben will, stell ich das gerne ein. Mir ist es ein Rätsel, warum die Fehlermeldung kommt.

PapaDox, 2023-03-24:
Ich habe noch 2 Standard Antenne bestellt, also nicht so lange. Mir denen geht es aus Anhieb in der Standard config. Wirklich unglaublich. 

Ich wollte die Integration über den IoBroker machen. Habe da Daten gesetzt und seit dem komme ich an einen nicht mehr ran. Keine Ahnung warum es nicht geht. Wollte es reseten per USB und über ahoy Webseite, aber der USB Port wird nicht zur Auswahl gestellt. Kann ich über langes drucken des Reset Knopfes was erreichen? 

Habt ihr ggfs. die IoBroker Integration gemacht? Bin da nicht so Form. Habe bisher nur einen sonoff POWR2 mit tasmota integriert. Was habt ihr als Adapter/Instanz gewählt (sorry falls ich das falsche wording benutze, bin unterwegs und kann nicht nachschauen)? Danke und Gruß

closed 2023-03-25 by lumapu