lumapu/ahoy #706: Häufige Limit-Änderung unterbricht Datenübertragung von WR an Ahoy.

Parodontitis, 2023-02-22:
## Hardware
  * [X] ESP8266

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [X] external antenna

## Software
* [X] AhoyDTU

### Version / Git SHA: 
Version: 0.5.66
Github Hash: f8fe044

### Desktop OS:
* [X] Linux

2 x HM-600
Um die Abgabe an das Stromnetz unter 400W zu halten, steuere ich einen WR über die API und "limit_nonpersistent_relative". Das funktioniert ganz gut. Der neuer Limit-Wert wird nur bei einer Änderung an ahoy geschickt, was aber alle 30 Sekunden passieren kann. Nach einiger Zeit, empfängt der WR weiterhin die Limit-Änderungen und setzt sie auch brav um, ahoy zeigt jedoch: 

Inverter #0: HM-600 (v10010) is not yet available
-> last successful transmission: 22.2.2023, 09:04:51

Ich erhalte von dem WR keine Daten mehr. Ändert sich die Limit-Steuerung einige Zeit nicht, dann wird wieder kommuniziert und ahoy erhält wieder die Daten, die er über mqtt an IoBroker schickt.

Gibt es Erkenntnisse, wie oft die Limit-Änderung an den WR geschickt werden darf? Gibt es alternativen? Was könnte ich verbessern?

[image]
Rot "Leistung" oben max 125% unten 0%
Gelb "Stromverbrauch" in W - Negativer Bereich = Überproduktion=Abgabe an Stromnetz.

MubiTec, 2023-02-22:
Hatte das gleiche Thema gestern auch, da wurde mir empfehlen die Sendeleistung hochzu Schrauben auf high. Habs aber noch nicht ausprobiert, ob das Besserung bringt.

Ollipop030, 2023-02-22:
Ich speise immer ca 100 Watt ein, eine Änderung erfolgt im 17 Sekundentakt. Keine Probleme auf einem ESP32. Was ich aber hatte: Ein 8266 mit gleicher Einspeiseregelung hatte öfters Aussetzer, aber nur wenn ein zweiter WR in Ahoy hinterlegt war (habe HM 1500 und HM 300). Deaktiviere den zweiten WR mal, bei mir hatte das geholfen.

Parodontitis, 2023-02-22:
Das Abschalten des zweiten WR hat nicht geholfen. Ich werde den Wemos durch einen ESP32 ersetzen und testen. Werde berichten. Danke.

Ollipop030, 2023-02-22:
> Ich werde den Wemos durch einen ESP32 ersetzen

Ich habe mit mehreren 8266 schlechte Erfahrungen gemacht. Ich kann den Umstieg nur empfehlen.

reserve85, 2023-02-22:
Gut ist es auch dass du kein gemeinsamer Teiler als Intervall einstellst. Wenn du beispielsweise alle 5 sekunden den WR in ahoy abfragst dann stelle keine 20s ein zur leistungsanpassung.

KG3RK3N, 2023-02-22:
Kann ich bei mir mit einem ESP8266 ebenfalls bestätigen.
Hast du die aktuelle Dev schon getestet? Ggf. funktioniert es damit schon besser. Steht auch bei mir noch auf der To-Do Liste das damit Mal zu testen.

Parodontitis, 2023-02-23:
Die Leistungsanpassung ist bei mir eine Reaktion auf die Werte die vom Smartmeter kommen und waren auf 30 Sek. eingestellt. Der ESP sendet auch alle 30 Sek die Daten. Jetzt habe ich den Smartmeter auf 23 Sek. umgestellt, damit sie nicht synchron arbeiten. Danke für den Hinweis. Leider kann ich es bei heutigen Wetter nicht testen. Parallel dazu werde ich ich die DEV Version testen und ein ESP32 ist schon auf dem Weg zu mir. Danke.

schliepi, 2023-02-24:
Vielleicht die Stromversorgung? Alle Hänger könnten immer damit zu tun haben. Ein leichter Browout und schon geht was schief.
Zwei identisch aufgebaute und konfigurierte 8266 (mit 2x100µ Elkos) zur Kommunikation mit jeweils einem HM300 bei mir. Der eine (am MacMini-USB) hat eine Uptime von 26d, der andere (am 10W Apple-Netzeil) schwankt zwischen 3h und 2d.

lumapu, 2023-02-25:
@beegee3 Ich habe mal darüber nachgedacht bzgl. Intervall und Powerlimits. 
Was hälst du davon ein `next-command-struct` in einem Array einer fixen Größe (um Heap-Löcher zu vermeiden) zu verwalten. es sollte sich wie ein FiFo verhalten.

Ich könnte mir das ungefähr so vorstellen:
```
typedef struct {
    uint32_t ts;
    uint8_t cmd;
    uint8_t ivId;
} ivCmd_t;

mCmdQueue[50];
```

Der `tickSend` müsste dann diese Queue prüfen, ob schon voll, wenn nicht dann kann man alle Inverter hinzufügen. Wenn ein `devcontrolrequest` kommt, dann muss dieser sofort ausgeführt werden.
Der `ts` ist drin, um zB. eine 'last-loop-time' oder so zu ermöglichen. Ich weiß noch nicht, ob dieser Parameter sinnvoll ist.

reserve85, 2023-02-25:
Sollte aber besser nur jeweils ein Objekt je Inverter drin sein. Wenn ein "Limit-Request" in der Queue ist und aus welchen Gründen auch immer nicht abgearbeitet wurde, dann sollte doch am besten bei einem neuen eingehenden Befehl das vorherige Limit-Objekt gelöscht oder überschrieben werden. Mit anderen Worten: wenn ein Limit gesetzt wird sollten mMn. alle anderen Limit-Requests gelöscht werden. Sonst hängt das ggf. ewig nach.

beegee3, 2023-02-25:
@lumapu das Problem ist doch, dass limit Anweisungen vorrangig behandelt werden. Was zunächst auch vernünftig ist. Aber ein neues Limit zu setzen macht doch erst Sinn, wenn die letzte Änderung vom WR umgesetzt wurde. Ich halte es ansonsten für einen Anwendungsfehler, der in der MQTT Steuerung korrigiert werden sollte und unabhängig von Ahoy ist.
Wie schon an anderer Stelle gesagt, sollte das vom WR gemeldete limit ein eigenes topic in MQTT haben, so dass Anpassungen darauf Bezug nehmen können (`MQTT_ACK_PWR_LMT` reicht nicht und kommt dafür zu früh!). Dieses könnte zusammen mit den WR Daten übertragen werden.
Kommen limit Anweisungen trotzdem zu schnell hintereinander, bleibt das Problem natürlich bestehen. Was man braucht, ist dass zwischen limit Anweisungen mindestens einmal `RealTimeRunData_Debug` abgefragt wird. Ich sehe noch nicht, wie deine angedachte `next-command-struct` das löst.
Ich würde einfach ts von limit und von RealTimeRunData_Debug speichern (z.B. als `tsLimit `und `tsRTRData`). Eine neue limit Anweisung wird nur ausgeführt, wenn `tsLimit < tsRTRData` ist. Dann kann auch die von @reserve85 angesprochene Problematik
nicht entstehen. Ggfs. muss die neuste limit Anweisung wiederholt werden.

MubiTec, 2023-02-27:
wäre natürlich wirklich gut, wenn man das über die MQTT Steuerung abfangen könnte. bei mir ist jedoch das Problem, dass ich alle 15 Sekunden das Limit nachregel, was auch an sich gut klappt. Jedoch ist Ahoy damit so ausgelastet, dass ich keine Daten des WRs mehr bekomme, sodass ich leider nicht die Change habe, auszuwerten, ob das Limit senden denn geklappt hat.

dtuuser, 2023-02-27:
Bei mir auch. Ich schicke das Limit nach Bedarf alle 60s. Der WR nimmt die Änderung an, Ahoy bekommt aber erst nach ca.2 Minuten die neuen Werte. Wenn dazwischen ein neues Limit gesetzt wird, dann verzögert sich das alles. Da ich das Limit anhand der wirklichen Erzeugung berechne, rechnet er wenn ich Pech habe immer mit den alten Werten. Und setzt das Limit so immer weiter runter.

beegee3, 2023-02-27:
@MubiTec , @dtuuser 
fragt ihr die WR Daten im gleichen Rhythmus ab (z.B. mit dem Default 30 Sekunden) ab? Würde erklären, dass nichts kommt. Bei einer limit Anweisung wird eine gleichzeitig anstehende Datenabfrage ignoriert. Probiert mal, die Intervalle für limit Anweisungen und Datenabfragen so zu gestalten, dass sie möglichst nicht zur selben Zeit ausgelöst werden. In einem anderen 
Issue (weiß leider nicht mehr wo) wurden Primzahlen für die Intervalle empfohlen (da gibt's die wenigsten Überschneidungen).
@dtuuser ein im Verhältnis zum limit kürzeres Datenabfrageintervall sollte helfen. Bei mir funktionieren 5 Sekunden ohne Probleme, nutze gerade sogar 2 Sekunden für Tests an einer Radio Modifikation.

Ollipop030, 2023-02-27:
Ich glaube die Hoymiles müssen für einen komplette Reset AC und DC seitig getrennt werden, 10 Sekunden lang wenn ich nicht irre. Also entweder warten bis es dunkel ist, oder Module abdecken und dann die MC4 Stecker ziehen. NICHT UNTER SPANNUNG DIE STECKER ZIEHEN.

Parodontitis, 2023-02-27:
Ein "Restart" unter "Serial/Control" hat geholfen.

Ollipop030, 2023-02-27:
> Was genau wird bei dem Restart gemacht?

Ich denke, das vielleicht einmal intern AC und DC getrennt werden. Aber gut, das es so leicht ging. Hat man sich den Weg mit der Wolldecke aufs Dach gespart.

dtuuser, 2023-02-27:
Hatte ich heute auch für 2 Stunden. Ich habe das WR Abfrage Intervall auf 17 (Primzahl)  geändert.
Limit wird über ein Windows Programm und CURL Aufruf bei Bedarf alle 60 Sekunden gesetzt. Limitänderungen wurden alle sofort in Ahoy angezeigt, aber es wurden  nur noch 0 Watt produziert.  Dann habe ich das Limit direkt unter Ahoy auf Maximum gesetzt und das Intervall wieder auf 15 Sekunden, es lief ab da wieder normal. Achso,Version 5.66.

Ollipop030, 2023-02-27:
Ich kann immer nur wieder das Script von @reserve85 empfehlen:
https://github.com/reserve85/HoymilesZeroExport

Läuft bei mir tagelang, ohne einen Aussetzer. Ich ändere auch alle 17 Sekunden. Habs auf dem PiHole laufen, das packt der locker.

dtuuser, 2023-02-28:
Heute das gleiche Phänomen wie gestern. Nach mehrmaligem Limitändern hängt der WR bei 0W Einspeisung, obwohl das Limit mit mehr als 10% angezeigt wird. Das Setzen vom 100% Limit hat nichts  gebracht. Dieses Mal half nur ein Steckerziehen.
Jetzt habe ich mein minimales Limit auf 20W und nach dem Limitsetzen eine 2 Minuten Pause bis zum nächsten Limitsetzen eingebaut. Wenn ich mehrfach in Folge das Limit setze, dann zeigt Ahoy mit die Limit Änderung direkt an, aber die aktuellen Werte werden erst ca. 2 Minuten nach der Änderung neu empfangen. Könnte man in Ahoy eine direkte Abfrage der Werte nach einem Limitsetzen einbauen?

beegee3, 2023-02-28:
@dtuuser welche Ahoy Version nutzt du? Die direkte Abfrage des Werts nach einem Limitsetzen ist seit einiger Zeit in den development Versionen eingebaut.

dtuuser, 2023-02-28:
Produktiv die 05.66. Dann werde ich Morgen mal den zweiten Stick mit der 05.92 "scharf" stellen. Oder soll ich lieber auf die nächste Version warten?

reserve85, 2023-02-28:
ja du musst zwingend updated, das ganze Limit-Setzen hat sich mittlerweile immens verbessert. Kein wunder dass du da teilweise ewig warten musst...

dtuuser, 2023-03-01:
Konnte noch nicht umstellen. Aber bin ein Stück weiter. Bei einem Minimum Limit von 10 Watt, steigt die Spannung der Zellen auf über 42 Volt. Dann steigt die WR Temperatur und der WR steigt dann mit 0 Watt Produktion irgendwann aus. Heute mit einem Minimum Limit von 20 Watt, bleibt die Spannung bei unter 40 Volt und es werden wirklich dien20 Watt produziert.
Werde trotzdem spätestens am Freitag auf die 5.92 umstellen.

Parodontitis, 2023-03-03:
Das mit dem 20 Watt war ein super Hinweis. Seit dem ich auf mindestens 4% runter regele, bleibt der WR nicht hängen.

dtuuser, 2023-03-03:
Bei mir alles okay. habe seit gestern die 5.92 am Laufen. Limit wird bei Bedarf alle 60 Sekunden gesetzt, wird sofort in % angezeigt und nach maximal 30 Sekunden wird der neue Erzeugungswert angezeigt.
Topp

lumapu, 2023-09-16:
habe einen Hinweis in der Doku hinzugefügt. Aktuell noch in der Dev-Version, aber in naher Zukunft auch als Release

closed 2023-09-16 by lumapu