lumapu/ahoy #709: 5.91 kein Inverter bei Serial/Control Auswahl

dtuuser, 2023-02-25:
Guten Morgen zusammen,
habe gerade die  5.91 installiert. WR läuft und produziert, wird aber unter Serial/Control (Select Inverter) zum Limit setzen nicht angezeigt

Ollipop030, 2023-02-25:
Live, Settings, System ist auch ausgeblendet.

lumapu, 2023-02-25:
ja das Menü kommt mobil nicht, mache heute einen hotfix

dtuuser, 2023-02-25:
Ist wieder da. 
Vielen Dank für die super schnelle Rückmeldung

Ollipop030, 2023-02-25:
Am Rechner mit 0.5.92 kommt das Menü aber nicht. Firefox und Edge getestet.

dtuuser, 2023-02-25:
Bei mir auch nur in Chrome. Das ist bei mir aber schon lange so

knickohr, 2023-02-25:
Das hatten wir doch schon mal 😲 Wurde bei mir als „zu alten“ Webbrowser abgetan 😉

Ollipop030, 2023-02-25:
Naja, Firefox ist 110, aktueller geht es nicht. Aber runter auf die 0.5.90, und Menü ist wieder da.

dtuuser, 2023-02-25:
Bei mir blockte die Firewall bzw. der Javafilter

knickohr, 2023-02-25:
Dann mach zu das Ding 😉

Ollipop030, 2023-02-25:
> Dann mach zu das Ding

Warum? Fehler ist noch vorhanden. Firefox ohne Addons, Edge und Firefox auf dem Handy, überall das gleiche Bild:

[image]

Ich glaube, es liegt daran:

[image]

Die Haken kann ich deaktivieren, sind aber nach dem Neustart wieder vorhanden. Evtl. funktionieren die Switches einfach nicht?

knickohr, 2023-02-25:
Ah OK, dachte das wäre mit der Firewall behoben. War wohl nur bei @dtuuser  so 😉

lumapu, 2023-02-25:
ok, die Haken haben einen Haken 😂, man müsste halt alles testen bevor man committed, bin da leider immer etwas schnell ...

MiniOh, 2023-02-27:
Ich musste auch auf die, für mich sehr stabile 0.5.89 zurück.
Wann gibts die 93?

lumapu, 2023-02-27:
noch nicht, dauert aber bestimmt nicht mehr lange 😉

lumapu, 2023-03-10:
Der Issue #699 hat das selbe Fehlerbild. Der Fehler wurde bereits behoben, dieser Kommentar dient nur der Dokumentation

closed 2023-03-27 by lumapu