lumapu/ahoy #710: Kein Verbindungsaufbau nach zweimaligem Routerausfall

NewU2023, 2023-02-25:
Hardware: ESP8266
 
Hallo,
wenn der Router einige Zeit ausgefallen ist (oder ausgeschaltet wird) und anschließend wieder neu gestartet wird, dann stellt Ahoy nur **einmalig** wieder die Verbindung her. Nach dem zweiten Ausfall kann nicht mehr auf die AhoyDTU zugegriffen werden. Es hilft nur ein Reset oder stromlos machen. 

Version 0.5.90; tritt auch bei älteren Vesionen auf.

lumapu, 2023-02-25:
danke für den für mich neuen use case 😎

fsck-block, 2023-02-26:
Hardware ESP8266
Software: Last Stable 0.5.66

> tritt auch bei älteren Vesionen auf.

Das kann ich zumindest für meine 0.5.66 nicht bestätigen. Habe derzeit Probleme mit meinem Kabelrouter. WLan fällt öfters aus und kann nur durch Neustart behoben werden. Ein reconnect von ahoy-dtu fuktioniert dabei immer zuverlässig.

@NewU2023 : Gibt dein Router Log Hinweise was beim (re)connect nicht klappt?

Ollipop030, 2023-02-26:
Liegt aber denke ich nicht an Ahoy. Wenn Ahoy das Netz verliert verbindet es sich eigentlich direkt wieder automatisch, egal wie oft das Netz weg ist. Bei mir zB ist jede Nacht WLAN weg, morgens verbindet es sich aber ordnungsgmäß wieder, auch am darauffolgenden Tag. Ich kann auch kurz WLAN abschalten, Ahoy ist nach ein paar Sekunden wieder verbunden. @NewU2023  In welchem Intervall wird denn der Router neu gestartet? Was ist das für ein Router? Und wieso fällt überhaupt der Router so oft aus?

NewU2023, 2023-02-26:
Ich habe mal die Version 0.5.66 geladen. Hier funktioniert es; Router mehrmals aus- und wieder eingeschaltet; Verbindung wird aufgebaut.

Bei der Version 0.5.90 ist die Diagnose beim Router: Pingtest fehlgeschlagen.

@Ollipop030: Es ist der Standardrouter von Vodafone (Kabeldeutschland). Ich mache es auch genauso: nachts wird Kabelmodem, Router, Telefon abgeschaltet; nur Ahoy und Auswertung bleiben an.
Aber jeden zweiten Morgen muß baut Ahoy keine Verbindung auf und ich muß es kurz abstecken. 

Evt. liegt es doch an der Dev-Version.

Ollipop030, 2023-02-26:
Habe fast alle Devs seit der letzten Stable probiert, 8266 sowie ESP32. Hatte mit der WLAN Verbindung nie Probleme, vor der 0.5.66 gabs da aber mal Schwierigkeiten. 
Vielleicht ein dirty Workaround: Wenn du nachts eh alles totlegst, schalt doch die DTU mit ab, und starte sie morgens mit dem Router zusammen wieder.

fsck-block, 2023-02-26:
> Vielleicht ein dirty Workaround: Wenn du nachts eh alles totlegst, schalt doch die DTU mit ab, und starte sie morgens mit dem Router zusammen wieder.
Falls du deine DTU im Bereich deines Kabelrouters (Vodafone Station) liegen hast: Die Vodafone Station hat einen unbenutzten USB Anschluss. Außer Strom liefern kann der sonst nichts. Dort einstecken dann läuft's syncron.

closed 2023-04-08 by lumapu