lumapu/ahoy #711: Daten an IoBroker senden über MQTT funktioniert nicht

ThomasKlaus-ai, 2023-02-25:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ x] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Die Darstellung von AHOY DTU funktioniert im Browser sehr gut. Leider kann ich die Daten aber nicht via MQTT in meinen Raspi bekommen. In Objekten erscheinen zwar meine beiden DTU‘s, aber nur mit Ordnern innerhalb meiner Sonoff Geräte. Es gibt dann nur die Ordner DTU und alive. Daten aber sehe ich keine. Normalerweise benötigt man im IoBroker immer einen Adapter, der eine Instanz erzeugt. Für OpenDTU habe ich einen solchen Adapter in der Beta Version gesehen. Wird es für Ahoy auch einen Adapter geben, oder gibt es eine andere Möglichkeit Daten in den IoBroker zu bekommen ?

lumapu, 2023-02-25:
kann ich nicht nachvollziehen, ich nutze selbst ioBroker mit Ahoy.
Wie sind deine Einstellungen?

Die Notwendigkeit für einen gesonderten Adapter erschließt sich mir noch nicht. Ich verwende den Standard MQTT Adapter

Tiezy, 2023-02-25:
Bei den Instanzen muss der MQTT Adapter installiert sein. Sieht dann so aus.
[image]

ThomasKlaus-ai, 2023-02-25:
Das sieht bei mir so aus. Ich empfange schon seit 1 Jahr Daten über Tasmota problemlos.Viele GrüßeThomas

ThomasKlaus-ai, 2023-02-25:
Welche Porteinstellung muß ich denn für Ahoy wählen ? Sonoff hat bei mir 1883 und welche Einstellung muß ich dann bei MQTT vornehmen ?

Tiezy, 2023-02-25:
z.b. 1885 in der Ahoy DTU und bei der MQTT Instanz einstellen.

c1328, 2023-02-26:
ggf. ist das hilfreich bei der Einrichtung: https://github.com/c1328/ahoy-iobroker-influx-grafana

ThomasKlaus-ai, 2023-02-26:
ich bekomme in den Objekten leider nur 2 Ordner: einen Ordner mit dem Topic des AHOYDTU und einen Ordner darunter mit der Bezeichnung alive. Die Häkchen für History Speicher und MQTT sind alle gesetzt. Bei meinen Tasmota Sensoren hat alles ohne Probleme funktioniert. Bei AhoyDTU bekomme ich leider keine Baumstruktur mit den Daten.Viele GrüßeThomas

AlterSagg, 2023-02-26:
Hi, hab es auch gerade gerafft:
Nimm im AHOY den mqtt-port auf  z.B. 1885; dann speichern, der AHOY startet neu.
Dann im iobroker dem mqtt-Adaper bei den Instanzen bei Verbindung auf "Server/Broker" und  auch Port 1885. 
Nach 'ner Minute hat der iobroker/mqqt-Adapter alle(?) Datenpunkte angelegt.

Bei mir fehlen nur die Sachen mit Sonnenauf und -untergang, obwohl im AHOY eingetragen und funktionabel.
Gruß

ThomasKlaus-ai, 2023-02-26:
So sehen meine Objekte aus. Bei DTU-Garage und DTU-Terrasse sehe ich keine Baumstruktur mit den Daten. Als Quelle sehe ich Sonoff und das stimmt wohl nicht, ist aber automatisch eingefügt worden.Bei meinen Tasmota Geräten sehe ich die Baumstruktur mit allen Daten. Warum geht das nicht mit DTU-Geräten ?Viele GrüßeThomas

ThomasKlaus-ai, 2023-02-27:
Wenn ich in Ahoy den MQTT Port 1885 eintrage und das gleiche im IoBroker MQTT, dann empfange ich meine Tasmota Sensoren nicht mehr. Gibt es eine MQTT Einstellung im IoBroker, die mehrere Portangeben zuläßt ?

AlterSagg, 2023-02-27:
@ThomasKlaus-ai 
aif der Adapterseite eine neue (zweite) Instanz des Mqtt-Adapters erstellen. Die eine als Client mit Port 1883, die zweite als Server mit Port 1885.

Gruß

ThomasKlaus-ai, 2023-02-27:
Vielen Dank, AlterSagg. Die 2.MQTT Instanz war die Lösung für mein Problem. Super, jetzt funktioniert es. Ich wußte nicht, daß man mehrere Instanzen mit MQTT installieren kann.

closed 2023-02-27 by ThomasKlaus-ai