lumapu/ahoy #713: [ESP8266/ESP32/RaspberryPi] Kein Powerlimit über mqtt

MrLee99, 2023-02-25:
### Platform

ESP32

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

Pinout

CS
GPIO5
CE
GPIO16
IRQ
GPIO17
LED0
off / default
LED1
off / default

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.92

### Github Hash

n/a

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

Pinout

CS
GPIO5
CE
GPIO16
IRQ
GPIO17
LED0
off / default
LED1
off / default

### Error description

power limit via mqtt since I assume .80 no longer operational.

sending mqtt.0.PV_Garage.ctrl.limit_nonpersistent_absolute.0 does not show any reaction.
Sending powerlimit to mqtt.0.PV_Garage.ctrl.limit_nonpersistent_absolute (not a special channel like 0 or 1) seems to work but only for the first WR.

Is it no longer possible to send a power limit to a SPECIFIC WR or is there any change to the mqtt channel?

Mr.Lee

MrLee99, 2023-03-03:
Moin!

gibts hier Neuigkeiten? Und vor allem...bin ich der einzige mit dem Problem?
Vielleicht gibt es hier noch weitere ioBroker Nutzer die das Problem ggf. nicht haben?

bis denne
Mr.Lee

lumapu, 2023-03-05:
ich glaube die subscription steht falsch und wertet aktuell die ID nicht aus.
Ich schaue rein

MrLee99, 2023-03-05:
Ok, Danke!

KG3RK3N, 2023-03-08:
Hab auch das Problem das ich per Mqtt kein Power Limit mehr setzen konnte auf dem esp8266 und einem Inverter mit v0.5.93. 
Bin erstmal auf die json API ausgewichen, damit klappt es.

In den logs konnte ich auch keinen Empfang von mqtt entdecken wie das sonst der Fall war. 
Ich vermute, er steigt hier aus https://github.com/lumapu/ahoy/blob/f5ef19ccab0f9c8bc68d339a72d4e19f0c248c05/src/publisher/pubMqtt.h#L343

Bei der 0.5.66 die ich vorher hatte klappte das noch wunderbar.

lumapu, 2023-03-09:
ich muss da nochmal debuggen, irgendwie habe ich in Hinterkopf, dass die ID des inverters nicht ausgewertet wird

lumapu, 2023-03-10:
bitte auch das [User Manual](https://github.com/lumapu/ahoy/User_Manual.md) für die neue Syntax beachten!

MrLee99, 2023-03-11:
5.98 Funktioniert, top!

closed 2023-03-27 by lumapu