lumapu/ahoy #714: Pinout not set to standard after initial flash

luftschlucker, 2023-02-25:
i've downloaded Ver. 05.90 from https://ahoydtu.de/getting_started/ and flashed it using nodeMCU-PyFlasher.
All Pinout settings in the System Config were set to D3(GPIO0). 
As it is not mentioned in the Guidelines to chenge these settings, it took me a while to fiddle it out.

Tiezy, 2023-02-25:
There is a more convenient way by simply using the ahoy webpage programmer and flash the latest release version automatically. The System Config is completely  included. Then download the latest development SW and run an OTA update from Ahoy DTU webfront. For sure you need to update settings for new features according to your usage condition.

luftschlucker, 2023-02-25:
i did what you suggested - everything went flawlessly. Thnx for your advice!

closed 2023-05-25 by lumapu