lumapu/ahoy #715: ESP32 - OTA funktioniert ab 0.5.89 nicht mehr

knickohr, 2023-02-25:
### Platform

ESP32

### Model name

DevKit C

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

05.89 ++

### Github Hash

Mehrere Versionen

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Sei der Verdion 0.5.89 geht der OTA nicht mehr. Das Binary wird offenbar noch geladen, dann aber passiert nichts mehr. Das Webinterface ist aber nach einem Refresh noch erreichbar und man kann die DTU manuell booten. Er kommt aber mit der alten Version wieder hoch.

Ein User berichtete sogar das es mit der 70 auch nicht geklappt hat, ein weiterer hatte mit der 89, wie ich auch kein Erfolg. Heute Morgen das gleiche Spiel dann auch mit der 90.

Da es aber bis zur 89 für funktioniert hat, gehe ich davon aus, das es in den neuen Versionen ab 89 ein Problem gibt. Ich weiß jetzt aber nicht ob es nur auf dem ESP32 beschränkt ist und es auch auf den 8266 zutrifft. Müßte man im Discord mal nachlesen.

bei anderen Usern scheint es aber offenbar zu funktionieren.

Screenshot vom USB-serial Log hänge ich an. Mehr als die letzten beiden Zeilen kommen während des Updates nicht. Ebenso keine Exception, es wird auch nicht rebootet.

[image]

knickohr, 2023-02-25:
Nachtrag :
Habe eben im Discord nachgelesen. Es betrifft sowohl den ESP32 als auch den 8266 😲

Kouichi-kun, 2023-02-25:
Hab ebenfalls von der 89 auf die 92 geupdatet, ich hab das Problem dass das Interface unbrauchbar geworden ist.
Es kommt der normaler Ahoy Startbild, es wird kein PW mehr abgefragt.

Wenn ich nun auf Login gehe und mein PW eingebe, komme ich auf die gleiche Seite wie im Screenshot.
Kein Setup oder anderer Menüpunkt mehr da.

VOrher hatte ich es so eingestellt, dass er immer das PW braucht.
Keine einzige Seite ohne PW

[image]

knickohr, 2023-02-26:
Du bist doch auf der 92. Wo ist das Problem ?

Dein Problem mit der unvollständigen Darstellung ist in #709 zu suchen ?

lumapu, 2023-02-26:
ich habe das Menü umgebaut, da ist scheinbar mit Passwort was gründlich schief gelaufen, werde es beheben

knickohr, 2023-02-26:
Und das hat mit den OTA-Updateproblemen zu tun ?  Ich nutze kein Paßwort 🤔

lumapu, 2023-02-26:
sorry @knickohr das bezog sich auf die Meldung von @Kouichi-kun 

Noch habe ich keine Idee was beim Update schief geht

reserve85, 2023-02-26:
Kann ich bestätigen, Update funktioniert nur noch via Flashen. Bei mir bricht beim Ausführen des OTA Updates immer die Verbindung ab. Dann ist der ESP nicht mehr erreichbar für 20 Sekunden, anschließend wieder online aber uptime unverändert.

beegee3, 2023-02-27:
hatte das Problem beim Update von '89 auf '90 auch. Gleiche Symptome wie bei @knickohr. Nach 4 Versuchen ging es. Seitdem aber kein Problem mehr.

lumapu, 2023-02-27:
@beegee3 kam bei dir `Update started` bzw. kommt es jetzt wieder?
Kann das mit der Fehlerbehebung beim Reboot (Exception) zusammenhängen?

beegee3, 2023-02-27:
@lumapu ja `update started` kam immer, nur  `update success` nicht, blieb also auf der alten Version. Allerdings nutze ich nicht die  .bin Versionen, sondern kompiliere den Quellcode lokal, da ich meine Modifikationen beibehalte (`ahoywifi` + `hmRadio`). Kann also nicht als Referenz herhalten. Welche 'Fehlerbehebung beim Reboot (Exception)' meinst du?

Könnte mir vorstellen, dass es die .bin Größe kombiniert mit schlechter Wifi Verbindung ist. Die Updatedatei ist fast mit jeder Version seit '66 größer geworden, also dauert die Übertragung entsprechend. Habe es gerade mit bewusst schlechtem AP (RSSI -80) ausprobiert und prompt riss die Verbindung während des Updates ab. Folglich ist das Update fehlgeschlagen, Ahoy hat natürlich die Wifi Verbindung neu aufgebaut, so dass es genau so aussieht, wie es die anderen hier beschreiben. Bei RSSI - 45 habe ich keine Schwierigkeiten, trotz meiner häufigen Updates, die schon mal beim tüfteln und testen anfallen. Übrigens war bei dem '89 auf '90 Update mein Standard AP auch gerade Offline. Will keine falsche Fährte auslegen, sehe es nur im Rahmen einer möglichen Erklärung.

knickohr, 2023-02-28:
Langsam, ich habe auch noch eine Theorie !

Ich glaube nicht das es an meiner WiFi-Verbindung liegt, die ist immer so zwischen -60 und -50dBm. Ich glaube es liegt daran das da was vom MQTT oder vom NRF-Radio dazwischen spukt, wie ich darauf komme ?  Irgendjemand hat einmal vor gewisser Zeit gesagt das wenn man das TX-Intervall zu den WRs aufs 10 runter setzt, es geschmeidiger läuft (was auch stimmt !). Und/oder das MQTT sendet ja auch nach jedem empfangenen Telegramm von den WRs (was man ja auch in meinem Screenshot oben sehen kann).

Habe es eben ausprobiert, der Downgrade auf eine frühere Version und der folgende Upgrade wieder zurück auf die 92 lief problemlos als ich das Intervall wieder auf 30s zurück stellte 😲

Also Pull Request :  Disable all transmittion (MQTT & Radio) while updating !

beegee3, 2023-02-28:
@knickohr ist schon so. Mit Update Start werden als erstes alle Ticker bis auf Wifi gestoppt (in `web.h`, `showUpdate2`, 1. Zeile: `mApp->setOnUpdate();`). Bereits gestartete MQTT & Radio Prozesse laufen noch zu Ende, was aber maximal ein paar Millisekunden dauert. Glaube nicht, dass das einen Einfluss hat. Neue Prozesse werden nicht gestartet. Während des Updates laufen also nur noch die Scheduler loop und die Wifi loop. Letztere macht nichts, solange kein Disconnect kommt. Theoretisch kann man auf die beiden Loops auch noch verzichten, aber die sind eher unproblematisch. Irgendwelche 'pending objects' durch asynchrone Wifi Kommunikation schließe ich zumindest bzgl. MQTT auch aus (das war bei meinem Update failure gestern nicht aktiviert). Bei der Web Oberfläche weiß ich nicht, ob da abgesehen vom Update noch etwas anderes läuft.
Habe aber bei meinem Test gestern den ESP am USB Port gehabt und im serial output gesehen, dass während des Updates die Wifi Verbindung abgebrochen ist und daher von Ahoy neu gestartet wurde.

knickohr, 2023-02-28:
@beegee3 

Ja, würde auch die letzte Zeile von meinem Screenshot oben erklären. Aber warum kommt da eine Meldung vom MQTT ?

„MQTT disconnected, reason :  TCP disconnect“

sieht für mich so aus, als würde die von Dir beschriebene WiFi-Verbindung in die Hose gehen, nur warum kotzt MQTT dann wenn da nichts mehr läuft ?

🤔

beegee3, 2023-02-28:
@knickohr beim Initialisieren von MQTT werden Event Callbacks für Connect, Disconnect und Message im MQTT Event Handler eingetragen. Der wiederum baut auf dem Wifi Event Handler auf. Events werden immer an die Callbacks weitergegeben und sind von den Tickern unabhängig. Die Ausgabe „MQTT disconnected...“ ist ist das einzige, was der MQTT Disconnect Callback macht. Ist halt der Hinweis auf Wifi Verbindungsverlust was hier gleichzeitig bedeutet, dass das Update fehlgeschlagen ist.

knickohr, 2023-03-07:
Eigentlich kann man das zu machen, oder ?

lumapu, 2023-03-09:
nicht ganz, mir ist in der `settings.h` noch aufgefallen, dass ich dort scheinbar einen Kommentar bei der Funktion `stop` hinterlassen habe.

knickohr, 2023-03-09:
Wenn ich das gerade mit den JSON lese. Die Rückspielung der Settings-Datei funktioniert auch noch nicht 🙁

lumapu, 2023-03-09:
Ich kann mir gut vorstellen, dass das wieder an den maximalen Buffergrößen liegt, die wir vorgeben. Haben wir ein gesonderten Issue dazu?

reserve85, 2023-03-10:
nur als Info: Zur Night Time funktioniert das OTA problemlos.

lumapu, 2023-03-10:
😊 deswegen habe ich nie Probleme, danke fürs teilen der Beobachtung

knickohr, 2023-03-10:
Der Update-Bug ist in der 97 wieder da 😪

Einmal kam sogar 
[image]

Workaround :  Inverter Communicatin und MQTT deaktivieren, dann klappt’s mit der Nachbarin 😉

lumapu, 2023-03-24:
ich mach hier zu 😊