lumapu/ahoy #717: [ESP8266] Resetting MQTT username and password to blank smashes whole configutation

Teggemann, 2023-02-26:
### Platform

ESP8266

### Model name

Hoymiles HM 1500

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

Device Host Name
Device Name 
AHOY-DTU
Network
WiFi
Enter the credentials to your prefered WiFi station. After rebooting the device tries to connect with this information.

Search Networks 
Avail Networks 
not scanned
 SSID 
FRITZ!Box Fon WLAN 7360
 Password 
•••••
Static IP (optional)
Leave fields blank for DHCP
The following fields are parsed in this format: 192.168.1.1

IP Address 
 Submask 
 DNS 1 
 DNS 2 
 Gateway 
Protection
Protection
Admin Password 
Select pages which should be protected by password

Index
Live
Serial / Console
Settings
Update
System
Inverter
Inverter
Inverter 0

Communication Enable
Serial Number (12 digits)*
116181858816
Name*
01HM1500
Max Module Power (Wp)
390
390
390
390
Module Name

Delete

General

Interval [s] 
30
 Max retries per Payload 
5
NTP Server
NTP Server
NTP Server / IP 
pool.ntp.org
 NTP Port 
123
 set system time  
Sunrise & Sunset
Sunrise & Sunset
Latitude and Longitude must be set to be stored! decimal separator: '.' (dot)

Latitude (decimal) 
51.46890
 Longitude (decimal) 
7.24314
 Offset (pre sunrise, post sunset) 
30 minutes

disable night communication 
MQTT
MQTT
Broker / Server IP 
192.168.2.82
 Port 
1883
 Username (optional) 
 Password (optional) 
 Topic 
inverter
 Discovery Config (homeassistant) 
System Config
System Config
Pinout (Wemos)

CS
D8 (GPIO15)
CE
D4 (GPIO2)
IRQ
D3 (GPIO0)
LED0
off / default
LED1
off / default
Radio (NRF24L01+)

Amplifier Power Level
HIGH
Serial Console

print inverter data 
Serial Debug 
Interval [s] 
5
Reboot device after successful save 

### Error description

When I reset username and password to blank ("") on settings -> MQTT the whole configuration of the dtu is smashed after saving. Evan the wiring settings of the radio interface are deleted.

I experienced this behavior twice.

It helps to reconfigure but since I hadn't the pin configurations for the radio module to hand I opted for factory reset. I don't exactly know why, but I performed the factory reset twice.

With the revirgined dtu every thing was fine again ... until I reset MQTT username and password...

Beside this, you made a great piece of software!

lumapu, 2023-02-26:
thank you, I hope I can reproduce that for debugging purposes.
Great to hear that your satisfied using Ahoy

Teggemann, 2023-02-26:
I saw in the sources that MQTT username and password where initialized with "\0". This might has something to do with the behavior.

DanielR92, 2023-08-26:
Hello @Teggemann, can you test with the new version of ahoy please?
Hope it is now solved. Thanks for your help. - I will close it now.

If something happend, please reopen here that session. Thanks in advance.

closed 2023-08-26 by DanielR92