lumapu/ahoy #721: [ESP32] obwohl Inverter schon lange nicht mehr produziert werden Werte angezeigt und MQTT Meldungen dazu versendet

Pixel-Chris, 2023-02-27:
### Platform

ESP32

### Model name

AZDelivery ESP32 NodeMCU Dev Kit C mit CP2102

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

Web-Settings:
CS - GPIO5
CE - GPIO4
IRQ - GPIO16

Wiring:
GND - GND (Pin1)
3,3V - VCC (Pin2)
G4 - CE (Pin3)
G5 - CS (Pin4)
G16 - IRQ (Pin8)
G18 - SCK (Pin5)
G19 - MISO (Pin7)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

Sunrise und Sunset sind gesetzt
disable night communication   off/on  beides getestet

### Error description

Hallo Zusammen,
hoffe ich habe kein Issue hierzu übersehen....

0.5.66:

Obwohl der Inverter nicht mehr produziert und auf der Startseite auch "is not yet available" angezeigt wird.
Werden in der Live-Ansicht unter P_DC (Total und der einzelnen Channels) noch Werte angezeigt.

Diese sollte, sobald der Inverter nicht mehr erreichbar ist auf 0 gesetzt werden.

Das gleiche gilt für die Nachschaltung, auch hier bleiben (Vermutlich die letzten Werte) immer ein paar Watt in der Anzeige und in den MQTT Nachrichten hängen.

unten steht aktuell bspw.:  
"Last received data requested at: 27.2.2023, 18:16:13" 
jetzt ist es 19:54 und angeblich werden 1,4W produziert.

P_AC ist hingegen auf 0

---

nebenbei:
ist es richtig, das wenn man Channels im Inverter Setup auf 0 setzt, diese trotzdem noch in der Live-Ansicht angezeigt werden?

Grüße
PixelChris

knickohr, 2023-02-27:
Aufpassen !  Einige Werte sind retained geflagt und werden somit im Broker vorgehalten.

Pixel-Chris, 2023-02-27:
und zum Thema "retained" ist es sinnvoll P_DC und P_AC mit dem flag zu verschicken? wenn das Teil nicht mehr produziert, sollte nichts mehr Übertagen werden (wenn überhaupt dann jeweils eine 0) oder verstehe ich da was nicht?

wie könnte eine automatische Löschung dieser Werte klappen?

Pixel-Chris, 2023-02-27:
lt. #722 scheint da was in der nächsten stabile zu kommen - sehe ich das richtig?

knickohr, 2023-02-27:
Welche Werte retained sind, siehst Du hier :
https://github.com/lumapu/ahoy/blob/main/User_Manual.md

Pixel-Chris, 2023-03-01:
@knickohr  gibt es eine Anleitung wie man eine DEV einspielt? und wie man die wo herunterlädt?

knickohr, 2023-03-02:
https://github.com/lumapu/ahoy/actions/runs/4269127247

beegee3, 2023-03-02:
oder so:
zip Datei mit Development Build von https://ahoydtu.de/web_install/ herunterladen und entzippen.
AHOY Update in der Web Oberfläche aufrufen (mit den Standardsettings also http://ahoy-dtu/update), die richtige bin Datei auswählen (je nach ESP das mit der Endung _esp8266.bin oder _esp32.bin) auf UPDATE klicken und sehen, was angezeigt wird (sollte sein: update started -> Update: success rebooting ... auto reload after 20s) -> dev. version -> 😃

Pixel-Chris, 2023-03-02:
Vielen Dank ! das hat geklappt, bin nun auf der 0.5.92

gibt es irgendwie die Möglichkeit den YieldTotal wieder auf den Wert von vor dem Update zu setzen? Der hat sich leider genullt.

Mit dem Offset klappt es leider nicht, obwohl man dort minus und plus Werte eintragen kann

ich müsste auf diese Werte kommen:
CH1: 5.974
CH2: 0.106
CH3: 4.482
CH4: 0.149
CH0: 10.711
Total: 10.711

Pixel-Chris, 2023-03-02:
oder ziehen die erst, wenn der Inverter wieder produziert? aber wie kriege ich da Kommawerte rein?

closed 2023-03-25 by lumapu