lumapu/ahoy #725: Unexpected MQTT values (drops & spikes)

nexulm, 2023-02-28:
Hello,

does anyone already observed some irregular values received via MQTT. I'm currently confused about some drops or spikes I see in my graphs over the day and more strange during the night as well. Please have a look on two screenshots of the same graph time duration from today. "Yield Day" is for me one of the most important values I'm monitoring of my HM1500 inverter. 
The configuration is: Reset values and YieldDay at midnight. Everything else is NOT set.
If I'll not find the source I've more effort in improving my filters to eliminate this behavior.

Would be nice if there is a solution or switch already available I'm not aware of. :-)

[image]

[image]

Argafal, 2023-03-01:
Mir fällt auf dass alle deine drops immer zum gleichen Wert (ca 800) gehen. Ich hatte kürzlich ähnliches drops (selten und unregelmässig, genau wie du) mit einem Wasserzähler. Da war die Ursache eine alte Nachricht mit einem vorherigen Wert, die auf "retained" stand. Diese wurde zwischendurch immer wieder gelesen. Ich habe dann alte retained Nachrichten gelöscht. Das hatte für mich das Problem gelöst. Vielleicht ist die Ursache ja hier ähnlich?

beegee3, 2023-03-01:
it seems that the drops and spikes occur where for some reason the connection to MQTT got lost. My guess: between 3:30 and 7:30 your WLAN is disabled. But I have no explanation why the Yield Day value is set to 750Wh. Looks like this is the Yield Day value from before midnight. Ahoy does not use any history. There is only one Yield Day value in memory. In your case this value is set to 0 at midnight. No possibility for Ahoy to get back the value from before midnight. And, if I'm right and the MQTT connection got lost, Ahoy can't send anything to MQTT.

@Argafal you're too fast 😄 I was thinking about a similar explanation too!

@lumapu Hinweis am Rande: in dem Zusammenhang ist mir aufgefallen, dass in `app::tickZeroValues` und `app::tickMidnight` die Zeile `if (!iv->config->enabled) continue; // skip to next inverter` fehlt. Füge sie doch bitte bei Gelegenheit ein.

knickohr, 2023-03-01:
Diese „Drops“ beobachte ich seit ein paar Versionen auch bei mir. Ist immer nur ein Inverter, so das man wohl Abschattungen ausschließen kann. Es ist aber nicht immer der gleiche Inverter. Da es ja vorher nicht auftrat tippe ich darauf das per MQTT irgendetwas falsches oder unvollkommenes ausgegeben wird.

[image]

Argafal, 2023-03-01:
Ich hab keine Drops mit 0.5.92. Ich hab das Auto-discovery auch nie benutzt (das sind retained messages, oder)? Könnt ihr alle retained messages clearen und dann sehen, ob das Problem noch auftritt? Ich mache das so:

```
sudo systemctl stop mosquitto.service
sudo rm /var/lib/mosquitto/mosquitto.db
sudo systemctl start mosquitto.service
```

Argafal, 2023-03-01:
@nexulm Can you clarify your exact settings please? Which resets did you enable in the inverter settings, what happens with communication at night (sunset settings), do the two graphs show different quantities? What do they show exactly? What do you expect to happen? For example, what happens in your top graph between 2:30 and 7:30 and why doesn't it happen to the bottom graph?

knickohr, 2023-03-01:
Ich ziehe meine Aussagen zurück. Es sind tatsächlich korrekte Werte. Die Leistung bricht aufgrund sehr kleiner Wolken ein. Es ist eben wieder passiert und ich konnte es sehen. Hätte nicht gedacht, das es so viel ausmacht. Bitte mein Gewäsch ignorieren, funktioniert alles bei mir wie es soll 😲 Sorry.

nexulm, 2023-03-01:
@Argafal: My inverter settings are:
[image]
---
I'm not an expert fro MQTT but:
1) WLAN isn't disabled at night
2) As I'm using HA since a few days I did always a "Dicovery Config" in case of minor modifications and restart of HA to get actual values from Ahoy, especially if the inverter is offline after sunset. Maybe 750 was the value for "Yield Day" last time I did "Dicovery config"-send?
3) The two graphs are showing on one hand the original HA values and on the other hand the same entity values logged to Influx DB which is visualized by Grafana for nearly the same timeslot.
4) The graphs should switch to zero (0) at midnight, stay there until the inverter starts producing. The graph should show the rising PV production as it does w/o those drops or spikes.
5) Both graphs are showing the same behaviour, especially between 02:30 - 07:30. The only different is that Grafana is/was configured to show dots for the datapoints and doesn't connect the 4 datapoints (22:00 - 07:30) as the default HA visualization does.
---

knickohr, 2023-03-02:
Hier mal die Aufzeichnung von der Nacht. Nichts was die DTU nicht machen soll, keine Spikes, keine Drops.

[image]

knickohr, 2023-03-02:
Keine Spikes und Drops in den letzten 24 Stunden. Das einzige was auffällt, der aufsummierte YieldDay wird vorher (beim ersten Zucker der Inverter) genullt, die Yields der einzelnen Inverter erst wenn sie produktiv gehen. Aber das kann man mit den Einstellungen verhindern.

[image]

nexulm, 2023-03-02:
I‘ll give a try with another MQTT broker. Good to know that na setup has the fault and not Ahoy. 😀👍🏻
Graph from the last 24h added:
[image]

knickohr, 2023-03-02:
Is this a Broker in the cloud ?  I‘m using a local MQTT Broker.

nexulm, 2023-03-02:
No, it’s the MQTT2-Server from FHEM which I‘m using since years for my home automation. As I‘m currently in the process of switching from one system (FHEM) to another one (HA).

Argafal, 2023-03-02:
> I‘ll give a try with another MQTT broker. Good to know that na setup has the fault and not Ahoy. grinning👍🏻

@nexulm Could you please try the following: turn off ahoy. Restart your MQTT broker. And then would you please post the mqtt messages you have for your ahoy mqtt topic?

knickohr, 2023-03-02:
Ja, hat mich auch gewundert. Es ist sogar noch grasser, er gibt nochmal einen alten Wert aus bevor er nullt 😵

[image]

(Glück gehabt, wat gerade noch drin im Grahen 😅)

beegee3, 2023-03-03:
ah, dann wird ja doch bei Statusänderung 'not available' zu 'available' gepublished.
**Also [pragmatischen Vorschlag](https://github.com/lumapu/ahoy/issues/725#issuecomment-1452336872) bitte sofort wieder vergessen.**
Ahoy nullt nur, wenn in den Settings bei mind. einem der 'Reset values' ein Haken gesetzt ist, was bei dir nicht der Fall ist.
Daher sind bei dir die alten Werte bis zur Statusänderung noch im Speicher.

@lumapu `isAvailable` reagiert ja nicht nur darauf, dass aktuelle Werte vom Inverter geliefert werden, sondern auch auf den `timestamp` von `recordInfo`, `recordConfig` und `recordAlarm`. Kann es sein, dass einer von denen ein 'available' auslöst, **bevor** neue Daten vorliegen? Z.B. in `getQueuedCmd` wird bei Bedarf `InverterDevInform_All` **vor** `RealTimeRunData_Debug` in die Queue gestellt. Das würde erklären, warum @knickohr noch einmal die alten Werte erhalten hat, denn der Inverter selbst wird wohl morgens keine alten Daten mehr liefern.

Sollte `isAvailable` vielleicht nur den `timestamp` von `recordMeas` prüfen, oder gibt es dann an anderer Stelle Probleme? Ansonsten würde eine Funktion helfen, die nur genau das abfragt und dementsprechend den 'available' Status in MQTT setzt, d.h.:
in `hmInverter`
```
        bool hasCurrentData(uint32_t timestamp) {
           return ((timestamp - recordMeas.ts) < INACT_THRES_SEC);
        }
```
in `pubMqtt`
```
         bool processIvStatus() {
         ...
                // inverter status
                uint8_t status = MQTT_STATUS_NOT_AVAIL_NOT_PROD;
                if (iv->hasCurrentData(*mUtcTimestamp)) {
         ...

```

nexulm, 2023-03-05:
After switching to another MQTT broker and using the newer Ahoy DEV 0.5.93 no drops & soikes observed.
[image]

closed 2023-03-05 by nexulm