lumapu/ahoy #726: MQTT - Retained-Flag von 2 Werten entfernen

knickohr, 2023-03-01:
Bitte bei den beiden P_AC und P_DC Werten vom Topic total das Retained-Flag wieder entfernen.

Die beiden Werte sind flüchtig und werden auch nicht aufsummiert. Ein Vorhalten als Retained im Broker ist somit sinnbefreit.

[image]

Pixel-Chris, 2023-03-01:
siehe dazu auch
#721

knickohr, 2023-03-03:
Hier alle Werte die retained geflaggt sind, also auch P_AC und P_DC (DEV 93) :

[image]

knickohr, 2023-03-12:
OK

knickohr, 2023-03-12:
Das war zu viel des Guten 😲

Bitte nur die beiden Werte P_AC und P_DC aus den retained Values entfernen, nicht alle 4 unter inverter/total 😉

Es fehlen jetzt auch die beiden grün markierten Werte YieldDay und YieldTotal 👍 

[image]

knickohr, 2023-03-14:
Jetzt paßts 🤔

closed 2023-03-14 by knickohr