lumapu/ahoy #727: Maximum length of MQTT password?

soebbing, 2023-03-02:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [ ] external antenna

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.92
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [X] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [X] Mac OS

### Debugging:
* [X] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I can receive and display values just fine with Ahoy, but the MQTT integration (to Homeassistant) doesn't seem to be able to connect (logs below). I double and triple checked username/password and the server configuration, but the problem remains.

My only guess at the moment is, that there is some sort of internal limit to the allowed length of a password? My password is quite long (64 chars), looking at the header files I couldn't find anything obvious to that regard.

I'd try and define some shorter password, but I still try to get around the hassle to reconfigure all the connected devices. 😓 

In any case, thank you for your awesome work and the great project!

```
13:45:21 I: MQTT disconnected, reason: not authorized
13:45:22 I: MQTT disconnected, reason: not authorized
13:45:23 I: MQTT disconnected, reason: not authorized
13:45:24 I: MQTT disconnected, reason: TCP disconnect
13:45:25 I: MQTT disconnected, reason: not authorized
13:45:26 I: MQTT disconnected, reason: not authorized
13:45:27 I: MQTT disconnected, reason: not authorized
13:45:28 I: MQTT disconnected, reason: not authorized
13:45:29 I: MQTT disconnected, reason: not authorized
13:45:30 I: MQTT disconnected, reason: not authorized
13:45:31 I: MQTT disconnected, reason: not authorized
13:45:32 I: MQTT disconnected, reason: not authorized
```

soebbing, 2023-03-02:
Some digging revealed this definition:
https://github.com/lumapu/ahoy/blob/f8fe044e1be86e8f2730807fb0da79d97e880feb/src/defines.h#L92
That looks like a 64 char password, plus zero byte `\0` should still be valid - is there any other possible reason I'm not seeing?

lumapu, 2023-03-09:
can you check another MQTT broker? Your code snippet is correct, it's allowed to have 64 chars of password + one extra char for string end.
Maybe it is possible to have a manual compile on your side and add some debug prints like `Serial.println(...);` with the password. So you're able to check whether the password is correctly stored.

soebbing, 2023-03-09:
Thank you for your feedback. I don't have any other broker running, unfortunately. I used OpenDTU successfully before, if that is off any informational value.

I'll try to add a few debug statements, that is a great idea! I'll get back if I have any more information!

soebbing, 2023-03-12:
Ok, when I tried to log some stuff I realized: MQTT was connecting and worked! I have no idea what was going on before or what I might have changed.

Anyway, sorry for the unnecessary issue here, closing it!

closed 2023-03-12 by soebbing