lumapu/ahoy #731: Live Ansicht leer mit Version 0.5.93

Ollipop030, 2023-03-03:
Moin,

gerade die 0.5.93 installiert, die Live Ansicht ist komplett leer, keine WR und keine Module werden angezeigt. Bei WR sind aber available und producing.

MiniOh, 2023-03-03:
Ja, hier leider auch, ... war aber, ja nach Anzahl der Invter auch schon on .91 unf .92 so ...

Ollipop030, 2023-03-03:
Ich habe nur 2 WR, mit der .92 wurden sie bei mir noch angezeigt. Aber der dark mode sieht echt mal gut aus.

lumapu, 2023-03-03:
wirklich? ich habe mir 4 virtuellen Invertern getestet.
Könnt ihr bitte prüfen was die Developer Konsole eures Browsers meldet?
In Firefox gibt es auch eine Netzwerkanalyse, hier müsste pro Wechselrichter eine Anfrage kommen

Ollipop030, 2023-03-03:
Das hier kommt im Firefox:

Quellübergreifende (Cross-Origin) Anfrage blockiert: Die Gleiche-Quelle-Regel verbietet das Lesen der externen Ressource auf http://10.20.3.44/api/inverter/id/1. (Grund: CORS-Anfrage schlug fehl). Statuscode: (null).

Uncaught SyntaxError: JSON.parse: unexpected end of data at line 1 column 1 of the JSON data
    p http://192.168.0.19/api.js:149
    getAjax http://192.168.0.19/api.js:140
    parseIv http://192.168.0.19/live:235
    p http://192.168.0.19/api.js:149
    getAjax http://192.168.0.19/api.js:140
    parse http://192.168.0.19/live:255
    p http://192.168.0.19/api.js:149
    getAjax http://192.168.0.19/api.js:140
    <anonymous> http://192.168.0.19/live:260

Wiederholt sich dann alle 6 Sekunden (meine Intervall ist so eingestellt). Aber unter Edge ist die Seite auch leer.

MiniOh, 2023-03-03:
Ich weiß nicht ob es hilft, bei der Suche was, wann verändert wurde.

Bei mir sag es wie folgt aus:
Bei der 0.5.90 ist bei mir bis Inverter 8 alles OK.
Bei Inverter 9 wird allerdings bei den Channels nicht mehr alles angezeit, wie auch schon in #682 beschrieben.

Bei der 0.5.92 sieht man bei 8 aktiven Invertern zwar noch die "Total Ansicht", diese zeigt aber 0 an. 
Inverter und Channels werden gar nicht angezeigt

Bei der 0.5.93 ist die Live Ansicht, wie oben beschrieben ganz leer.

lumapu, 2023-03-03:
habs gefunden - sorry mein Fehler ich habe meine Test-IP vergessen zu löschen 🙈
Ich werde gleich committen

lumapu, 2023-03-03:
https://github.com/lumapu/ahoy/actions/runs/4321762295

Ollipop030, 2023-03-03:
Besten Dank, installiert und läuft!

MiniOh, 2023-03-03:
Bei 8 Invertern leider auch nich nicht funktional.
Bei der 0.5.90 war das OK.

Sieht so aus in der [90955b9]

[image]

[image]

MiniOh, 2023-03-03:
Kannst du evtl auch mal mit 8 virteullen Invertern testen?
Es reicht auch schon aus, wenn 7 angelegt, aber inaktiv sind. Dann funktioniert der 8. dennoch nicht in der Live Ansicht.

lumapu, 2023-03-03:
@MiniOh ok mache ich
@Ollipop030 das freut mich - habe euch auch ein paar Tage glühen lassen =)

Ollipop030, 2023-03-03:
Kann es sein, das unter Total die Werte alle paar Sekunden nur addiert werden? Wenn man ein bisschen zusieht steigen AC Power, Yield Day, Yield Total, DC Power, Reactive Power immer weiter. Angezeigt werden über 26.000 Watt und weiter steigend.

MiniOh, 2023-03-03:
> Kann es sein, das unter Total die Werte alle paar Sekunden nur addiert werden? Wenn man ein bisschen zusieht steigen AC Power, Yield Day, Yield Total, DC Power, Reactive Power immer weiter. Angezeigt werden über 26.000 Watt und weiter steigend.

Das klaingt super, da ist der Akku schnell voll ;)

mtet2, 2023-03-03:
Mit F5 im Firefox sind die Werte dann immer für eine Weile IO bis auf das die Tagesleistung nicht mit KWh beschriftet ist.

closed 2023-03-27 by lumapu