lumapu/ahoy #732: Correct spelling of unit "var" -> "VAr"

Argafal, 2023-03-03:
Patch below:

```
diff --git a/src/web/html/visualization.html b/src/web/html/visualization.html
index 4c3c6a4..781893a 100644
--- a/src/web/html/visualization.html
+++ b/src/web/html/visualization.html
@@ -81,7 +81,7 @@
                             ml("div", {class: "hr"}),
                             ml("div", {class: "row"}, [
                                 numMid(total[3], "W", "DC Power"),
-                                numMid(total[4], "var", "Reactive Power")
+                                numMid(total[4], "VAr", "Reactive Power")
                             ])
                         ])
                     ])
@@ -115,7 +115,7 @@
                                 numMid(obj.ch[0][1], "A", "Current"),
                                 numMid(obj.ch[0][3], "Hz", "Frequency"),
                                 numMid(obj.ch[0][9], "%", "Efficiency"),
-                                numMid(obj.ch[0][10], "var", "Reactive Power"),
+                                numMid(obj.ch[0][10], "VAr", "Reactive Power"),
                                 numMid(obj.ch[0][4], "", "Power Factor")
                             ])
                         ])

```

knickohr, 2023-03-03:
Bei der Gelegenheit auch die Zeilenabstände zwischen den Feldern im grünen Bereich etwas größer machen. Verwirrt Beil lesen sonst etwas.

[image]

beegee3, 2023-03-03:
... und die % Werte (Efficiency, Irradiation) mit 2 Nachkommastellen. Sowieso bei allen Werten mit Nachkommastellen diese fix, also auch, wenn da nur Nullen stehen. Sonst springen die Werte in der Anzeige unangenehm (hab' aktuell 2 Sekunden Intervall).

roku133, 2023-03-03:
Wenn mir persönlich auch das Einheitenzeichen "VAr" für die Blindleistung besser gefiele, ist es "var":
Blindleistung Var
1 var = 1 W
https://de.wikipedia.org/wiki/Var_(Einheit)

DB1BMN, 2023-03-04:
Vorsicht!
Das Einheitenzeichen für die Blindleistung ist 'var', also alles klein geschrieben.
Ähnlich wie bei 'Ampere', 'Volt' und 'Farad', wähend die namensgebenden Physiker Ampère, Volta und Faraday hießen

lumapu, 2023-03-06:
was jetzt genau, ich glaube wir haben jetzt alle Varianten gesehen, lowercase, uppercase und camelcase
Was ist jetzt richtig?

stefan123t, 2023-03-06:
@lumapu Siehe Wikipedia: https://de.wikipedia.org/wiki/Var_(Einheit)
Das **Var** (mit dem [Einheitenzeichen](https://de.wikipedia.org/wiki/Einheitenzeichen): **var**) ist in der [elektrischen Energietechnik](https://de.wikipedia.org/wiki/Elektrische_Energietechnik) eine gesetzliche Einheit für die [Blindleistung](https://de.wikipedia.org/wiki/Blindleistung). Das Var wurde früher auch Blindwatt (Einheitenzeichen: bW) genannt.

__Ursprung__
Die Bezeichnung **Var** kommt vom Begriff „**V**olt**a**mpere **R**eaktiv“ und wurde 1932 von dem Rumänen [Constantin Budeanu](https://de.wikipedia.org/wiki/Constantin_Budeanu) vorgeschlagen und von der [International Electrotechnical Commission](https://de.wikipedia.org/wiki/International_Electrotechnical_Commission) (IEC) festgelegt.[[1]](https://de.wikipedia.org/wiki/Var_(Einheit)#cite_note-iec_hist1-1)[[2]](https://de.wikipedia.org/wiki/Var_(Einheit)#cite_note-bio_budeanu-2)

Also laut Wikipedia ist die **kleingeschrieben** Variante `var` das eigentliche **Einheitenzeichen**. 
`VaR` in **camel-case** war der ursprüngliche Vorschlag und `Var` ist der **Name der Einheit**.

stefan123t, 2023-03-06:
Wie @roku133 richtig geschrieben hat `VAr` ist für jeden vermutlich einfacher zu lesen, eben VoltAmpere bzw. VA reaktiv fast selbsterklärend ist, aber formell richtig ist einzig `var` weil es die SI-Einheiten so vorgeben.

knickohr, 2023-03-06:
Ich kenne es auch nur als VA (ohne r).

Ollipop030, 2023-03-06:
> Ich kenne es auch nur als VA (ohne r).

Das ist dann aber Maßeinheit für die Scheinleistung.

Argafal, 2023-03-13:
Fair enough :)

closed 2023-03-13 by Argafal