lumapu/ahoy #733: Ahoy 05.93 läuft aber ohen Nokia-Display

schneire, 2023-03-03:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-03-03:
Hast Du in der Display Config unter den Settings auch das Display konfiguriert ?

[image]

knickohr, 2023-03-03:
Der obige Screenshot ist für ein ESP32.

Hier der für ein 8266 :

[image]

ich kann das Problem nicht nachstellen 🤔

[image]

schneire, 2023-03-03:
Ja des passt schon nur Kontrastwertunterschiede!

 

 <https://user-images.githubusercontent.com/63048210/222761488-ab9b59fc-7819-40c8-a4b4-b22c5601f47d.png>

Loetnase, 2023-03-05:
Bei mir läuft des Nokia unter 05.93 mit diesen Einstellungen auch sehr schön.
[image]

1. Nimm mal den Contrast auf 100% und drücke ein wenig unten am breiten Rand auf den Rahmen.
Ich habe da ein Display das hat da schlechten Kontakt LCD zur Platine und zickt da rum, es geht da an und aus.
Vieleicht ist auch der Rahmen mit den 4 Klips irgendwo ausgehängt.

2. Alle meine drei Nokia Displays haben da das gleiche Problem mit Statischer Aufladung.
Sobald ich mit einem Finger auf irgend einen Kontat tippe, GND oder +3,3V oder sonst was geht meist das Display aus, der Rest auf dem Wemos läuft aber sauber weiter. Über den Brouser ist alles da und weiter bearbeitbar.
Erst mit Reset drücken oder USB Stecker raus und wider rein kommt das Display wieder.

knickohr, 2023-03-05:
@Loetnase 

Warum Pixel Shifting an einem LCD ?  Bringt überhaupt nichts, bei LCD gibt es kein Einbrenn- und Alterungseffekt. Macht nur das Display unruhig.

Schneidet bei Dir der Wert nicht das Logo links und rechts ab ?  Deshalb habe ich das deaktiviert.

Interessant !  Mein Nokia hatte das gleiche Problem mit dem Leitgummi. Der Kontakt war schlecht und die Anzeige ganz flau. Auch war ein Clip ausgehängt. Scheint wohl ein Produktionsproblem zu sein 😵

Loetnase, 2023-03-05:
@knickohr 
Ja da hast du vollkommen recht, Pixel Shifting bringt eigentlich nichts,
aber es ist ein bischen mehr Bewegung auf dem Display, zu viel Ruhe ist auch nix :-)
Ja, das Logo läuft auch etwas raus, aber das Display gibt eben nicht so viel her ist aber schön groß, größer als die Oleds.
Meines sieht gerade so aus, im Schrumpschlauchgehäuse, mit Nachtdesign. Mit zwei Adapterplatinchen selbstgebastelt.
[image]

knickohr, 2023-03-05:
😎 Negligae 🤪

closed 2023-03-25 by lumapu