lumapu/ahoy #737: Änderung bei Ausgabe .../api/live  unter 5.93 geändert?

dtuuser, 2023-03-05:
Guten Morgen,
habe gerade mein Testsystem auf die 5.93 geändert. Im Browser sieht es so besser aus. Daumen hoch
Aber beim API Aufruf der aktuellen Daten hat sich was geändert, oder?

Ich bekam in den "alten" Versionen beim .../api/live Abruf diese Anzeige. Daraus habe ich mir die Werte rausgefiltert.
{"generic":{"wifi_rssi":-72,"ts_uptime":240060,"menu_prot":false,"menu_maskH":0,"menu_maskL":61,"menu_protEn":false,"esp_type":"ESP8266"},"inverter":[{"enabled":true,"name":"Garten","channels":2,"power_limit_read":100,"ts_last_success":1678005435,"ch":[[236.9,0.26,60.7,50.02,1,9.7,292.269,58,63.5,95.591,0],..........................

Unter 5.93 bekomme ich keine Werte mehr angezeigt
{"generic":{"wifi_rssi":-37,"ts_uptime":520,"menu_prot":false,"menu_mask":61,"menu_protEn":false,"esp_type":"ESP8266"},"refresh":15,"ch0_fld_units":["V","A","W","Hz","","°C","kWh","Wh","W","%","var"],"ch0_fld_names":["U_AC","I_AC","P_AC","F_AC","PF_AC","Temp","YieldTotal","YieldDay","P_DC","Efficiency","Q_AC"],"fld_units":["V","A","W","Wh","kWh","%"],"fld_names":["U_DC","I_DC","P_DC","YieldDay","YieldTotal","Irradiation"],"iv":[true,true,false,false,false,false,false,false,false,false]}

Habe alle API durch, aber keine gefunden, wo ich die Werte als String finden kann.

dknubben, 2023-03-05:
Darüber habe ich mich auch gewundert.
Die Werte werden wohl nun unter http://[IP]/api/record/live ausgegeben, allerdings in anderer Form und müssen irgendwie aufsummiert werden (besonders bei mehreren Modulen/Invertern). Das Datenformat mit der JSON-Ausgabe erschliesst sich mir noch nicht.
Daher funktionierte auch bei mir mein 2. ESP32 mit LCD-Anzeige der Daten nicht mehr. Ich bin nun temportär zurück zu 0.5.89

lumapu, 2023-03-07:
ja ich musste an der Stelle was ändern, da es mit mehreren Invertern zu einem zu großen JSON String geführt hat.
Die Werte der einzelnen inverter sind jetzt über `/api/inverter/id/0` bis `/api/inverter/id/9` abrufbar. Unter `/api/live` gibt es ein Array `iv` welches zeigt welche Inverter aktiv sind.

beegee3, 2023-03-07:
da es sich um live Daten handelt, fände ich es besser, diese auch unter `/api/live` abzurufen, also z.B. `/api/live/inverter/id/0`, oder vielleicht kürzer als `/api/live/iv0` usw.

lumapu, 2023-03-07:
@beegee3 `/api/live/iv0` existiert aber bis dato noch nicht

dtuuser, 2023-03-07:
@lumapu  mit /api/inverter/id/0 kann ich leben. Die Reihenfolge der Daten in den Eckigenklammern hat sich ja nicht geändert.
Danke für den api Tipp

GallusMax, 2023-03-24:
Unter /api/live waren noch aufsummierte Werte vorhanden? Sind die weiter verfügbar, oder nur noch die  Einzelwerte unter /api/inverter/id/0..9 ?

lumapu, 2023-03-25:
Ahoy hat doch nie die aufsummierten Werte per API angeboten. Die Summen wurden schon immer per Javascript im Browser erzeugt, ist auch in der aktuellen Release `0.5.66` so.

closed 2023-03-27 by lumapu

beegee3, 2023-03-30:
> Dort fehlen jetzt die topics unter inverter/total, etwa inverter/total/P_AC

die topics gibt es sobald alle Inverter (mit setting: communication enabled) Daten liefern.