lumapu/ahoy #739: Keine Anzeige der Live-Ansicht auf dem Samsung Fernseher

Mart-Muc, 2023-03-05:
Hallo Zusammen,
Für einen schnellen Überblick (für meine Familie) habe ich die „Internet“-App unseres Samsung Fernsehers (BJ2023) nutzen wollen.

alle Menüpunkte wie Einstellungen, etc werden korrekt dargestellt, nur die Live-Übersicht bleibt unausgefüllt.

Javascript habe ich kontrolliert, ist aktiviert.

Gibt es eine Lösung? Was könnte ich übersehn haben?

Danke, Mart

Mart-Muc, 2023-03-05:
V. 5.66

dtuuser, 2023-03-05:
5.93 bei Philips auch nicht

tqma1, 2023-03-05:
Kann für Samsung bestätigen 0.5.93 - keine Live Anzeige

lumapu, 2023-03-06:
gibt es da eine Art debug Konsole oder eine Info worauf der Browser basiert?

dtuuser, 2023-03-07:
Konsole habe ich beim Philips nicht gefunden. Auf dem TV läuft Android mit dem Standard Browser

lumapu, 2023-03-08:
könnt ihr versuchen genaueres über den Browser über zB. http://wieistmeinuseragent.de/ herauszufinden?

tqma1, 2023-03-09:
[image]

Samsung TV

dtuuser, 2023-03-11:
Beim Philips wie oben beim Samsung. 
Unterschied
Linux Android 9.KHTML like Gecjo, Chrome 77.0.3865.120 Safari 536.36 OPR 46 0.2207.0 OMI 4.20.3.56 Typhoon 59 Philips MT 5599

DanielR92, 2023-08-26:
Wenn ich mein Agent auf die von obrigen Daten ändere. Ändert sich bei mir nichts.
Sieht alles i.O aus. ... hmm vielleicht ist es nicht der Agent unbedingt, sondern eher der Browser vom TV was es ja falsch darstellt.

Irgendwo gab es eine Tabelle zum Thema Stylesheet... welche Browser was genau Unterstützen.

lumapu, 2023-09-14:
es ging hier vielmehr darum den eingesetzten Browser zu erfahren und dessen Fähigkeiten

closed 2024-08-13 by lumapu