lumapu/ahoy #740: Home Assistant MQTT Autodiscovery schlägt fehl wegen Nullwert in 'mf'

equinoxiq, 2023-03-05:
### Platform

ESP8266

### Model name

Wemos D1 mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

irrelevant

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.66

### Github Hash

f8fe044

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

irrelevant

### Debug Serial Log output

```shell
irrelevant
```

### Error description

Hallo und zuallererst vielen Dank für das Projekt und die Arbeit, die da drin steckt.

Mein Problem:
Eine typische MQTT Autodiscovery-Nachricht für Home Assistant sieht z.B. so aus:

`{
  "name":"Garage U_AC",
  "stat_t":"inverter/Anlagenname/ch0/U_AC",
  "unit_of_meas":"V",
  "uniq_id":"123456123456_ch0_U_AC",
  "dev":{
    "name":"Anlagenname",
    "ids":"123456123456",
    "cu":"http://192.168.1.2",
    "mf":null
  },
  "exp_aft":65,
  "dev_cla":"voltage",
  "stat_cla":"measurement"
}`

Eine Nachricht dieser Art führt zu einem Fehler in Home Assistant, bemängelt wird das 'null' für "mf", weil für dieses Feld 'null' nicht erwartet wird.
Die Spezifikation für MQTT und Home Assistant ist hier: https://www.home-assistant.io/integrations/sensor.mqtt/

Gemäß dessen ist der "manufacturer" optional und als String definiert... der Wert 'null' ist aber eben kein String.

Vielleicht war das Autodiscovery früher entspannter, bei mir zumindest (Home Assistant 2023.3.1) funktioniert wegen dieses Problems das Autodiscovery nicht.
Ersetze ich die Nachricht z.B. durch "mf":"Hoymiles" oder lasse das "mf" ganz weg, funktioniert das Autodiscovery sofort.

Vorschlag:
Da es aktuell wohl keine Möglichkeit gibt, den Wert für 'manufacturer' explizit zu setzen, den Parameter in der Autodiscovery-Nachricht weglassen (ist sowieso optional).
Oder aber: Ein Feld vorsehen, dass in den MQTT-Settings befüllt werden kann und das dann in der MQTT-Nachricht gesendet wird.

knickohr, 2023-03-05:
Vielleicht das Feld nicht als null ausgeben, sondern empty. Wenn das auch nicht funktioniert, als Workaround „ „ oder „-„

KG3RK3N, 2023-03-05:
Es gab mal Probleme bei der Auto Discovery Message das teilweise die Werte nicht vollständig waren da es zu lang wurde. Hast du die aktuelle Dev schon ausprobiert?

faeibson, 2023-03-05:
Hatte gerade genau das gleiche Problem mit dem Release. In der Dev läuft alles ("Hoymiles" wird dann als Manufacturer gesendet).

equinoxiq, 2023-03-06:
Ja... lt. dem Quellcode der aktuellen 0.5.66 wird für "mf" tatsächlich "Hoymiles" gesendet, was das Problem lösen dürfte.
Von meiner Seite habe ich das Problem schon manuell geregelt bekommen (man braucht das Discovery ja nur einmal) und der Fall hier kann von meiner Warte aus geschlossen werden.

So als Verbesserung und "nice to have" fände ich es trotzdem schön, wenn es in den MQTT Settings befüllbare Felder für Hersteller und Modell gäbe und diese dann im Discovery mit gesendet werden würden.
(Im Moment ist der Hersteller sicher immer Hoymiles aber wer weiß... vielleicht gibt es ja Nachbauten von anderen Anbietern oder andere Anbieter nutzen das gleiche Funkprotokoll, dann wäre Ahoy-DTU auch dafür tauglich.)

Danke nochmal.

closed 2023-03-06 by equinoxiq

AndreasCS, 2023-03-06:
Das Problem besteht auch mit Version 0.5.66, ist aber vermutlich davon abhängig, welche Daten (Länge) gesendet werden. Bei mir war "mf" nach wie vor "null", obwohl im Quellcode der Wert "Hoymiles" gesetzt wird (Datei /src/publisher/pubMqtt.h). Ich habe die Daten (MQTT-Pakete) im Netz mitgeschnitten und dabei ist mir aufgefallen, dass in den Messages etwas fehlt. Laut Quellcode müsste nach "mf" noch "mdl" geschickt werden. Das passiert aber nicht mehr.
Ich habe daher mal testweise die Reihenfolge geändert (zuerst "mf" dann "cu"  - Zeilen 152 / 153 getauscht). Den Code übersetzt und eingespielt. Danach wurde "mf" übertragen, bei "cu" dafür "null" und "mdl" fehlte nach wie vor.

Ich bin kein C-Experte, vermute aber, dass zu wenig Platz allokiert wird. Daher habe ich Zeile 149 geändert:
DynamicJsonDocument deviceDoc(192);
Nach dem übersetzen und übertragen an das Gerät, wurden alle Werte übertragen, inkl. "mdl" und siehe da, das Geräte taucht in HomeAssistant auf.
Vielleicht könnte sich das jemand mit etwas mehr Ahnung ansehen und "korrekt" beheben.

Gruß Andreas

KG3RK3N, 2023-03-06:
> Das Problem besteht auch mit Version 0.5.66, ist aber vermutlich davon abhängig, welche Daten (Länge) gesendet werden. Bei mir war "mf" nach wie vor "null", obwohl im Quellcode der Wert "Hoymiles" [...]

Wie gesagt, diese Thematik besteht in der aktuellen Release (0.5.66), sollte aber bei der aktuellen dev Version (~0.5.93) bereits behoben sein.